vroland/epdiy #203: Adding ESP32S3 Platform Support

THSLP13, 2022-08-29:
### A new board from Lilygo published.
### It uses ESP32S3(N16R8) as controller,and uses ED047TC1(Soft ver.) as display.
I try to make a patch to my own code,but arduino IDE only throws a lot of(99+) error to me.
When I simply compile code from your example,IDE throws errors again,and throws the same errors.
So I write the issue.

The code from Xinyuan-LilyGO works normal and fully functional,but they hasn't update their code such as update mode,they just update the pinmap and I2S Sources.
(Hint:Select S3 branch to get S3 version driver from lilyGo)

Most errors contains declaration error.
Please notice me if you need further information about this problem.

martinberlin, 2022-08-29:
Sorry but you started wrong, because there is a PR that is still not merged, that will enable EPDiy to compile in IDF v 5.0
Until that is merged, you should start on that branch, instead of EPDiy master. 
I was going to make my own PR to add this board but if you manage to do it, go ahead.

martinberlin, 2022-08-29:
@THSLP13 since it's on your own best interest to have this working on Arduino, please help me test this PR:

https://github.com/vroland/epdiy/pull/180#issuecomment-1230878963
So you can give some feedback about it. Just get that PR branch and use it to compile your EPDiy library on Arduino IDE.

THSLP13, 2022-08-29:
> @THSLP13 since it's on your own best interest to have this working on Arduino, please help me test this PR:
> 
> [#180 (comment)](https://github.com/vroland/epdiy/pull/180#issuecomment-1230878963) So you can give some feedback about it. Just get that PR branch and use it to compile your EPDiy library on Arduino IDE.

Hi,thanks for your reply.
Due to my coding skill level I couldn't give much advice,but I will do my best to test the PR.

martinberlin, 2022-08-30:
When my Lilygo S3 arrives I will try to port the updates over the existing working branch on IDF v5. 

I started on my fork but is very raw still and needs to be debugged.

vroland, 2022-09-02:
I'm curious how the S3-based board will perform. I have some normal S3 dev boards on the way since the Lilygo board is a bit expensive and not flexible enough for prototyping, but I'm confident @martinberlin can make the necessary changes :)
Since we're planning an S3-based epdiy for the future, this will be good ground work anyway.

martinberlin, 2022-09-08:
@THSLP13 you can try this:

https://github.com/martinberlin/epdiy-rotation/tree/lilygos3

Just checkout the lilygos3 branch. Note that it will work only with V5.
So far I only tried the dragon. Some statistics:

```
update_screen() 957 ms to draw dragon  on ESP32
update_screen() 2176 ms to do the same in LilyGo S3
```

It's much slower. I don't know if it would be possible to improve that (Measured with CPU at 240Mhz, is not related to CPU, is the **esp_lcd** data speed)

vroland, 2022-09-11:
@martinberlin Using the LCD peripheral to output each line separately probably also adds a lot of overhead. I'll be back at home where I have my oscilloscope in ~10 days and by then my S3 boards will hopefully have arrived, so I can do some experiments.
It would be nice to just start one LCD transaction per frame and control the remaining signals via interrupts / external logic. That would probably be the fastest way to do this.

vroland, 2022-09-11:
Nice work though for porting to esp_lcd as a prototype :) Did you try higher pixel clocks and did that make a difference?

martinberlin, 2022-09-11:
> Nice work though for porting to esp_lcd as a prototype :) Did you try higher pixel clocks and did that make a difference?

Thanks Valentin but in fact I just cleaned up Lilygo's take on this and adapted it to the latest version. I tried different clocks including the fastest one as described [here in esp_lcd](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/clk_tree.html#_CPPv424soc_periph_lcd_clk_src_t) docs. Note that the 240Khz that uses PLL_D2 drops a lcd_i80_select_periph_clock(490): unsupported clock source: 7
With the fastest clock source that I could use: LCD_CLK_SRC_PLL160M (using pclk_hz at 50 * 1000 * 1000)

The dragon takes: update_screen() **1620 ms** -> to be drawn. Still high but better than 2 seconds.

> It would be nice to just start one LCD transaction per frame and control the remaining signals via interrupts / external logic. That would probably be the fastest way to do this.

Yes that is a great idea. In my opinion this is slower because in the older version you filled a buffer while sending the other. But this one uses just 1 so it's like double slow. I'm quite sure that well optimized can be some good percentage slower with S3 that is at least 15% faster and maybe has a faster PSRAM also. 

On other rants that crossed my head [using this board](https://twitter.com/martinfasani/status/1568334101307920384): What makes me crazy about this new Lilygo S3 board is that they did something weird with the Serial. A flash monitor often delivers:
E (204) esp_image: invalid segment length 0xffffffff
E (205) boot: Factory app partition is not bootable

And you have to flash it 2 times in order to get the new program flashed. This and the fact that after 2 days of using it has a dead row make me discard this as anything that will use on the future, also is a great info for us, so we can make a powerful S3 board.

martinberlin, 2022-09-12:
Attention: Additional weirdness found in this LilygoS3 device:
[image]

This is my BLE receive JPG demo. Just an experimental receive and decode JPG that is too slow to be anything useful (100Kb in about 30 seconds, but hey consumes less than WiFi)
Well here I was having a lot of weird resets due to Brownout detector. So I connected my Nordic PPK power kit to check out what is going on.

And what I found is this:
If you toggle epd_poweron() / off() just to make short updates, that generates a spike of about 600mA that is what makes this device reset. It triggers mostly when you use BLE or WiFi, meaning for me that there is something badly designed, because in the old one I never had this resets.
And this is BLE, so it consumes a fraction of WiFi, and still generates a 600mA (or more) spike when you do an epd_poweron() that of course means immediate reset on USB since the PC can deliver only 500mA. 
So check my last commit to see how I had to remove this in order to make it work. 
Still in my humble opinion, this is a highly experimental device, that I won't think even remotely to use in any real-life project.  Is a pity, because the idea is good, and has SD card and an RTC on board. A lot of goodies, but this failures make it really untrustworthy.

vroland, 2022-09-13:
Damn, that sounds bad. Is the lilygo board still using the LT1945-based power generation? I think epdiy v5 had a rather high inrush current as well, but the TPS65185 in v6 should not have this problem. In a way, it's good to have them do some mistakes, so we can learn from them and avoid them ;)

Regarding the clock speed: If we can get DMA working and fill the buffer while the LCD peripheral is feeding the EPD, a clock speed of ~30MHz should be sufficient for all EPDs i've checked. Maybe 35MHz if we account for some overhead. So the trick will be in efficiently using DMA and both cores of the ESP.

I'm currently thinking of the following architecture:
Run the LCD peripheral continuously and use the [bounce buffer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html#bounce-buffer-only) method, where an interrupt is called to re-fill the LCD peripheral buffer. 
Then, we start with the difference buffer in PSRAM as in epdiy. We have an additional queue / buffer of lines ready for output in SRAM. Two tasks for both cores of the ESP32 run continuously and fill the line queue in SRAM by performing the waveform lookup on the difference buffer / framebuffer. Every time the LCD peripheral buffer runs low, it calls an interrupt to fill its bounce buffer. In that interrupt, we fetch lines from the line buffer and put them in the LCD peripheral buffer. 
This does not use DMA, but probably has the lowest latency.
Alternatively, we could have two framebuffers in PSRAM instead of having a line buffer in SRAM. We fill one framebuffer with the two CPU tasks while the other is written out by the LCD peripheral using DMA. This uses less CPU resources but requires a write-back to PSRAM, which may or may not be slower. An it adds one frame = ~12ms delay. But if the first method does not work out it is worth a try, since memory bandwith in the ESP32-S3 is much higher than in the ESP32.

martinberlin, 2022-09-13:
Hi Valentin,

> Is the lilygo board still using the LT1945-based power generation?

Yes they are still using them: U2 ( [Schematics](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/schematic) )
But I think they made some modifications on this board because the other one had no this spikes. Maybe a smaller capacitor, no idea since I'm not familiar with the older PCB versions like you ;)
About the ideas, they sound great!  
Would love to follow them and test how they perform.
By the way I've got a full refund for this board and I have no use for it at all, so if you want to have one to experiment, is all yours.
I would love to see a design based on S3, I think the [ESP32-S3-WROOM-1(N8R2) with 2MB PSRAM](https://jlcpcb.com/partdetail/3198302-ESP32_S3_WROOM_1_N8R2/C2913204) would suffice (Still about 190 available)

vroland, 2022-09-13:
I think the S3 dev board will probably arrive before your lilygo board, so I'm not sure if it's worth it. Once we have our own board it will (hopefully) be obsolete anyway ;) But if you really don't need it / want it it could be useful as a backup :)
Regarding the module: My dev kit will be N8R8 and if I remember correctly, there is a difference in how the PSRAM is connected on some modules. I.e., some modules support octal SPI and some don't. So in case we go with the double framebuffer in PSRAM method, having the higher bandwith is probably nice. So let's see what we need, hopefully JLCPCB will have more stock soon anyway, since the S3 is relatively new.

liangyingy, 2022-09-16:
> Attention: Additional weirdness found in this LilygoS3 device: [image]
> 
> This is my BLE receive JPG demo. Just an experimental receive and decode JPG that is too slow to be anything useful (100Kb in about 30 seconds, but hey consumes less than WiFi) Well here I was having a lot of weird resets due to Brownout detector. So I connected my Nordic PPK power kit to check out what is going on.
> 
> And what I found is this: If you toggle epd_poweron() / off() just to make short updates, that generates a spike of about 600mA that is what makes this device reset. It triggers mostly when you use BLE or WiFi, meaning for me that there is something badly designed, because in the old one I never had this resets. And this is BLE, so it consumes a fraction of WiFi, and still generates a 600mA (or more) spike when you do an epd_poweron() that of course means immediate reset on USB since the PC can deliver only 500mA. So check my last commit to see how I had to remove this in order to make it work. Here in more detail: [image]
> 
> Still in my humble opinion, this is a highly experimental device, that I won't think even remotely to use in any real-life project. Is a pity, because the idea is good, and has SD card and an RTC on board. A lot of goodies, but this failures make it really untrustworthy.
> 
> [Short video explanation](https://twitter.com/martinfasani/status/1569428745286549504)
> 
> To discard that is my part of porting this to the new version of EPDiy I just ran [their code](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3) with a simple program that does only this in arduino-esp32:
> 
> ```c
> void loop() {
>   epd_poweron();
>   epd_clear();
>   epd_poweroff();
>   delay(1000);
> }
> ```
> 
> No Wifi, or BLE, just that and this is the result: [image]
> 
> About 294 mA spikes every time epd_poweron() triggers the HV pumps (Or what is that otherwise) But if you use this with WiFi or BLE then it sums up to the other consumption making that huge spike that surpasses 500mA and that's the real problem. Apart that it resets it will eat your battery like cookies!

Here are our test results:

[image]

The test code is:

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/perfect-refresh/examples/drawImages

The full refresh time is about 450ms.

liangyingy, 2022-09-16:
I think double buffering, the improvement may be less than 5%, because the current method only takes some more copy time than buffering, which is almost negligible.

martinberlin, 2022-09-16:
My test was after being ported to EPDiy using IDF v5.

```
update_screen() 957 ms to draw dragon  on ESP32
update_screen() 2176 ms to do the same in LilyGo S3 -> Could get it down to around 1600 ms. updating some settings.
```

That is the update_screen() which is sending the parallel data. Note that this version supports Waveforms, I don't have any idea if that can add more time, maybe Valentin can tell more about that. 
What I know is that in my tests this S3 version is definitively slower than the old one using EPDiy as a controller. Now if this new version is itself slower than the one that you are using I don't know, but I could not replicate with other epapers. 

About the spikes in consumption in epd_poweron() @liangyingy you should take a look for yourself. This will reset the device if it's using while other consuming things are taking place, like WiFi or BLE. And in my humble opinion, this makes your S3 device hardly usable, if you need to draw stuff while downloading an image. 
The workaround is to leave the epaper on at all time, if you need to draw a progress bar, something that is unwise if you make something that is battery powered. 
The last device you designed based on ESP32-WROVER has not this problem. And works stable with all examples, while this new one does not, I had to modify the BLE example and left it with epd_poweron() all the time in order to render the image.

liangyingy, 2022-09-16:
We should have a unified interpretation of the concept of full refresh.

The full refresh time I think is the time it takes to call `epd_hl_update_screen` or `epd_draw_image`.

liangyingy, 2022-09-16:
There is a reset problem using ble or wifi, is it found by which code test? We want to reproduce this bug.

martinberlin, 2022-09-16:
Hi Liang,
sure I agree with this point.
You can find my BLE example here:

https://github.com/martinberlin/epdiy-rotation/tree/lilygos3
pull that repository.  It's on examples/ble-image
I'm sure it can be ported to Arduino esp32 pretty easily.

Here you can find an article I wrote about [BLE receive JPG](https://fasani.de/2022/08/31/making-our-own-esp32s3-ble-receive-jpeg-image-firmware/).
As BLE client to send the image you can use my service: https://cale.es
I can create an account for you if you wish with the right screen and a demo image gallery. Then you just need to build and flash the ble-image Firmware.
This will work now in S3 since I added some **ifdef** pre-compiler conditions, to avoid doing epd_poweroff() in your device.
Just tell me if you need help, I can also try to port it to arduino-esp32, and make a pull request in your Repository.

liangyingy, 2022-09-19:
@martinberlin 

I ran [ble-image](https://github.com/martinberlin/epdiy-rotation/tree/lilygos3/examples/ble-image) on both boards and no device reset occurred.

The current is about 250ma.

[image]

martinberlin, 2022-09-19:
It occurs only when it receives data. When it's idle of course does not consume.
Note that I also commented the epd_poweoff() just for your board since the spike comes when is in off state and it's triggered to on while receiving data.
Try to leave the power on /off and send some data using cale.es client and you will see what I mean. If you want I can prepare an account for you with the right configuration and image gallery so you can log in and try it right away. My email is on my GitHub profile

liangyingy, 2022-09-19:
The above test result is to use https://cale.es/ to send the image to the device through ble.

[image]

martinberlin, 2022-09-19:
Great. But did you tried to remove my ifdef pre-compiler directives so it does an epd_poweron() / epd_poweroff()

Try to change: 

```C
void epaper_power_off() {
    #ifndef CONFIG_EPD_BOARD_REVISION_LILYGO_S3_47
      epd_poweroff();
    #endif
}
```

That function wrapper I did just to avoid switching power on /off and avoid the reset. Just leave it with the epd_poweroff() and try to receive the image again.

liangyingy, 2022-09-20:
@martinberlin 

You can apply the following patch and run it on your device if there is still a device reset issue:

Please add the following patch to [epdiy-rotation](https://github.com/martinberlin/epdiy-rotation):

```diff
diff --git a/examples/ble-image/main/gatt_server_receive.cpp b/examples/ble-image/main/gatt_server_receive.cpp
index d38744c..1357820 100644
--- a/examples/ble-image/main/gatt_server_receive.cpp
+++ b/examples/ble-image/main/gatt_server_receive.cpp
@@ -222,9 +222,9 @@ extern "C"
  *        That means you cannot turn on -> draw something fast -> turn off or you will have a Reset
  */
 void epaper_power_off() {
-    #ifndef CONFIG_EPD_BOARD_REVISION_LILYGO_S3_47
+    // #ifndef CONFIG_EPD_BOARD_REVISION_LILYGO_S3_47
       epd_poweroff();
-    #endif
+    // #endif
 }
 
 void progressBar(long processed, long total)
@@ -303,7 +303,7 @@ int decodeJpeg(uint8_t *source_buf, int xpos, int ypos) {
     if (jpeg.decodeDither(dither_space, 0))
       {
         time_decomp = (esp_timer_get_time() - decode_start)/1000 - time_render;
-        ESP_LOGI("decode", "%d ms - %dx%d image MCUs:%d", time_decomp, jpeg.getWidth(), jpeg.getHeight(), mcu_count);
+        ESP_LOGI("decode", "%ld ms - %dx%d image MCUs:%d", time_decomp, jpeg.getWidth(), jpeg.getHeight(), mcu_count);
       } else {
         ESP_LOGE("jpeg.decode", "Failed with error: %d", jpeg.getLastError());
       }
@@ -489,7 +489,7 @@ static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
     }
     case ESP_GATTS_READ_EVT: 
     {
-        ESP_LOGI(GATTS_TAG, "GATT_READ_EVT, conn_id %d, trans_id %d, handle %d\n", param->read.conn_id, param->read.trans_id, param->read.handle);
+        ESP_LOGI(GATTS_TAG, "GATT_READ_EVT, conn_id %d, trans_id %ld, handle %d\n", param->read.conn_id, param->read.trans_id, param->read.handle);
         esp_gatt_rsp_t rsp;
         memset(&rsp, 0, sizeof(esp_gatt_rsp_t));
         rsp.attr_value.handle = param->read.handle;
@@ -518,7 +518,7 @@ static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
                 (param->write.value[2] << 8) +
                 (param->write.value[3] << 16) +
                 (param->write.value[4] << 24);
-                ESP_LOGI(GATTS_TAG, "0x01 content-lenght received:%d", received_length);
+                ESP_LOGI(GATTS_TAG, "0x01 content-lenght received:%ld", received_length);
 
                 esp_log_buffer_hex(GATTS_TAG, param->write.value, param->write.len);
             }
@@ -531,8 +531,8 @@ static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
                 time_receive = (esp_timer_get_time()-start_time)/1000;
                 decodeJpeg(source_buf, 0, 0);
 
-                ESP_LOGI("ble-rec", "%d ms - download", time_receive);
-                ESP_LOGI("render", "%d ms - copying pix (JPEG_CPY_FRAMEBUFFER:%d)", time_render, JPEG_CPY_FRAMEBUFFER);
+                ESP_LOGI("ble-rec", "%ld ms - download", time_receive);
+                ESP_LOGI("render", "%ld ms - copying pix (JPEG_CPY_FRAMEBUFFER:%d)", time_render, JPEG_CPY_FRAMEBUFFER);
 
                 epd_poweron();
                 epd_hl_update_screen(&hl, MODE_GC16, temperature);
@@ -546,7 +546,7 @@ static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
                 memcpy(&source_buf[img_buf_pos], param->write.value, param->write.len);
                 img_buf_pos += param->write.len;
 
-                ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, value len %d, total:%d", param->write.len, img_buf_pos);
+                ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, value len %d, total:%ld", param->write.len, img_buf_pos);
             }
 
             // Debug received bytes
[... 66 lines cut ...]
+  push_cfg(&config_reg);
+
+  config_reg.ep_stv = false;
+  push_cfg(&config_reg);
 }
 
 static void epd_board_poweroff_touch(epd_ctrl_state_t *state) {
diff --git a/src/epd_driver/i2s_data_bus_s3.c b/src/epd_driver/i2s_data_bus_s3.c
index 48b0b8f..7910f5b 100644
--- a/src/epd_driver/i2s_data_bus_s3.c
+++ b/src/epd_driver/i2s_data_bus_s3.c
@@ -53,7 +53,7 @@ static bool notify_trans_done(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_
 
 void i2s_bus_init(i2s_bus_config *cfg)
 {
-    ESP_LOGI(TAG, "Initialize Intel 8080 bus\n6:%d 7:%d 4:%d 5:%d 2:%d 3:%d 0:%d 1:%d\nstart_pulse:%d clock:%d\nepd_row_width:%d",
+    ESP_LOGI(TAG, "Initialize Intel 8080 bus\n6:%d 7:%d 4:%d 5:%d 2:%d 3:%d 0:%d 1:%d\nstart_pulse:%d clock:%d\nepd_row_width:%ld",
     cfg->data_6,cfg->data_7,cfg->data_4,cfg->data_5,cfg->data_2,cfg->data_3,cfg->data_0,
     cfg->data_1, cfg->start_pulse, cfg->clock, cfg->epd_row_width);
 
```

Here is the current when rendered (at the red arrow):

[image]

martinberlin, 2022-09-20:
Nice catch 👍
I will try this morning, report how it works, and leave that change ready so we can merge this new board in epdiy. Thanks for your time!

martinberlin, 2022-09-20:
Thanks @liangyingy 
This has reduced the consumption when epd_poweron is triggered (and epd is off)
I’m just experiencing a strange f effect after this update. If I send a JPG image that is not full screen and there is white space in the side then I see some horizontal lines in the esp32S3 version. Using the same demo in the older ESP32 display this does not happen:
[image]

please note that the T5 below has just different gamma value so it’s lighter. Note that this seems to appear each time the progress bar is updated.
But check the strange horizontal lines, can you reproduce that?

liangyingy, 2022-09-21:
Yes, we have reproduced the problem.

vroland, 2022-09-26:
@martinberlin Btw, I've got an ESP32S3 dev board a few days ago and so far, things are looking promising. I can use the LCD peripheral for all but the CKV signal, which is again generated by the RMT peripheral. Hence, no need for line-wise output. The limiting factor will be getting the framebuffer from PSRAM and looking up the waveform. The prototype is still in a rough shape, but I can draw the dragon in ~850ms single-threaded.

martinberlin, 2022-09-27:
Awesome @vroland !
That's great news. 

> I'm currently thinking of the following architecture

Looking forward we can try this architecture's you proposed before to see if we can optimize even further. I would like to know also if I can help you testing that in Lilygo S3 hardware. 
By the way thanks @liangyingy for helping solving this lines bug, if it can be solved by software, and I'm also interested to know Valentin if you seen that when updating the epaper (Check my post before)
I would like to know if it's something that happens only with BLE image download (Don't think so, since JPG is decompressed at the end) or it's just reproducible drawing any image leaving some white space in the left.
This kind of effect is the first time I see it and I don't have any idea of what triggers it. I will keep on helping wherever I can. Also planned to design a Lilygo Adapter to see if we can also [drive a 9.7" with their hardware](https://github.com/martinberlin/H-spi-adapters/tree/master/S3-AXE544127A-to-33), it's kind of a shot in the dark, but I would like to give it a try so I will send 5 PCB's to fabrication.

vroland, 2022-10-09:
I have some news to report on the S3 version:
I implemented multithreading and now I can get ~550ms updates with the vendor waveform on my 9.7" screen and ~450ms updates if I push it a bit further, though that can become unstable. EPD builtin waveforms are even faster, since they use less phases :)
However, there's a setback: I was experimenting with driving a 16bit display as well, and while I can get some output, I noticed that the S3 LCD peripheral apparently only supports up to 1024 lines of total vertical resolution, including blanking :/
That's too little for the 1872 x 1404 display I'm testing. Looks like we need to do some more hacking: my current idea would be to either re-start the LCD peripheral in the middle of the frame to draw the frame in two halves, or actually drawing two frames of different sizes right after another. Unfortunately, both solutions will force me to use the low-level abstractions and not their driver component :/
@martinberlin: While the prototype is relatively simple (see picture, just a bunch of jumper wires and voltages from an epdiy board), an oscilloscope is very much necessary at the current state, since I do some hacking with the RMT and LCD peripherals. However, once that's stable you should be able to follow along with just an esp32 dev board :)
[image]

martinberlin, 2022-10-10:
That's great progress!
About the LCD peripheral limitation there is not any other way to overcome this?
Like no configuration in menuconfig or similar that lets you have a bigger vertical rows resolution?
That's a showstopper... What a pity.
But well if not as you say is possible to draw in 2 rounds. Interesting research.
Do you think in some months we can have a new EPDiy S3 based controller?

vroland, 2022-10-11:
I had a look into the technical reference manual, and the corresponding register only has 10bits for hight, so 0..1023. Since that's how the hardware is built, there's afaik not much I can do about that :/ 
Still, for the smaller displays that fit in one frame this approach is very fast, and the larger displays are going to be slower anyways. I'll definitely do an S3-based epidy, the results for the smaller displays where just too good ;) How long will depend on how many of those hidden issues I find next, though :D

huming2207, 2022-10-11:
Hi all, I've just roughly read through this thread and a bit of the code in `s3-lcd` branch and I've got some questions: 

1. If we only consider 8-bit panels, would it be faster if we use SPI in octal mode? I remember the SPI2 on ESP32S3 can do octal SPI up to 80MHz.
2. For @vroland I'm not quite understand why you have the 1024 resolution limit? Isn't that LCD peripheral can be used as a plain 8080 parallel port? 
3. I've proven 16MB PSRAM (APS12808L) works on ESP32-S3 earlier, see [here](https://github.com/huming2207/jumbo-s3). Since we have a bigger caching space, would it be helpful if we somehow pre-cache the waveform while the image is being downloaded?

martinberlin, 2022-10-11:
Dear @huming2207 

> I had a look into the technical reference manual, and the corresponding register only has 10bits for hight, so 0..1023

If that's how the hardware is built, reading vroland comment above, there is no way you are going to overcome this limitation. If the register has 10 bits and with that you can count up to 1024 then that is the limit. But it will be not a problem to render the larger panel even with that limitation.

vroland, 2022-11-02:
@martinberlin @huming2207 I'm using the RGB mode, since with that mode I can directly derive some necessary EPD signals that I can't get from the I8080 bus :/ Also, transfer speed is not the limiting factor, both with the old I2S driver nor with the LCD peripheral. Rather, it's processing the data using the wavform lookup table.

Nevertheless, after throwing out the driver by espressif, some hacking and plenty of debugging I goth something working!
This is all pretty hacked together right now, but it is possible to chain multiple LCD frames and adjust timings, etc. in the frame end interrupt. Then you need to fiddle a bit with DMA and signal delays to make sure no data is lost. 

https://user-images.githubusercontent.com/5757380/199618844-c505e831-6e10-404d-912c-af1800ca9cc1.mp4

The display is a 1872 x 1404 7.8" 16-bit parallel display. There are still some visual artifacts (these vertical, light "columns") and the waveform doesn't fit for the display, but it's progress ;)

huming2207, 2022-11-02:
Nice work!

>  I'm using the RGB mode, since with that mode I can directly derive some necessary EPD signals that I can't get from the I8080 bus 

Thanks, I've just roughly read your code again and now I somewhat understand why RGB mode is necessary :+1: 

> Also, transfer speed is not the limiting factor, both with the old I2S driver nor with the LCD peripheral. Rather, it's processing the data using the waveform lookup table.

Yes indeed. I remember earlier this year I've done some testing before by writing a big chunk of random data to PSRAM, and reading it out from PSRAM. Even though I've set the PSRAM in octal 80MHz and the physical limit should be higher, it was still around 30MB/s anyway. I guess it was either misconfigured or miscalculated or maybe because of the overhead of RTOS and Espressif's low-level stuff. 

I guess maybe we can improve the performance by using either a larger 16MB PSRAM (by sacrificing more space & optimizing access alignment??) or something like this instead of the FreeRTOS queue to avoid unnecessary memory copy: https://github.com/DNedic/lfbb

vroland, 2022-11-03:
With the ESP32-S3 I could get up to 80MB/s, though the theoretical limit should be 160MB/s. I guess some is protocol overheads and some is the CPU just waiting for the cache to fill, since CPU-based reads always go through the cache. At least for the S3, you can observe this by changing the DRAM cache line size.
Still, avoiding the queue overhead could be very nice, so thanks for the recommendation! Aside from that, I want to experiment with either copying the data via DMA to bypass the cache or use cache preloading to do this...

huming2207, 2022-11-03:
> I could get up to 80MB/s, though the theoretical limit should be 160MB/s

Hmm will it be that fast?? Have you tried filling in random data for a few MBs? (which lets the cache run out)?

> Aside from that, I want to experiment with either copying the data via DMA to bypass the cache or use cache preloading to do this...

Yea true, I think it might be helpful to use the DMA-based async memcpy thing together with that LFBB library.

vroland, 2022-11-07:
I tried adding some cache preloading to my speedtest and I got ~114MB/s max. I wonder if I could squeeze out more, but not sure how. I also tried using the async memcpy driver, but somehow it was very slow, only 30-40MB/s.
Regarding the ringbuffer library: Somehow it looks like the library you linked gets in a weird state where the buffer is full but I still can't read data. After trying to debug it for way too long I threw it out again and wrote my own queue, which seems to work.

vroland, 2022-11-07:
https://user-images.githubusercontent.com/5757380/200430958-20964c8c-7a4c-4cd6-ab0c-4aacd758350e.mp4

Btw, with cache prefetching the 1200x825 display goes brrrr: 480ms for a full draw (of 38 frames with vendor waveforms). All that with that wonky prototype :D

huming2207, 2022-11-07:
> Btw, with cache prefetching the 1200x825 display goes brrrr: 480ms for a full draw (of 38 frames with vendor waveforms). All that with that wonky prototype :D

Wow that's really fast! Nice work!

> I tried adding some cache preloading to my speedtest and I got ~114MB/s max. I wonder if I could squeeze out more, but not sure how. I also tried using the async memcpy driver, but somehow it was very slow, only 30-40MB/s.

I think that 114MB/s might be hitting the SRAM cache? I think for an 80MHz Octal-SPI (8-bit) connection, it at most reaches 80MB/s in one direction? Since the CPU also needs to issue fetch commands to the PSRAM, plus some overheads on FreeRTOS, I think 30-40MB/s is the speed we can actually get at most.

vroland, 2022-11-07:
It's a DDR (double data rate) SPIRAM, clocking out data at the positive and negative edge, hence the 160MB/s theoretical maximum. With the above we already have 1200 * 825 * 38 / 0.48 B/s, which is ~78MB/s ;)

huming2207, 2022-11-08:
> It's a DDR (double data rate) SPIRAM, clocking out data at the positive and negative edge, hence the 160MB/s theoretical maximum. With the above we already have 1200 * 825 * 38 / 0.48 B/s, which is ~78MB/s ;)

Oh yes, it's DDR... 😅 Then maybe I misconfigured my stuff when I did the testing earlier...

martinberlin, 2022-11-12:
Dear @vroland 
Tried to test what you are doing today, copying what you did in the s3_lcd branch, to my working S3 Lilygo branch (Yes I have one that is still working ;) 
And so far could not make it work. This errors are what is coming on IDF 5:

```
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c: In function 'rmt_pulse_init':
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c:54:18: error: 'ETS_RMT_INTR_SOURCE' undeclared (first use in this function)
   54 |   esp_intr_alloc(ETS_RMT_INTR_SOURCE, ESP_INTR_FLAG_LEVEL3,
      |                  ^~~~~~~~~~~~~~~~~~~
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c:54:18: note: each undeclared identifier is reported only once for each function it appears in
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c: In function 'pulse_ckv_ticks':
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c:82:6: error: 'rmt_dev_t' has no member named 'conf_ch'
   82 |   RMT.conf_ch[row_rmt_config.channel].conf1.mem_rd_rst = 1;
      |      ^
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c:83:6: error: 'rmt_dev_t' has no member named 'conf_ch'
   83 |   RMT.conf_ch[row_rmt_config.channel].conf1.mem_owner = RMT_MEM_OWNER_TX;
      |      ^
/home/martin/esp/projects/epdiy-rotation/src/epd_driver/rmt_pulse.c:84:6: error: 'rmt_dev_t' has no member named 'conf_ch'
   84 |   RMT.conf_ch[row_rmt_config.channel].conf1.tx_start = 1;
```

With what version of idf are you building in this branch?  I'm missing something or RMT needs to be changed in the future?

lbuque, 2022-11-30:
> Thanks @liangyingy This has reduced the consumption when epd_poweron is triggered (and epd is off) so now I can receive the full BLE image. I’m just experiencing a strange effect after this update. If I send a JPG image that is not full screen and there is white space in the side then I see some horizontal lines in the esp32S3 version, only on the left side of the display. Using the same demo in the older ESP32 display this does not happen: [image]
> 
> please note that the T5 below has just different gamma value so it’s lighter. Note that this seems to appear each time the progress bar is updated. Update: the horizontal lines appear also when the progress bar is disabled so it does not seem to be that the issue. But is visible only when the image is white.
> 
> But check the strange horizontal lines, can you reproduce that?

@martinberlin 
We recently received customer feedback that T5-4.7 Plus has a serious refresh problem on epdiy, if there is any, please let us know and try to solve it.

In addition, We used [epd_draw_image](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/src/epd_driver.c#L764) on LilyGo-EPD47 to refresh the picture, and did not find afterimages.

For example this code:

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/examples/demo/demo.ino#L92-L103

Thanks!!!

lanistor, 2022-11-30:
> output2.mp4 
> Btw, with cache prefetching the 1200x825 display goes brrrr: 480ms for a full draw (of 38 frames with vendor waveforms). All that with that wonky prototype :D

Is this by S3? It render so fast, compared to ESP32.

martinberlin, 2023-07-30:
Related to #207 @vroland when S3 V7 is merged we can also close this one since there are currently 2 Issues open regarding esp32S3 implementation

martinberlin, 2023-08-25:
Note: I closed #207 since there are 2 issues about S3 support.

I will copy here an additional wish before the pull-request is merged:

#### Question for @vroland and interested parties: 

Every time I use **epdiy** as an idf component in a project I face the challenge of adding manually a CMakeFiles. Could we add an additional one like this in the root of the project so it's easier for everybody:

```
set(basedir "src/epdiy")
set(app_sources "${basedir}/epdiy.c"
                "${basedir}/render.c"
                "${basedir}/output_lcd/render_lcd.c"
                "${basedir}/output_lcd/lcd_driver.c"
                "${basedir}/output_i2s/render_i2s.c"
                "${basedir}/output_i2s/rmt_pulse.c"
                "${basedir}/output_i2s/i2s_data_bus.c"
                "${basedir}/output_common/lut.c"
                "${basedir}/output_common/line_queue.c"
                "${basedir}/output_common/render_context.c"
                "${basedir}/font.c"
                "${basedir}/displays.c"
                "${basedir}/board_specific.c"
                "${basedir}/builtin_waveforms.c"
                "${basedir}/highlevel.c"
                "${basedir}/board/tps65185.c"
                "${basedir}/board/pca9555.c"
                "${basedir}/board/epd_board.c"
                "${basedir}/board/epd_board_common.c"
                "${basedir}/board/epd_board_lilygo_t5_47.c"
                "${basedir}/board/epd_board_v2_v3.c"
                "${basedir}/board/epd_board_v4.c"
                "${basedir}/board/epd_board_v5.c"
                "${basedir}/board/epd_board_v6.c"
                "${basedir}/board/epd_board_v7.c"
)

# Can also use IDF_VER for the full esp-idf version string but that is harder to parse. i.e. v4.1.1, v5.0-beta1, etc
if (${IDF_VERSION_MAJOR} GREATER 4)
    idf_component_register(SRCS ${app_sources} INCLUDE_DIRS "${basedir}/include" REQUIRES driver esp_timer esp_adc esp_lcd)
else()
    idf_component_register(SRCS ${app_sources} INCLUDE_DIRS "${basedir}/include" REQUIRES esp_adc_cal esp_timer)
endif()

# Does not seem to exist anymore lut.c in new version?
#set_source_files_properties("${basedir}/lut.c" PROPERTIES COMPILE_OPTIONS -mno-fix-esp32-psram-cache-issue)
```

Just a suggestion. Also this lut.c if I'm looking right does not exist anymore in latest version that is still not merged. 
This will make in my humble opinion this component much easier to add in future projects.

martinberlin, 2023-08-25:
Hello @THSLP13
if you switch to **s3_lcd** branch and create a board configuration that matches Lilygo S3 pin-out this should be already working for the board. Maybe there are some additional things to correct but the parallel communication is working already in another boards so feel free to give it a try. Sorry this time I won't add this board myself since I don't have the hardware and have really little interest into buying one.

lanistor, 2023-08-26:
The rendering clarity or rendering speed of v7 board, seems not so good as v5, v6 board. 
Here is videos of the demo code, we can sees the behavior clearly, and all the screen resolution is 1024 * 758, model is `ED060XCD`

v7 board:
https://github.com/lanistor/assets/blob/master/epaper/v7/v7-board-demo.mov

v6 board:
https://github.com/lanistor/assets/blob/master/epaper/v7/v6-board-demo.mov

mickeprag, 2023-08-28:
> Every time I use epdiy as an idf component in a project I face the challenge of adding manually a CMakeFiles. Could we add an additional one like this in the root of the project so it's easier for everybody:

I think this is a good idea. We should make it easy to integrate with different build platforms. Also we should keep the PlatformIO release in sync with the GitHub releases. I can help with writing a GitHub Action for this if that is something you want help with?

vroland, 2023-08-29:
@lanistor In the videos v7 seems faster to me, just in the v6 demo the animation delays are removed?

And yes, I'll look into if there is a better way to structure the component to not have the hacky arduino-only headers and make it compatible for both.

lanistor, 2023-08-31:
@vroland I don't know, i didn't change the demo code, i see it's time interval code is `vTaskDealy(100)`, same as v6's demo.

If not consider the rendering invertal, focus on the rendering clarity, it seems to has some little problem, i mark the difference in the below images:

v6-board:
[image]

v7-board:
[image]

Besides, if i change the mode to `MODE_DU` in the demo code(v7 board), the rendering font becomes blurry.
[image]

vroland, 2023-08-31:
Yes, that is a waveform issue. Currently, v7 ignores the timings in the epdiy waveforms, making them not work correctly. The vendor waveforms should work though, because they assume constant timing. Can you try assigning the ED047TC2 waveform to the display? Then the rendering should be sharper.

martinberlin, 2023-11-02:
S3 support is added. Only someone needs to make a pull request to add the new Lilygo S3 board. I currently do not own that hardware so this time someone else needs to add the board.
Is not hard just check how existing boards are added and use the Lilygo Schematics or image with GPIOs for reference

ggros, 2023-11-13:
@martinberlin by S3 support added, you mean it is avail in main branch?
If so I will test, I do have an ED047TC1 LilyGo

vroland, 2023-11-13:
Adding support for their board should be straightforward, but we don't have any hardware to test. I don't feel like buying one from my own money though, i'd rather spend it on epdiy boards. 
But in theory, it would just be adding a new board definition with the correct pins and it should work, if they didn't do any unexpected changes from the ESP32 version.

martinberlin, 2023-11-13:
@ggros esp32s3 is already supported. As @vroland mentioned you need to add a board.
If no one wants to do the pull request I can do it, but someone has to test it because I also don't have a Lilygo S3.
https://github.com/martinberlin/epdiy-rotation/blob/lilygos3/src/epd_driver/board/epd_board_lilygo_s3_47.c -> That's when I tested it. 
I cannot make a PR from there thought since I started from an outdated master. Just tell me if you are interested to try it once is implemented.

ggros, 2023-11-13:
Yes interested to test, if I get it to work I will do the PR.

martinberlin, 2023-11-14:
Hello @vroland 
I just wanted to inform that if someone wants to resync their fork with this repository (just did)
the latest commit you see making a **git log** is this one:

commit 615648e7dc9262dd76bc2e0e2bc9e77c687b5237 (HEAD, upstream/master, origin/master, origin/HEAD, dev/bmp_download)
Author: jdoubleu <hi@jdoubleu.de>
Date:   Tue Aug 29 16:28:31 2023 +0200

    fix temperature readings

I don't know exactly why. Maybe it's because some older commits where rebased, but the fact is, that is going to be harder to contribute after this. Maybe we try to make a "fake commit" just updating an space in the README or whatever so it's syncs again?
Or that is a bad idea?

[14 more replies]

closed 2024-01-03 by martinberlin