vroland/epdiy #204: vbaksa LilyGo T4.7 fails to compile due to error in epdiy - HELP ;)

galandilias, 2022-09-26:
Source of the problem is here:

https://github.com/esphome/issues/issues/3372

Other folks have no idea how to solve that - I suppose that sth is either improperly imported or set - @vroland - can you please help :)

martinberlin, 2022-09-26:
This must be because the IDF version is different. Can you find out what version it uses @galandilias ?

```
error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
```

galandilias, 2022-09-26:
Well - as far as I understand all of that ;)

https://esphome.io/components/esp32.html

[image]

Should be the recommended one - I can try to compile with any other perhaps - just point me one with which it should work :)

galandilias, 2022-09-26:
okay - version issue indeed - I have tried like below - and works:
`esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1`

Now - how to pass this to esphome maintainers? I guess that defaults have to be altered...

martinberlin, 2022-09-27:
Ok I'm not sure how this can be solved in our side so it builds also with an older version. But versions move forward so for sure in a near future they will use a new version as the default.

galandilias, 2022-09-27:
I am a bit afraid about that - because the ticket I refer to is already couple months old ;) but lets see.

martinberlin, 2023-03-03:
Closing this since it's not really related to the library but more to how the esphome handles the library. Just contact me if it needs to be reopened @galandilias

closed 2023-03-03 by martinberlin