vroland/epdiy #205: Non-updated parts of screen fade more gray over time

dot4qu, 2022-10-21:
I've found that on boot when I run `epd_fullclear`, the entire screen is set a flat white and anything drawn stands out with a good contrast as expected. Over time I updated small portions of the screen (some text, some raw data images), leaving the full screen alone and never refreshing the entire thing.
This results in "boxes" around the re-rendered areas that are the same good white as the initial draw. The areas of the screen that aren't updated seem to degrade after the first 2 or 3 partial draws to more gray. This results in a screen that you can clearly see the bounding boxes of the updated areas and doesn't look very good. Interestingly enough, the outer edges of the screen seem to stay at the original white, suggesting that only columns/rows that have been partially updated start to fade their non-updated pixels.

Here is a picture to show what I mean more clearly:
[image]

Is there something I'm misunderstanding around waveforms, or is this just a drawback of doing partial screen updates? Maybe a different update mode instead of `MODE_GL16` is better?

The functions I'm using for partial updates are things like `epd_draw_rect` or `epd_write_text`, and then executing my render function which the main logic is:
```
    epd_poweron();
    enum EpdDrawError err = epd_hl_update_screen(&hl, MODE_GL16, 25);
    (void)err;
    epd_poweroff();
```

vroland, 2022-11-02:
This will always happend with partial screen updates evetually, in my experience.To slow this effect down, make sure VCOM is properly calibrated to your display, maybe also try to set it a little bit higher / lower than the value printed on the cable. The waveform doesn't change much, since I suspect the greying is caused by the particles slowly mixing again, which is sped up if VCOM is slightly wrong. Hope it helps :)

seishinkouki, 2022-11-03:
This always happen, Even commercial products need a full-update after several partial-update to eliminate this effect

dot4qu, 2022-11-03:
Ah I forgot about VCOM I can fiddle with that. I know I originally set it to the voltage printed on the screen barcode but I'll try tweaking it.

@seishinkouki I don't see this at all on my amazon kindle paper white when it does partial updates, but maybe there aren't enough partial updates in a row to show the behaviour.

I wouldn't have a problem doing a periodic full screen refresh, but is there a better way or mode to to do that doesn't take ~5-10 seconds? Right now I'd have to to a full screen erase which causes multiple flashes and takes multiple seconds, then redraw all the content. It would be nice to basically just blank the whole screen white with no flashing then quickly re-render everything on top (like a kindle looks if you 'turn the page' of a book, no noticeable flashing or blank screen)

martinberlin, 2022-11-03:
Also note that Amazon Kindle has a full fledged Linux inside. Speed of sending the data / waveform cannot be matched with EPDiy. 
I usually do every now and then a full refresh, that's the best way to go in my opinion!

vroland, 2023-01-01:
If you have the vendor waveforms of the display, just re-drawing the image the the GC16 waveform would be a nice way to refresh the display. It could be worth exploring if this works with the built-in waveforms as well, and offer an API function for a "single-flash", slightly lower quality, screen refresh.

dot4qu, 2023-02-04:
Okay I know I've left this open super long and I'm sorry but I have been working on it. My solution so far isn't perfect (read: a dirty hack) but it at least mitigates the individual box problem.

I manually mark all lines as 'dirty' before calling the epdiy high-level render function by XORing all the bits in the front and rear frame buffer. I technically don't even have to flash to erase anything, as once I do that, turn the display on, push the frame buffer, then turn it off, the previously black pixels fade away to white. It's not perfect because there's about 1 second where the new pixels and old pixels are all black and they overlap, but for something like updating the minute in a time it's not obtrusive or too noticeable.

The problem I'm running into now is that my screen aggressively washes out to gray after I write to it. Like, if I full clear with multiple flashes, then draw and render, within a minute the screen has faded from black and white to a washed out dark gray and light gray. I had to switch displays recently (both still ED060SC4), so I'm not sure if this is just a screen issue or not yet. I'll test a new one when it comes in a couple weeks. I've tweaked VCOM up and down between -1V and -1.6V (there's no sticker on the flex cable for the correct value) but the behavior is the same.

I've tried to roll back a lot of my changes to see if it's something I've added in my logic that's causing it, like maybe leaving the display powered instead of turning off power to it after drawing, but I can't seem to isolate the cause. I'll attach comparison pictures to show the "washed out" colouring later today

dot4qu, 2023-02-11:
Received a new display and I've confirmed that the slow fade mentioned at the end of my previous comment is a problem with just this specific display.

Now, by marking the entire buffer as dirty and then redrawing, I can successfully 'refresh' the behavior I detailed in the initial issue post. There is a noticeable pulse of all of the black pixels when I do this, so I'm using it more sparingly than every minute update.

So now I still have the same issue I created this GitHub issue about, but just around the time numbers when they update every minute. Every other part of the screen that I update individually, I do the full dirty screen refresh. I'm going to investigate optimising this further by marking just the bounding box around the time as dirty every minute. I think that should result in the black pixel 'pulse' that I'm mentioning just around the time text, which is much less noticeable than doing the whole screen every minute.

I'm still curious as to why you mentioned that I would need the vendor waveforms to do a GC16 refresh @vroland  - is it because having the specific waveform for this display would reduce the black pixel pulse that I'm seeing, or something else? I can provide a video of this 'pulse' I'm describing if it would clarify what I'm talking about.

closed 2024-01-04 by second-string