vroland/epdiy #206: How to refresh full screen without clean the buffer?

lanistor, 2022-11-24:
How to refresh full screen without clean the buffer?
`epd_clear` seems will clean the buffer data, then no way to render again.

vroland, 2023-01-01:
Hm, I suppose the highlevel API currently does not allow this, as it looks for parts that have changes first. Maybe an additional function like `epd_hl_redraw` would be nice that just re-daws everything. I suppose that would also help with #205 ...

lanistor, 2023-01-03:
Thanks for reply.

martinberlin, 2023-08-25:
Hello @lanistor 
What is the status of this?
You made #217 pull-request but later was closed. Shall we add this to #205 as **vroland** suggested?

lanistor, 2023-08-26:
I will create a new PR after tested v7, now i close this issues.

closed 2023-08-26 by lanistor