vroland/epdiy #207: Epdiy S3 board discussion

mcer12, 2022-12-06:
Since @vroland resumed work on the S3 board, let's throw ideas and notes about the schematic first, then the pcb layout and features. Don't discuss this on slack because the conversation will get lost!

Yardie-, 2022-12-06:
Here are some of my coments from todays discusson on Slack
Regarding Why we still have the IO expander in the design.

pins 35 36 and 37 cant be used on the R8 Octal PSRAM S3 also to get the voltage over the ADC you need a pin. so with 16 data lines there is not much left
5:10
You could add an ADC expander or an I2C voltage gauge which would allow it though.
5:11
I have built a board with them all connected but it is only 8 data lines
5:12
Also I mucked up and it will only work with an R2 Quad PSRAM S3 (ie I used the naughty pins)
5:13
I'm just waiting for some S3R2 modules to arrive so I can test 

I like the idea of doing away with the port expander however it becomes a choice of 
8bit screens or 2MB PSRAM

If you only used a Quad PSRAM S3 (2MB) you could squeeze a 16 bit on though. You would just have to do everything else over Serial or I2C

mickeprag, 2022-12-06:
One idea is to make two boards. For 8 bit displays we can skip the portexpander and another board for 16 bit displays with a portexpander.

Yardie-, 2022-12-06:
With the S3 you can also skip the Serial to USB. 
I have built many boards that program seemlessly directly over USB via the CDC interface. With no specific setup changes and no need for the Vulcan grip of death either.
However I can see the point of keeping it for as Valetin said the other day "having the external uart could be valuable, if you want to connect a keyboard or something"

mcer12, 2022-12-06:
My notes about the schematic:
1. USB to Serial - use CH340C, simple to use, nice (but will need a second regulator to switch power)
2. MCP1700 - even though it works, it's not a suitable regulator for this purpose. RT9080 is pretty much perfect with great transient response and should deal with transmission spikes hopefully just with the small ceramic capacitors.
3.  TP4057 instead of TP4056 to save some space
4. add 34pin and 33pin connectors for 6" XC6/XD4 displays and 9.7" displays respectively

mickeprag, 2022-12-06:
A note on 1, the CH340C can work without a secondary regulator.

mcer12, 2022-12-06:
@mickeprag it can but it will be 5V logic which may or may not work. Maybe it will but as opposed to ESP8266 I don't think ESP32 is 5V tolerant. Datasheet says absolute maximum 3.6V and who am I to question espressif datasheet :)

martinberlin, 2022-12-08:
Hi there, I've been busy so didn't had time to do much in latest weeks. I'm fighting hard to integrate (And make some money to pay bills) in Barcelona. 
First of all amazing that @vroland is working on V7 that is something great since finally it will be possible to have a new PCB and hopefully also a simpler controller, I mean now that LCD module in S3 will do most of the transmitting parallel data.
One thing I would like to propose Valentin is to add a new **Discussions** area to this repository, that is easily doable just activating a checkbox in settings, and then we could discuss there without creating a new Issue. 

[image]

Not that it bothers me, more on the contrary, I'm happy to collaborate in any discussion. But then having this area we could also **move Issues that are not real bugs or enhancements to that area**, having a clearer focus and less Issues. 

### Onto the discussion now:

- Serial chip:  Even though everybody says for the new S3 you don't need a USB to UART chip, not everybody got it working as is, and for me at least it did not work out all of the time. UART programming is also more than just having Serial to program, it also toggles DTR / RTS that put the MCU in the right mode to program. 
If it would be my opinion to choose I disagree completely in removing UART. 

I use CP2102 that is too big and there are smaller things for the task. Espressif official boards also have UART and an UART chip in place so I see no reason to save space and then have additional problems in the most basic thing that is flashing the Firmware. But that is only my opinion about it.

What it would be very nice to achieve for me is to have 16 datalines support so we can try new models (V5, Inkster & V6 all give support to all existing models)
My desire for this project will be always to innovate and support whenever possible new models so we don't need to buy expensive and complicated controllers, we can just keep on using EPDiy as a rock solid project for all parallel projects.
But I agree would be nice to have a PCB that offers also 33 and 34 pin connectors plus a 40 pin FPC connector. And everyone decides what will be soldered where. But of course that is quite a challenge to make in quite a reduced space.
I see also Micke point of having an IO expander if after 16 data-lines plus other control-lines there is just a few available GPIOs to do something custom.  Maybe what it could be done also is to add a Groove connector (4 pin) with I2C. So everyone can expand the PCB with anything I2C, like a port expander, or any I2C sensor / Touch whatever that is driven by this protocol.
I hope Valentin soon will share the Schematics in SLACK or here and everyone can see better where is V7 board going.

Yardie-, 2022-12-08:
I only use CDC to program my S3 and C3 units and only have a problem when there is a bad solder joint. I do use esp-idf on linux though. Itcould be different for different OS and tool-chains though.

One thing I do do and maybe it is worth adding to the design is

-  add 0 ohmn resistor in line on the D+ and D- lines this is in the S3 design guidelines so maybe that helps. They do explain it in the guide.

mcer12, 2022-12-09:
@Yardie- can you share the document?

vdp, 2022-12-09:
Does anyone have experience and/or opinion on using MAX17135 vs TPS651851? Glancing through the datasheet it seems roughly analogous to TI's IC, but slightly simplified. Could be easier to solder, because although it's still 6mm package it has only 32 pins and the pitch is the same as for TPS65185RGZ(the 7 mm package) and, from experience, for novice solderers that could make a lot of difference. In my experience TPS651851 is also rather fragile and easy to damage when soldering.

mcer12, 2022-12-09:
The issue here, first and foremost is the availability. TPS65185 is widely available, sometimes expensive yes, but available. MAX17135 not so much, especially on lcsc which most of us use for pcbs and assembly. Novice solderer shouldn't solder QFN package but start with something simple :D

mcer12, 2022-12-09:
@Yardie- I have to say I have an S3 devkit at home with native USB broken out but I didn't test it yet :D I'll try to get to it over the weekend, see how difficult it is to work with it and if it's easy to set up, we can make some kind of easy guide and ditch the CH340, I'm all for it. For development tho, having the external chip for serial communication with autoreset is awesome. If it's more than just pushing a button to put it in flash mode, I'm not going to like it very much :D

Yardie-, 2022-12-09:
Just heading out for the weekend but I will find the section about the 0ohmn resistors.
CDC programing as I have tried to explain needs no buttons or special configuration . 
It just works.
Resets into flash mode and uploads the code then restarts the S3 or C3.
Many of my C3 PCBs had no buttons at all. They like the S3 just program and spit any printf statements out over USB.
You don't include serial or anything.

The S3 doesn't even have the glitch of not restarting when you upload like that.

Anyway have a great weekend everyone I'm off.

Yardie-, 2022-12-11:
Ok @mcer12  
Here is the document https://www.espressif.com/sites/default/files/documentation/esp32-s3_hardware_design_guidelines_en.pdf
On page 16 it states
```
2.9 USB
ESP32-S3 has a full-speed USB On-The-Go (OTG) peripheral with integrated transceivers. The USB peripheral is
compliant with the USB 2.0 specification. GPIO19 and GPIO20 can be used as D- and D + of USB respectively.
It is recommended to populate zero-ohm series resistors between the mentioned pins and the USB connector.
Also reserve a footprint for a capacitor to ground on each trace.
ESP32-S3 also integrates a USB Serial/JTAG controller that supports USB 2.0 full-speed
```

At one stage they recommended 22pF capacitors on these two lines as well. I did that and it was totally wrong and USB did not work. After removing the capacitors everything works fine so I no longer add these to my designs.

However I also just noticed that these resistors are not included in the DevKit design. So maybe they are not necessary. I will still add them as This is one area I have absolutely no problems with.

I also recall they had issues with USBC to USBC connectors in as much as the Chip will only handle USB2 and the system was trying to use USB3 over USBC. The solution they came up with at the time was to USB a USB2 Hub but I find it easier to just add micro USB connectors to the PCB.

vroland, 2022-12-25:
Sorry for the silence, I was sick for a week and then busy with chistmas... Here's a new version of the schematic though ;)
So far, I've addressed the following points:
 * Added the 33 and 34 pin connectors. Let's see what sticks when it comes to designing the PCB.
 * Swapped the CP2102 for the 340C. After reading the discussion I'm not entirely convinced CDC is as reliable for programming and resetting as the UART version, which is known to work reliably. Also, I'd like to experiment with using a board as a USB host at some point, e.g. reading USB keyboards and the like, so I prefer to have the dedicated USB host free. 
 * Changed the MCP1700 for the RT9080, which seems more available and has slightly better characteristics.
 * RTC crystal has two 12pF caps.
Feedback is welcome :)
[epaper-breakout.pdf](https://github.com/vroland/epdiy/files/10300755/epaper-breakout.pdf)

vroland, 2022-12-25:
@mcer12 You recommended the TP4057 over the TP4056, but the only datasheet I can find is in chinese, so are there any differences aside from the footprint?

mcer12, 2022-12-25:
@vroland lower maximum charge current (600mA instead of 1A) due to smaller package size. TP4054 can be used but has only CHRG led, not DONE led so it just looks off when fully charged nad I don't like that.

Yardie-, 2022-12-27:
@vroland you might want to consider treating pins 35 36 and 37 as NC.
The new 2 series S3 only have Octal PSRAM according to https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf

Yardie-, 2022-12-27:
Yeah I think a warning is a good call as it is not obvious to new users of the S3. And the 2 series will take  while to come online mainstream.

mcer12, 2022-12-27:
@vroland sure no problem, I was under impression there's size constraints since you wanted to remove some connectors, that's why I suggested it. TP4056 is more generic part so it's better usually.

mcer12, 2022-12-30:
@vroland I was checking out the schematic again and I see GPIO0 is not used for anything and could free up some other more useful pin.

Yardie-, 2022-12-31:
GPIO0 is boot on the S3

mcer12, 2022-12-31:
@Yardie- You mean like on all other ESP modules? :) It can be used unless you pull it low during boot, so for driving high impedance inputs it's perfectly fine. Epdiy V5 uses GPIO0 btw.

vroland, 2023-01-01:
The board will be biggern than the V6 board anyway, so I'll just make it wide enough to fit everything comfortably this time :)
@mcer12 Good catch, I should absolutely use GPIO0 for something else. I'll have a look what goes well with the layout, probably some clock or enable signal.

PickaxeHit, 2023-01-02:
there is a question that esp32-s3 doesn't have parallel I2S support. It must use LCD peripheral or Octal SPI to drive the panel.

Yardie-, 2023-01-02:
@PickaxeHit that's what the S3 code @vroland is working on is all about. This is not a simple thing and will take some time. He is also working towards 16 bit parallel displays in the same code. 
It is pretty much the point of this thread to make the current information/discussion available publicly.

Yardie-, 2023-01-02:
Yes @mcer12 I use it regularly fair point to break it out with a note.

martinberlin, 2023-04-15:
I wanted to comment about this point:

> USB to Serial - use CH340C

CH340 has issues with all latest Ubuntu versions. You just cant flash any MCU with this chip unless you know how to fix it.There is a Braille reader associated directly with CH340.

In case that happens and you don't use a braille display this should do the trick:

sudo apt remove brltty

lanistor, 2023-06-01:
@martinberlin @vroland Hi, is S3 board usable now? which branch can i use it?

martinberlin, 2023-06-01:
Hi Lanistor, 
there is a branch called s3_lcd. @vroland has to confirm when it's ready. 
It might have some modifications pending. Please **do not send anything to production** until Valentin gives the green flag

vroland, 2023-06-12:
@lanistor The tracking pr is #251. I think the hardware should be mostly solid now, there is just a lot of software work to do, which is currently constrained by my time...

lanistor, 2023-07-24:
@vroland @martinberlin 
The computer cannot recognize and read the s3 board.
After plugging in the data cable, the green light is on, but the board cannot be recognized by the command `ls /dev/tty.*`.
The driver of CH340 has been installed, and there is no problem with the recognition and build, flash of the v5 board.
Does the s3 board need to install a new driver?

martinberlin, 2023-07-24:
> The driver of CH340 has been installed, and there is no problem with the recognition and flash of the v5 board.
Does the s3 board need to install a new driver?

Not really. Are you sure that you updated your branch before sending this to fabrication?
send a bit more details please and check that the USB to UART wiring is correct

lanistor, 2023-07-25:
> Not really. Are you sure that you updated your branch before sending this to fabrication? send a bit more details please and check that the USB to UART wiring is correct

I am using the design in the code 2 months ago, is there a problem with the previous PCB design?

martinberlin, 2023-07-25:
If you didn't use last update called "v7 hardware fixes" 
https://github.com/vroland/epdiy/pull/251/commits (Scroll down commit 30ed99b4aae2695df2c928ec879f4a50866ec0a7 )

Then yes it's a problem since D- D+ where swapped. Please check because there where also some other minor corrections.
Check in this [image initial prototype](https://twitter.com/martinfasani/status/1678659803034394625/photo/1) by VRoland.

mcer12, 2023-07-25:
@martinberlin I'm sorry to be that guy but untested boards shouldn’t be on github or at least properly labeled as untested ;)

martinberlin, 2023-07-25:
Well this is still in a Pull-Request that is not merged and hence a clear **Work in progress**. I've been always loud and clear to wait until clear confirmation from @vroland that the PCB part is ready to send release 1 to fabrication.

mcer12, 2023-07-25:
@martinberlin Sorry I was under impression it’s already merged…

lanistor, 2023-07-25:
> If you didn't use last update called "v7 hardware fixes" https://github.com/vroland/epdiy/pull/251/commits (Scroll down commit [30ed99b](https://github.com/vroland/epdiy/commit/30ed99b4aae2695df2c928ec879f4a50866ec0a7) )
> 
> Then yes it's a problem since D- D+ where swapped. Please check because there where also some other minor corrections. Check in this [image initial prototype](https://twitter.com/martinfasani/status/1678659803034394625/photo/1) by VRoland. I sent one PCB with this corrections to him by post but it has no ESP32S3 soldered so we need to wait and confirm that everything is working as expected.
> 
> This release was actually a work in progress so we need to be patient to check that most issues are fixed or risk an early production where it can be that things still need minor corrections.

Thanks a lot, i'm making the new boards.

martinberlin, 2023-07-25:
@lanistor please do not trash the ones you already mounted. Maybe some other people can try if you don’t have the time to fix them and you can offer them in the EPDiy Slack group

lanistor, 2023-07-26:
> @lanistor please do not trash the ones you already mounted. Maybe some other people can try if you don’t have the time to fix them and you can offer them in the EPDiy Slack group

I can't use my computer to connect to the boards I made last time, so I don't know how to fix them.

martinberlin, 2023-07-26:
Just try to fix it inverting the USB to UART cables. Make sure also to check that GBR files and drill files are in the latest stand before sending it to fabrication. In any case just open the V7 in KiCad and regenerate them just in case.
Please always check that you are in the right branch and in the latest stand before sending anything to production.

lanistor, 2023-07-26:
> Just try to fix it inverting the USB to UART cables. Make sure also to check that GBR files and drill files are in the latest stand before sending it to fabrication. In any case just open the V7 in [...]

Sorry, I'm not a PCB designer and don't know how to use KiCad. I sees the [latest commit](https://github.com/vroland/epdiy/commit/30ed99b4aae2695df2c928ec879f4a50866ec0a7) has changes of gerbers, so the gerbers should  be updated already?

martinberlin, 2023-07-26:
@vroland I just opened latest **s3_lcd** v7 hardware and I think that Drill files are not updated. At least opening this in KiCad settings I see some difference:

	modified:   epaper-breakout-NPTH.drl
	modified:   epaper-breakout-PTH.drl

[image]

But I'm not 100% sure since I'm using version 7 and this was done with v6 of KiCad so maybe is just that.
If this is like this can you please update it so people trying to send this to fabrication do not miss latest updates?

> Sorry, I'm not a PCB designer and don't know how to use KiCad. 

Sorry @lanistor but this is open source hardware and you are risking here sending to fabrication something that is still not merged. I'm also not a PCB designer but I forced myself to learn KiCad and at least understand how this is exported to Gerbers.
I highly recommend you to follow the same path. Sending something to fabrication without this knowledge has a high risk of failure, since even if you trust the source, this is open source and we all can make mistakes or forget to update something, etc. In the top of that maybe you want to add a custom message or URL in the Silkscreen or any indication so I highly recommend to learn a bit how is the process.

lanistor, 2023-08-05:
> @vroland I just opened latest **s3_lcd** v7 hardware and I think that Drill files are not updated. At least opening this in KiCad settings I see some difference:
>
> ```
> modified: [...]

Thanks, i create new gerbers and make new boards.

But i got an error when start the board, here is the error log:
```
E (302) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (306) cpu_start: Failed to init external RAM!
```

martinberlin, 2023-08-05:
Hi @lanistor can you please post a close up photo of the MCU?
Sounds strange this error, did you configured PSRAM using menuconfig? Please check the [epdiy wiki page](https://github.com/vroland/epdiy/wiki#epdiy-v7) where I'm starting to document stuff and hopefully will soon migrate the docs in readthedocs.

Make sure you enable PSRAM (Octal for this board)

Component config --> ESP PSRAM --> Enable SPI-connected RAM --> Config --> QUAD/OCT

**Additionally:** Set RAM clock to max. speed (80MHz clock speed)

martinberlin, 2023-08-05:
Hello @vroland 
compiling last **s3_lcd** I'm getting this error:

    src/epd_driver/display_ops.c:61: undefined reference to `i2s_deinit'

If I see this correctly that file is missing an include: #include "render_method.h"  which defines RENDER_METHOD_LCD and then this i2s_denit is excluded.
I will be testing different things on each update to make sure there are no errors building different examples. Will need to correct also my www-image download jpg demos since after IDF v 5 there are a lot of mistakes if you don't cast (int) in the Logging functions (It says to include <inttypes.h> in console to fix it, but it really does not)

lanistor, 2023-08-07:
@martinberlin Yeah, i did the settings according to the wiki, here is the full log and the board picture:
[image]

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x187c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3130
entry 0x403c9934
I (31) boot: ESP-IDF v5.1-476-g3187b8b326 2nd stage bootloader
I (31) boot: compile time Aug  7 2023 10:33:30
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v0.1
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32s3: Boot SPI Speed : 80MHz
I (50) boot.esp32s3: SPI Mode       : QIO
I (54) boot.esp32s3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00177000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=d59d0h (874960) map
I (243) esp_image: segment 1: paddr=000e59f8 vaddr=3fc93900 size=02858h ( 10328) load
I (245) esp_image: segment 2: paddr=000e8258 vaddr=40374000 size=07dc0h ( 32192) load
I (255) esp_image: segment 3: paddr=000f0020 vaddr=42000020 size=21364h (136036) map
I (278) esp_image: segment 4: paddr=0011138c vaddr=4037bdc0 size=07ac4h ( 31428) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (302) cpu_start: Multicore app
E (302) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (306) cpu_start: Failed to init external RAM!

abort() was called at PC 0x4037564c on core 0
0x4037564c: call_start_cpu0 at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/cpu_start.c:466 (discriminator 3)

Backtrace: 0x40375a9a:0x3fceb220 0x4037bfa5:0x3fceb240 0x403818b2:0x3fceb260 0x4037564c:0x3fceb2d0 0x403cda98:0x3fceb340 0x403cde91:0x3fceb380 0x403c999d:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375a9a: panic_abort at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/panic.c:452

0x4037bfa5: esp_system_abort at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x403818b2: abort at /Users/Shared/B/SDK/esp/esp-idf/components/newlib/abort.c:38

0x4037564c: call_start_cpu0 at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/cpu_start.c:466 (discriminator 3)

0x40045c01: ets_run_flash_bootloader in ROM
```

huming2207, 2023-08-07:
> E (302) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode

What's your PSRAM config? That looks like PSRAM is dead, not present or misconfigured?

lanistor, 2023-08-07:
> > E (302) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
> 
> What's your PSRAM config? That looks like PSRAM is dead, not present or misconfigured?

Here is my config:
[image]

martinberlin, 2023-08-07:
Wrong!
Is **OCTAL Mode PSRAM**, not Quad mode, I wrote it in the WiKi but also here @lanistor 
You need to really focus and be aware on the details here when you read an explanation on how to configure this.
If it's not that then it must be the S3 module that is different please post exactly what part you got soldered there. Here a side to side comparison
[image]

lanistor, 2023-08-07:
> Wrong! Is **OCTAL Mode PSRAM**, not Quad mode, I wrote it in the WiKi but also here @lanistor You need to really focus and be aware on the details here when you read an explanation on how to [...]

Thanks, i fix this error by set `OCTAL Mode`, but i got a new error `EPD_DRAW_LOOKUP_NOT_IMPLEMENTED`, here is the  log:
```
I (31) boot: ESP-IDF v5.1-476-g3187b8b326 2nd stage bootloader
I (31) boot: compile time Aug  7 2023 10:33:30
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v0.1
I (39) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32s3: Boot SPI Speed : 80MHz
I (50) boot.esp32s3: SPI Mode       : QIO
I (54) boot.esp32s3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00177000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=d5a30h (875056) map
I (243) esp_image: segment 1: paddr=000e5a58 vaddr=3fc94300 size=02c08h ( 11272) load
I (245) esp_image: segment 2: paddr=000e8668 vaddr=40374000 size=079b0h ( 31152) load
I (255) esp_image: segment 3: paddr=000f0020 vaddr=42000020 size=21360h (136032) map
I (278) esp_image: segment 4: paddr=00111388 vaddr=4037b9b0 size=08858h ( 34904) load
I (291) boot: Loaded app from partition at offset 0x10000
I (292) boot: Disabling RNG early entropy source...
I (303) cpu_start: Multicore app
I (303) octal_psram: vendor id    : 0x0d (AP)
I (304) octal_psram: dev id       : 0x02 (generation 3)
I (307) octal_psram: density      : 0x03 (64 Mbit)
I (312) octal_psram: good-die     : 0x01 (Pass)
I (317) octal_psram: Latency      : 0x01 (Fixed)
I (323) octal_psram: VCC          : 0x01 (3V)
I (328) octal_psram: SRF          : 0x01 (Fast Refresh)
I (334) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (339) octal_psram: BurstLen     : 0x01 (32 Byte)
I (345) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (351) octal_psram: DriveStrength: 0x00 (1/1)
I (357) MSPI Timing: PSRAM timing tuning index: 5
I (362) esp_psram: Found 8MB PSRAM device
I (366) esp_psram: Speed: 80MHz
I (370) cpu_start: Pro cpu up.
I (374) cpu_start: Starting app cpu, entry point is 0x4037551c
0x4037551c: call_start_cpu1 at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (793) esp_psram: SPI SRAM memory test OK
I (802) cpu_start: Pro cpu start user code
I (802) cpu_start: cpu freq: 240000000 Hz
I (802) cpu_start: Application information:
I (805) cpu_start: Project name:     firmware
I (810) cpu_start: App version:      1.0.1-78-g30ed99b-dirty
I (817) cpu_start: Compile time:     Aug  7 2023 14:46:37
I (823) cpu_start: ELF file SHA256:  885204d567489924...
I (829) cpu_start: ESP-IDF:          v5.1-476-g3187b8b326
I (835) cpu_start: Min chip rev:     v0.0
I (840) cpu_start: Max chip rev:     v0.99
I (844) cpu_start: Chip rev:         v0.1
I (849) heap_init: Initializing. RAM available for dynamic allocation:
I (856) heap_init: At 3FC9A730 len 0004EFE0 (315 KiB): DRAM
I (863) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (869) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (876) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (883) spi_flash: detected chip: gd
I (887) spi_flash: flash io: qio
W (891) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (904) sleep: Configure to isolate all GPIO pins in sleep state
I (911) sleep: Enable automatic switching of GPIO sleep configuration
I (918) app_start: Starting scheduler on CPU0
I (923) app_start: Starting scheduler on CPU1
[... 42 lines cut ...]
0x4037c911: esp_system_abort at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40382315: __assert_func at /Users/Shared/B/SDK/esp/esp-idf/components/newlib/assert.c:81

0x40377a74: lcd_feed_frame at /Users/lanistor/C/hardware/epdiy/src/epd_driver/render_lcd.c:152 (discriminator 9)

0x403777b2: feed_display at /Users/lanistor/C/hardware/epdiy/src/epd_driver/render.c:251 (discriminator 1)

ELF file SHA256: 885204d567489924

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420127c3
0x420127c3: __sfvwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fvwrite.c:221

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x187c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3130
entry 0x403c9934
```

martinberlin, 2023-08-07:
This you need to check how to solve I ask the same up there. Stay tuned and do your own research, try to understand what **vroland** did to make this work in S3. Please collaborate and try to help us expand the WiKi with your tests.
Try a 16 bit display also in the 40 FPC connector.

vroland, 2023-08-07:
Hi @martinberlin , 
Sorry, the v7 driver is unfortunately still in a rough state, I hope to be able to clean it up a bit next week. You're right about the includes, my plan is to conditionally compile the render method based on the chip used, but make the display config dynamic.
You get the queue error if the line / waveform computation can't keep up with sending data to the LCD. This will be reworked into a nicer error hopefully. But at the speed you have it set to, It's likely a problem with either SPIRAM bandwidth (as @lanistor figured out) or CPU core frequency.

About that last assertion: Currently, the S3 driver code always draws full frames and doesn't skip empty lines. I'm wondering why you hit that assertion though, I think there is
```c
  diff_area.x = 0;
  diff_area.y = 0;
  diff_area.width = EPD_WIDTH;
  diff_area.height = EPD_HEIGHT;
```
in highlevel.c in the s3 branch... Also, not all buffer modes are supported right now, e.g., 2PPB packing and monochrome mode. So examples that only use the highlevel API should work, but the ones that use 4bit-per-pixel buffers don't right now. As I said, the driver is still work in progress. Sorry about that, there is such much else going on and the driver work needs a block of focus time, it's not something I can just work on effectively for 30mins inbetween other things :/

martinberlin, 2023-08-21:
@lanistor could you try the example/dragon. It is working for me correctly with a V7 board. 
Check latest updates in **s3_lcd** there is a way to set board and display dynamically now.

martinberlin, 2023-08-22:
Thanks @vroland . Question: 

Every time I use **epdiy** as an idf component in a project I face the challenge of adding manually a CMakeFiles. Could we add an additional one like this in the root of the project so it's easier for everybody:

```
set(basedir "src/epdiy")
set(app_sources "${basedir}/epdiy.c"
                "${basedir}/render.c"
                "${basedir}/output_lcd/render_lcd.c"
                "${basedir}/output_lcd/lcd_driver.c"
                "${basedir}/output_i2s/render_i2s.c"
                "${basedir}/output_i2s/rmt_pulse.c"
                "${basedir}/output_i2s/i2s_data_bus.c"
                "${basedir}/output_common/lut.c"
                "${basedir}/output_common/line_queue.c"
                "${basedir}/output_common/render_context.c"
                "${basedir}/font.c"
                "${basedir}/displays.c"
                "${basedir}/board_specific.c"
                "${basedir}/builtin_waveforms.c"
                "${basedir}/highlevel.c"
                "${basedir}/board/tps65185.c"
                "${basedir}/board/pca9555.c"
                "${basedir}/board/epd_board.c"
                "${basedir}/board/epd_board_common.c"
                "${basedir}/board/epd_board_lilygo_t5_47.c"
                "${basedir}/board/epd_board_v2_v3.c"
                "${basedir}/board/epd_board_v4.c"
                "${basedir}/board/epd_board_v5.c"
                "${basedir}/board/epd_board_v6.c"
                "${basedir}/board/epd_board_v7.c"
)

# Can also use IDF_VER for the full esp-idf version string but that is harder to parse. i.e. v4.1.1, v5.0-beta1, etc
if (${IDF_VERSION_MAJOR} GREATER 4)
    idf_component_register(SRCS ${app_sources} INCLUDE_DIRS "${basedir}/include" REQUIRES driver esp_timer esp_adc esp_lcd)
else()
    idf_component_register(SRCS ${app_sources} INCLUDE_DIRS "${basedir}/include" REQUIRES esp_adc_cal esp_timer)
endif()

# Does not seem to exist anymore lut.c in new version?
#set_source_files_properties("${basedir}/lut.c" PROPERTIES COMPILE_OPTIONS -mno-fix-esp32-psram-cache-issue)
```

Just a suggestion. Also this lut.c if I'm looking right does not exist anymore in latest version that is still not merged. 
This will make in my humble opinion this component much easier to add in future projects.

lanistor, 2023-08-25:
@vroland Is `EPD_DRAW_EMPTY_LINE_QUEUE` an error? I got the below message while running latest code:
```
W (27381) epd_lcd: draw frame draw initiated, but an error flag is set: 400
```

Here is the whole message: 
```
I (27) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8 2nd stage bootloader
I (27) boot: compile time Aug 25 2023 15:27:11
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00011000 0000e000
I (73) boot:  1 phy_init         RF data          01 01 0001f000 00001000
I (81) boot:  2 factory          factory app      00 00 00020000 00400000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00020020 vaddr=3c030020 size=f6f30h (1011504) map
I (282) esp_image: segment 1: paddr=00116f58 vaddr=3fc96900 size=02dcch ( 11724) load
I (285) esp_image: segment 2: paddr=00119d2c vaddr=40374000 size=062ech ( 25324) load
I (294) esp_image: segment 3: paddr=00120020 vaddr=42000020 size=23f10h (147216) map
I (323) esp_image: segment 4: paddr=00143f38 vaddr=4037a2ec size=0c5ach ( 50604) load
I (334) esp_image: segment 5: paddr=001504ec vaddr=600fe000 size=00070h (   112) load
I (342) boot: Loaded app from partition at offset 0x20000
I (342) boot: Disabling RNG early entropy source...
I (354) cpu_start: Multicore app
I (355) octal_psram: vendor id    : 0x0d (AP)
I (355) octal_psram: dev id       : 0x02 (generation 3)
I (358) octal_psram: density      : 0x03 (64 Mbit)
I (364) octal_psram: good-die     : 0x01 (Pass)
I (369) octal_psram: Latency      : 0x01 (Fixed)
I (374) octal_psram: VCC          : 0x01 (3V)
I (379) octal_psram: SRF          : 0x01 (Fast Refresh)
I (385) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (391) octal_psram: BurstLen     : 0x01 (32 Byte)
I (396) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (403) octal_psram: DriveStrength: 0x00 (1/1)
I (408) MSPI Timing: PSRAM timing tuning index: 11
I (413) esp_psram: Found 8MB PSRAM device
I (418) esp_psram: Speed: 80MHz
I (422) cpu_start: Pro cpu up.
I (425) cpu_start: Starting app cpu, entry point is 0x4037573c
0x4037573c: call_start_cpu1 at /Users/Shared/B/SDK/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (884) esp_psram: SPI SRAM memory test OK
I (893) cpu_start: Pro cpu start user code
I (893) cpu_start: cpu freq: 240000000 Hz
I (893) cpu_start: Application information:
I (896) cpu_start: Project name:     firmware
I (901) cpu_start: App version:      1.0.1-89-g9f3f19a-dirty
I (908) cpu_start: Compile time:     Aug 25 2023 15:27:38
I (914) cpu_start: ELF file SHA256:  d3231b7370a6d78d...
I (920) cpu_start: ESP-IDF:          v5.1.1-1-gd3c99ed3b8
I (926) cpu_start: Min chip rev:     v0.0
I (930) cpu_start: Max chip rev:     v0.99
I (935) cpu_start: Chip rev:         v0.1
I (940) heap_init: Initializing. RAM available for dynamic allocation:
I (947) heap_init: At 3FC9CC40 len 0004CAD0 (306 KiB): DRAM
I (953) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (960) heap_init: At 600FE070 len 00001F78 (7 KiB): RTCRAM
I (967) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (975) spi_flash: detected chip: gd
I (978) spi_flash: flash io: dio
I (982) sleep: Configure to isolate all GPIO pins in sleep state
I (989) sleep: Enable automatic switching of GPIO sleep configuration
[... 101 lines cut ...]
I (13711) epdiy: highlevel diff area: x: 219, y: 484, w: 591, h: 109
I (13711) epdiy: starting update, phases: 30
actual draw took 346ms.
PG is up
I (15241) epdiy: calculating diff..
I (15441) epdiy: highlevel diff area: x: 15, y: 15, w: 795, h: 753
I (15441) epdiy: starting update, phases: 30
W (15461) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (15461) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (15471) epd_lcd: draw frame draw initiated, but an error flag is set: 400
......
W (15871) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (15871) epd_lcd: draw frame draw initiated, but an error flag is set: 400
actual draw took 446ms.
E (15961) demo: draw error: 400
PG is up
I (21141) epdiy: calculating diff..
I (21341) epdiy: highlevel diff area: x: 266, y: 25, w: 492, h: 743
I (21341) epdiy: starting update, phases: 30
W (21351) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (21351) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (21361) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (21361) epd_lcd: draw frame draw initiated, but an error flag is set: 400
W (213
```

martinberlin, 2023-08-25:
Just put a lower bus_speed in display.c this is because at that speed the LCD cannot keep it at so fast.
On 097OC4 at 8 bit I use 15 MHz, while with others can work faster, like 097TC2 was tested by Valentin at 22 MHz.

lanistor, 2023-08-25:
> Just put a lower bus_speed in display.c this is because at that speed the LCD cannot cope with the sending so fast. On 097OC4 at 8 bit I use 15 MHz, while with others can work faster, like 097TC2 was tested by Valentin at 22 MHz.

This works 👍.

But the rendering clarity or rendering speed, seems not so good as v5 board.

martinberlin, 2023-08-25:
> But the rendering clarity or rendering speed, seems not so good as v5 board.

I do not agree with this. Both the speed and rendering quality seem good to me. If you would like to contribute please read to the detail the [LCD module](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html)
It's easy to complain about something without giving any facts like measured statistics or video proof of what you experiment. This is also open source and everyone using it should realize that is a collective effort to make it happen.
Closing here:
There is a double Issue with S3 let's please keep this discussion at #203

[7 more replies]

closed 2023-08-25 by martinberlin