vroland/epdiy #211: LilyGo T4.7 software port to V5 on Arduino ?

gfaman, 2023-01-12:
Hi,
I modified and added functions on an application with the board and display from LilyGo T4.7 (without touch screen). The original software was the OWM one in the LilyGo and on Arduino.
I'm trying now to port this application to the V5 board, using ED097TC2 display. 
I removed the libraries from LilyGo, replace by the epdiy master, add the V5 board in the boards list but now, I'm struggling with many specific functions seems developed by Lilygo.
Anyone tried already successfully to transfer an LilyGo application (with their library) to a V5 board (with epdiymaster library) ?
Thanks,
Laurent

martinberlin, 2023-01-14:
Reopening after discussing this with @gfaman and inviting him to Slack. Please add details of what are the functions that are missing or give errors when you switch to EPDiy. 
But in the future please use only EPDiy to start projects and the newest examples from this repository. Starting with Lilygo repository that uses epdiy v.2 is a bad starting step because you will lack the newest updates that are done here.

gfaman, 2023-01-16:
Thanks Martin. Understand the point to start directly with EPDiy. However, I had first the HW of Lilygo and their SW demo was fully compatible with Arduino. That's the original reason.
So far, the transfer from LilyGo to EPDiy is still on going (not yet finished neither working). If someone wants to do it, some points I can highlight:
Add:
#include "epd_highlevel.h"
#define WAVEFORM EPD_BUILTIN_WAVEFORM
EpdFont  currentFont;
EpdFontProperties font_props;
_swap_int function

Replace:
- epd_poweroff_all  by epd_poweroff
- Rect_t by EpdRect
- get_text_bounds by epd_get_text_bounds
- FontProperties by EpdFontProperties
- write_string by epd_write_string  and add a parameter &font_props
- GFXfont  by EpdFont
- in all .h icon file, change some type variables with the epd one
- GFXglyph by EpdGlyph

Missing functions:
- epd_write_line declaration. Moreover, it seems this function has 2 different definitions, one in epd_driver.c and one in font.c

Change:
- drawRect, fillRect (type parameters are differents)

Can compile but not yet link...

Laurent

melmager, 2023-02-04:
just try to compile esphome 
since yesterday i got compile error:

Compiling .pioenvs/lilygo/lib56b/epdiy/board_specific.c.o
.piolibdeps/lilygo/epdiy/src/epd_driver/board/epd_board_v6.c: In function 'epd_board_init':
.piolibdeps/lilygo/epdiy/src/epd_driver/board/epd_board_v6.c:102:7: error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
       ^
.piolibdeps/lilygo/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
                    ^
.piolibdeps/lilygo/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: note: each undeclared identifier is reported only once for each function it appears in
*** [.pioenvs/lilygo/lib56b/epdiy/board/epd_board_v6.c.o] Error 1

martinberlin, 2023-02-05:
You can remove that line of clk_flags it should work also without it.
What version of ESP-IDF is using the ESP-Home to compile this? (Or if it's using Arduino ESP32 framework, what version is using)
There can be the solution for this. I'm quite sure this appeared after last ESP-IDF version 5 merge in master #180

martinberlin, 2023-03-03:
Closing this. Unless there is common approval I really would prefer to focus testing and working in the new PCB versions.

@mcer12 & @vroland  IMHO:
Lilygo should maintain and keep updated their own repository based on this one and not the other way around

closed 2023-03-03 by martinberlin