vroland/epdiy #213: EPDiy driver doesn't work on the monitor ttgo t5 4.7 plus with microcontroller esp32s3

gianlucarossi15, 2023-01-21:
Hi,
My team and I have encountered a problem with the EPDiy driver on the monitor version ttgo t5 4.7 plus from LilyGo.
 
[image]

Basically the library is not able to detect anymore the GPIO's because they have been changed with the new monitor version ttgo t5 4.7 plus.
What should we do?

martinberlin, 2023-02-05:
You can try my branch here:
https://github.com/martinberlin/epdiy-rotation/tree/lilygos3

This new board is not added to epdiy yet. I had one of this but at the moment I do not find it so someone else should try to add it!

martinberlin, 2023-06-08:
Unless someone adds this and makes a pull request with the new display I’m not really making the effort this time. Simply because I don’t have a Lilygo S3 at hand at the moment and also does not motivate me that Lilygo uses a really old version of this library,
It’s Lilygo team who should fork this and add the new display and not the other way around.
Just as a heads starter take a look at **s3_lcd** branch.