vroland/epdiy #214: Option 8M working for V5 board ?

gfaman, 2023-01-26:
Hi,
I did not succeed to use the 8M option in the menu of the V5 under Arduino.

It should accept 3MB application size but it makes endless reset on the ESP32 which is confirm 064 (8MB size).
To bypass this, I have to create a new partition files with these datas:
 # Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     36K,     20K,
otadata,  data, ota,     56K,     8K,
app0,     app,  ota_0,   64K,     3M,
app1,     app,  ota_1,   ,        3M,
ffat,    data,     fat,  ,        1500k,

Also, in the boards.local.txt, I changed some lines:
epdiy_v5.upload.maximum_size=4026368 (before it stuck sometimes here due to size 2MB hier than 1.3MB)
epdiy_v5.build.flash_size=8MB (not sure it is necessary, it was 4MB before)

After that, partitions.csv + boards.local.txt, works fine with the 1.5MB FFAT and 3MB app with OTA

Is it normal ? 

Laurent

martinberlin, 2023-06-09:
Hi @gfaman 
Did you succeed with this?
Because I don't think this Issue is related to the EPDiy library but more a issue you are having with your own configuration.

Also you apart of the 8 Mb flash you are not really clarifying in this topic what are you trying to build, neither providing a log so we can check more. Please give more feedback before the end of June 2023 or this issue will be automatically closed.

martinberlin, 2023-08-25:
Closing due to lack of feedback in last 2 months

closed 2023-08-25 by martinberlin