vroland/epdiy #215: waveshare 6inch HD e-Paper support

3058132083, 2023-01-28:
https://www.waveshare.net/w/upload/e/e9/6inch_HD_e-Paper_Specification.pdf
There is no mention of the model number in the document, I think it is the same as the ED060KC1

martinberlin, 2023-01-28:
This one should work without any problems. Just get one PCB made or get an Inkster in Tindie and try it out

3058132083, 2023-01-28:
[image]
[image]

The documentation mentions: 25V VGG is required, but in the schematic, it is marked 22V

3058132083, 2023-01-28:
I remade a small size version to use this screen, here is the link, not verified yet
https://oshwhub.com/lxhcat/bing-kou-dian-zi-mo-shui-ping-qu-dong-esp32-xiu-gai-zi-github-kai-yuan-xiang-mu-epdiy
[image]

martinberlin, 2023-01-28:
Looks nice. But I will try with an existing model that works to make sure it will drive that screen.
I don't think the voltage will be a problem since all this parallel epapers use mostly the same voltages just a different VCOM that usually is printed on an adhesive label on the display.

martinberlin, 2023-06-09:
@3058132083 if you will keep researching on this one then we keep it open. 
But if there is no more updates on this till end of June 2023 then this issue will be closed.

closed 2023-08-06 by martinberlin

markbirss, 2025-04-04:
> https://www.waveshare.net/w/upload/e/e9/6inch_HD_e-Paper_Specification.pdf There is no mention of the model number in the document, I think it is the same as the ED060KC1

I uses this display with a small modification 
https://github.com/vroland/epdiy/issues/70#issuecomment-801908988

just tried this again with current tree and v5 board and needed update (idf 5.4)

nano ../../src/displays.c
const EpdDisplay_t ED097TC2 = {
    .width = 1456,
    .height = 1072,
    .bus_width = 8,
    .bus_speed = 22,
    .default_waveform = &epdiy_ED097TC2,
    .display_type = DISPLAY_TYPE_ED097TC2,
};

and for the example you want to compile set #define DEMO_BOARD epd_board_v5 in my case (since im still having v5 board)

nano main/main.c
#ifdef CONFIG_IDF_TARGET_ESP32
#define DEMO_BOARD epd_board_v5