vroland/epdiy #219: I2C error issue on V6 boards when using arduino

deskl0, 2023-02-26:
hi, on V6 board, with utilizing of GPIO32, 33 as I2C pins and esp32-arduino 2.0.7, I have got a constant I2C communication error. Which is 
<code>E(658) PCA9555: pca9555_read_input failed
E (1224) PCA9555: pca9555_read_input failed
E (1790) PCA9555: pca9555_read_input failed
E (2357) PCA9555: pca9555_read_input failed
E (2923) PCA9555: pca9555_read_input failed
E (3490) PCA9555: pca9555_read_input failed
</code> 

Hardware: ESP32-WROVER-E
ENV: Visual Micro+arduino1.8/Arduino 1.8 IDE/Arduino2.0 IDE (all of them have the same issue mentioned above), windows11 Pro
PROGRAM: Dragon-Demo(given as an example in the repository)

When I use I2C scanner on GPIO32, 33 as I2C ports, the program freezes if I use<code>Wire1.begin();</code>, but I2C scanner is able to run flawlessly in using <code>Wire1.begin(32,33,400000);</code> and the scanner found everything attached to I2C port on GPIO32, 33 correctly. 
I have tried esp32-arduino 2.0.1-2.0.7, no changes. For esp32-arduino 2.0.0 and below, the compiler reports missing esp32-hal-i2c-slave.h
Since I know little about esp32 IDF and this repo isn't using <code>wire</code>library provided by arduino, I'm not sure what's going on there, any clues?

deskl0, 2023-03-10:
hi

martinberlin, 2023-03-10:
No idea. But I could not reproduce this. On ESP-IDF does not happen, I don't have one V6 now, but it worked out correctly using this IOs and could communicate with TPS65135  and the IO expander PCA9555 correctly.  Check the connections to PCA either there is something wrong and not correctly connected, or PCA is dead, no idea. But for some reason is not responding.

deskl0, 2023-03-10:
> No idea. But I could not reproduce this. On ESP-IDF does not happen, I don't have one V6 now, but it worked out correctly using this IOs and could communicate with TPS65135 and the IO expander [...]

Thanks for your reply! I have everything checked immediately once the issue occurred, unfortunately, connections are fine. And if you use<code>Wire1</code> instead of <code>Wire</code> in an ordinary I2C scanner on esp32-wrover module, the issue occurs no matter what I2C slave devices attached. The fix could be <code>Wire1.begin(32,33);</code>, so I wonder it might be something wrong with io mux config error. Maybe there is somewhere in the repository that I can initialise I2C port in advance, before making I2C transfers?

vroland, 2023-03-11:
@deskl0 Hi, I got your email, I'm just a bit slow to catch up with issues at the moment, as my dayjob is taking a lot of time :/
Did you try with the ESP-IDF as well or just with the Arduino IDE? I don't think the libary was tested yet with arduino-esp32 2.0, only 1.x.

deskl0, 2023-03-12:
> @deskl0 Hi, I got your email, I'm just a bit slow to catch up with issues at the moment, as my dayjob is taking a lot of time :/ Did you try with the ESP-IDF as well or just with the Arduino IDE? I don't think the libary was tested yet with arduino-esp32 2.0, only 1.x.

Thanks for your reply! I'm just use the arduino IDE, for arduino above 2.0.1-2.0.7 the issue is mentioned above. And for below 2.0.1, the compiler reports missing <code>esp32-hal-i2c-slave.h</code>, so I'd like to know which version of arduino when you have the version6 tested? As for ESP-IDF, not have had a try yet, due to busy daily work recently, I will have a try when I can take a break.

martinberlin, 2023-06-09:
Hi @deskl0 could you find some time to try ESP-IDF?
I highly recommend it. It's a very nice way to go to low level C and get to know much better the Espressif chips.
Please give some feedback about this if you find some time.

martinberlin, 2023-08-07:
Is there anyone else here that has a V6 board and can try this with Arduino to help @deskl0 ?

deskl0, 2023-09-19:
Sorry for being late, I'm quite busy with my research job. Unfortunately, nothing changed since the last comment, however, I managed to make it work in an eccentric way. I found the board could be functional when the esp32 module has been heated to ~200degC, it malfunctioned immediately once the temperature dropped. I was confused, totally, beside that, nothing changed. The library still not working...

martinberlin, 2023-09-19:
I do not have at the moment any V6 board to test this.

> esp32 module has been heated to ~200degC

Would not really recommend to heat the PCB like that. This could make other things go wrong and then you won't be able to test this anymore.
That is very close to the soldierijg temperature and at the edge of some components resistance.

deskl0, 2023-09-20:
Exactly, esp32 should not be used in such extreme circumstances. Quite strange, I guess it is some sort of impedance issue. Anyway, I may fallback to previous versions if I failed to locate this bug. Thanks for your reply :P

martinberlin, 2023-10-04:
I tested this with a v7 board, that has a similar I2C port extender: 
https://github.com/vroland/epdiy/tree/s3_lcd/hardware/epdiy-v7 PCA9535PW (U2)

Added the branch **s3_lcd** of epdiy as an Arduino library. 
And do not get this error at all in Arduino.  So IMHO this is a fault in your PCB but not an issue of this repository, otherwise at least one more person using v6, should have the same problem.

deskl0, 2023-10-05:
There is no progress from my side, it seems to be some i2c1-related registers failed to reset themselves in the starting sequence. Pure Arduino environment gets no such error, maybe Arduino drivers initialise them manually as a precaution? I've consulted with one of my friends who is an IC engineer, he suspects there may exist some minor hardware design defects in iomux or something interconnection-related self-reset circuits, which led to this issue. Anyway, I managed to bypass this by simply generating the required voltages from separated LDOs. Meanwhile, by the publishing and testing of epdiy-v7, there should be no further similar issues, this issue lost the purpose it should have, so I'm closing it after this comment. And, I DO appreciate your reply and support, thanks a lot!

closed 2023-10-05 by deskl0