vroland/epdiy #220: GPIO12 unusable

Fabian-Schmidt, 2023-03-01:
Hi,
I am using GPIO pin 12 as an UART pin and the library disables it every time on powerdown of the display.

https://github.com/vroland/epdiy/blob/12c67cead7a76e90f692fa2c3afcda87831badd0/src/epd_driver/i2s_data_bus.c#L168

Can this behavior be changed? I understand the power savings desire. Is it possible to add at least a `#if` directive to disable this behaviour?

Thanks,
Fabian

mickeprag, 2023-03-01:
Is this a custom board or the official?
For a custom board I would suggest you add your own board implementation in your project. Copy from epdiy and change whatever you need.

Fabian-Schmidt, 2023-03-01:
It is a Lilygo T5-4.7 board.

Fabian-Schmidt, 2023-03-02:
I am using a Lilygo T5-4.7 Board which is supported by the library and therefore I would like to avoid forking the library. I did fork it to validate that removing this line will fix my issue, but I would rater like to keep it together in your branch.

The problem is that the Lilygo T5 device has so few ports (4) and I need unfortunately all of them.

mickeprag, 2023-03-02:
You do not need to fork the library. It is possible to supply your own board implementation and still use the mainline library.

martinberlin, 2023-03-03:
@Fabian-Schmidt this IO is not used in any PCB version.
You can just make a merge request removing this line and we can review it testing in before merging it. IMHO don't think that this will affect consumption at all.
Only note that I'm aware of in the datasheet about this IO is this one:

> GPIO12 is internally pulled high in the module and is not recommended for use as a touch pin.

So keep that in mind in your design, if it's pulled-up high, then reading that IO should return a 1 by default. Give it a try.

martinberlin, 2023-10-07:
On hold till https://github.com/vroland/epdiy/pull/251 is merged

martinberlin, 2023-11-19:
Hello Fabian, can you make the pull-request but from the current **main** branch please?
Thanks!

martinberlin, 2023-12-30:
Closing here since the request was merged

closed 2023-12-30 by martinberlin