vroland/epdiy #221: LilyGo T5 screen clears with lines

skorokithakis, 2023-03-02:
I tried the LilyGo T5 PlatformIO example, which works very well and flashes easily, but doesn't display properly. When the display tries to draw something after clearing, there are random vertical lines that run through the length of the display (like a barcode), leading to ghosting.

Has anyone else seen this? The official library works fine.

martinberlin, 2023-03-03:
I had this on the T5 S3 version and it's a hardware issue on the screen.
It's just a very bad quality cheap screen. The older version was better. So I guess there is nothing that can be done in the library side to fix this, sorry

skorokithakis, 2023-03-03:
The official example works fine though?

martinberlin, 2023-03-03:
@skorokithakis can you please add details: Exactly what version post a link please to the product (You know there is T5 T5-pro since Lily uses this T5 denomination for many devices)
Then also what program, what settings you are using . Without full details is impossible to reproduce please keep it in mind when adding an issue in any repository

skorokithakis, 2023-03-03:
This is the listing:

https://m.aliexpress.com/item/1005002006058892.html?spm=a2g0n.order_detail.order_detail_item.4.38c3f19cyLUiOB

This is the code that works great:

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/examples/demo

[image]

There's no ghosting at all with it, I'm not sure what waveform it's using, but I assume it's in the repo.

martinberlin, 2023-03-03:
Try using the waveform selecting it from menuconfig or inserting it as a **build_flag**:

     ED047TC1 LILYGO 4.7 inch, correct grays

skorokithakis, 2023-03-03:
Here's a photo of the effect:

[image]

I'm using the `lilygo-t5-47-epd-platformio` example.

skorokithakis, 2023-03-03:
The example already has `EPD_DISPLAY_TYPE_ED047TC1` configured, as far as I can see.

martinberlin, 2023-03-03:
Thanks will try it tonight and let you know the result in mine

martinberlin, 2023-03-03:
[image]
Tried and demo looks normal. Just tried also to draw a central filled circle.
This lines are drawn only for 50 millis or so when updating them comes clear with the image. Sadly I’ve no idea how to reproduce your issue

skorokithakis, 2023-03-03:
Ah, that's too bad. I wonder if I have a different version of this, because it happens consistently for me, and only with your library. The lines are always in the same place, as well.

Thanks for your time, I'll update here if I have any more information at some point.

martinberlin, 2023-03-03:
It’s not actually mine is VRoland. But is actually not the component. There must be something different in your configuration.
I suggest forgetting the demos and based on that make your own test program and start drawing something directly with EPDiy instead of using the old version that Lilygo uses.

skorokithakis, 2023-03-03:
I'll try that, thanks!

retsamedoc, 2023-04-02:
I've seen this issue using the T5-4.7 demo in this repo. It appears to be something to do with the refresh? When running the demo, the issue appears after waking the device up from sleep.

Running this command to clear the screen seems to do better:
`epd_clear_area_cycles(epd_full_screen(), 2, 12);`

matthewfcarlson, 2023-04-16:
I'm also seeing this but it seems to be worse in portrait compared to the landscape. I'm digging in and investigating the differences between the lilyGo version and the epdiy version. Running the clear command at the beginning gets it nice and clear, it's just drawing text that seems to cause problems. 
[image]

matthewfcarlson, 2023-04-16:
It seems strange, sometimes it draws and sometimes it doesn't.

matthewfcarlson, 2023-04-16:
Honestly, I wonder if this is related: https://github.com/vroland/epdiy/issues/203#issuecomment-1259978229
It seems like when I do a full screen draw it works great but other times it's pretty usable. 
https://user-images.githubusercontent.com/6557490/232271298-9fae9dcb-99b8-47c0-9312-7f86d3ba37c1.mp4

martinberlin, 2023-06-08:
The issue with this is that Lilygo team kept using an older version of EPDiy V2 while this library kept evolving. I cannot reproduce this for me it’s drawing fine.
What I won’t do is to try with their library since it lacks many things that where added here like bugfixes, software rotation and much more.

martinberlin, 2023-06-09:
Please paste here the program where you are experiencing this. I will try it on same LilyGo and we can then check if it's your display or there an issue

skorokithakis, 2023-06-09:
It's this:

https://github.com/vroland/epdiy/blob/master/examples/lilygo-t5-47-epd-platformio/src/main.cpp

martinberlin, 2023-06-09:
It must be something with your display. You should contact Lilygo and ask for a replacement.
Just tried the demo, opened it with Platformio, and it works as expected. I don't see any of this lines in the screen. 
[image]
[image]

closed 2023-06-09 by martinberlin