vroland/epdiy #223: can‘t start

3058132083, 2023-03-10:
My epdV6 does not start normally, and the serial port outputs the following information
I (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (29) boot: chip revision: v3.0
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32: SPI Speed      : 80MHz
I (38) boot.esp32: SPI Mode       : QIO
I (43) boot.esp32: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00177000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=d5680h (874112) map
I (339) esp_image: segment 1: paddr=000e56a8 vaddr=3ffb0000 size=022cch (  8908) load
I (342) esp_image: segment 2: paddr=000e797c vaddr=40080000 size=0869ch ( 34460) load
I (356) esp_image: segment 3: paddr=000f0020 vaddr=400d0020 size=1e838h (124984) map
I (391) esp_image: segment 4: paddr=0010e860 vaddr=4008869c size=09054h ( 36948) load
I (412) boot: Loaded app from partition at offset 0x10000
I (412) boot: Disabling RNG early entropy source...
I (423) quad_psram: This chip is ESP32-D0WD
I (424) esp_psram: Found 8MB PSRAM device
I (424) esp_psram: Speed: 80MHz
I (425) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (433) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (442) cpu_start: Pro cpu up.
I (446) cpu_start: Starting app cpu, entry point is 0x400813d0
0x400813d0: call_start_cpu1 at /COMPONENT_ESP_SYSTEM_DIR/port/cpu_start.c:142

I (438) cpu_start: App cpu up.
I (922) esp_psram: SPI SRAM memory test OK
I (930) cpu_start: Pro cpu start user code
I (930) cpu_start: cpu freq: 240000000 Hz
I (930) cpu_start: Application information:
I (933) cpu_start: Project name:     firmware
I (938) cpu_start: App version:      1.0.1-43-g12c67ce
I (944) cpu_start: Compile time:
I (949) cpu_start: ELF file SHA256:  c723dbdaed9e05e3...
I (955) cpu_start: ESP-IDF:          v5.0.1-dirty
I (960) cpu_start: Min chip rev:     v0.0
I (965) cpu_start: Max chip rev:     v3.99
I (969) cpu_start: Chip rev:         v3.0
I (974) heap_init: Initializing. RAM available for dynamic allocation:
I (981) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (987) heap_init: At 3FFB2E50 len 0002D1B0 (180 KiB): DRAM
I (994) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1000) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1007) heap_init: At 400916F0 len 0000E910 (58 KiB): IRAM
I (1013) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1021) spi_flash: detected chip: generic
I (1025) spi_flash: flash io: qio
W (1029) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (1040) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Heap summary for capabilities 0x00000800:
  At 0x3ffae6e0 len 6432 free 108 allocated 5808 min_free 108
    largest_free_block 96 alloc_blocks 34 free_blocks 1 total_blocks 35
  At 0x3ffb2e50 len 184752 free 172184 allocated 11608 min_free 172184
    largest_free_block 172032 alloc_blocks 5 free_blocks 1 total_blocks 6
  At 0x3ffe0440 len 15072 free 14656 allocated 0 min_free 14656
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112968 allocated 0 min_free 112968
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x400916f0 len 59664 free 58836 allocated 20 min_free 58836
    largest_free_block 57344 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 358752 allocated 17436 min_free 358752 largest_free_block 172032
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194304 free 4192136 allocated 0 min_free 4192136
    largest_free_block 4128768 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 4192136 allocated 0 min_free 4192136 largest_free_block 4128768
Dimensions after rotation, width: 1024 height: 758

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d9204
0x400d9204: epd_board_poweron at /COMPONENT_EPD_DRIVER_DIR/board/epd_board_v6.c:208 (discriminator 1)

file: "C:/msys64/home/Administrator/epdiy/src/epd_driver/board/epd_board_v6.c" line 208
func: epd_board_poweron
expression: tps_write_register(config_reg.port, TPS_REG_ENABLE, 0x3F)

abort() was called at PC 0x4008b358 on core 0
0x4008b358: _esp_error_check_failed at /COMPONENT_ESP_SYSTEM_DIR/esp_err.c:47

Backtrace: 0x4008216d:0x3ffb4f40 0x4008b361:0x3ffb4f60 0x40090241:0x3ffb4f80 0x4008b358:0x3ffb4ff0 0x400d9204:0x3ffb5020 0x400d7ac5:0x3ffb5050 0x400d6a79:0x3ffb5070 0x400d6e8e:0x3ffb5140 0x400ee2cc:0x3ffb5160
0x4008216d: panic_abort at /COMPONENT_ESP_SYSTEM_DIR/panic.c:423

0x4008b361: esp_system_abort at /COMPONENT_ESP_SYSTEM_DIR/esp_system.c:153

0x40090241: abort at /COMPONENT_NEWLIB_DIR/abort.c:38

0x4008b358: _esp_error_check_failed at /COMPONENT_ESP_SYSTEM_DIR/esp_err.c:47

0x400d9204: epd_board_poweron at /COMPONENT_EPD_DRIVER_DIR/board/epd_board_v6.c:208 (discriminator 1)

0x400d7ac5: epd_poweron at /COMPONENT_EPD_DRIVER_DIR/display_ops.c:40

0x400d6a79: idf_loop at /COMPONENT_MAIN_DIR/main.c:81

0x400d6e8e: app_main at /COMPONENT_MAIN_DIR/main.c:261 (discriminator 1)

0x400ee2cc: main_task at /COMPONENT_FREERTOS_DIR/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

ELF file SHA256: c723dbdaed9e05e3

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7248
load:0x40078000,len:16484
ho 0 tail 12 room 4
load:0x40080400,len:4236
0x40080400: _init at ??:?

entry 0x40080664

3058132083, 2023-03-11:
Do you have a discord channel to discuss these issues

vroland, 2023-03-11:
We don't have discord, but there is a slack channel ;)
Apparently the TPS chip does not respond to I2C. Usually this is a soldering issue. Can you double-check the soldering? Also, does the TPS chip show up in an I2C scan?

3058132083, 2023-03-11:
OK,I'll check it.

3058132083, 2023-03-12:
I rewelded, and now there is no error message, but there is still no voltage

martinberlin, 2023-05-05:
> I rewelded, and now there is no error message, but there is still no voltage

Check the return error messages that come from TPS65185 from I2C. If PWR_GOOD signal does not turn HIGH there is some issue with starting the voltage pumps. Check datasheet, I use to have to have some similar issues and I remember it helped to read the values.

https://www.ti.com/product/TPS65185#tech-docs

**Read datasheet with detail:**

- Feature Description (continued)  8.3.8 Fault Handling And Recovery
- Table 8. INT_EN1 Register Field Descriptions 
- Table 9. INT_EN2 Register Field Descriptions

Understanding the values of those two registers can be the key to understand what is going on. 
Check also that the power to TPS PMIC is stable.

martinberlin, 2023-05-05:
@vroland I think that it will be nice in this topic, for example for the developers starting with V6, to have a Firmware demo that just starts TPS65185 and makes some checks. Also delivers the INT_EN1, INT_EN2 values maybe with some additional info coming from the hardware that we can define. So it's easier to find where the error is.
Because I know how frustrating can be to debug it when you don't get the high voltages running...

mickeprag, 2023-05-05:
This patch at least prints the INT1/2 registers to the log:
https://github.com/vroland/epdiy/pull/248/commits/13fd04ef93b47559c3799bdd5170e538d4fb579b

martinberlin, 2023-06-09:
Thanks Micke,
@3058132083 there is more than 2 months without any feedback from your side. Closing this unless you really want to keep on trying.
Please contact me at the email that you can see in my github profile and I will reopen this then.

closed 2023-06-09 by martinberlin