vroland/epdiy #229: egermacegermac

chromoxdor, 2023-04-14:


closed 2023-04-14 by chromoxdor

martinberlin, 2023-04-15:
For gods sake @chromoxdor what are you doing here?
If there is one more Issue like this I will block your account and will never have the chance to interact anymore with this repository.

chromoxdor, 2023-04-15:
Hey, i am really sorry for that. I was hoping to delete these messages and leave silently without anybody noticing.
It is really embarrassing. 😳

> what are you doing here?
It was actually a microcontroller of mine going rogue. It was programmed as a HID device sending some characters (as a kind of hello world example) to my computer.
My mistake was being on the phone and at the same time producing and uploading shitty code without really noticing.
I really have no clue why this repository and how this happend since i can not remember to have visited this repo.
But again, i was on the phone and somehow lost focus on what i was doing... 🤷‍♂️

There will be no issues from me anymore. I promise :)
Sorry again....

chromoxdor, 2023-04-15:
But this lead me to look at you work. Very interesting! :)