vroland/epdiy #249: new v6 assembled strange display behavior

grelasroberto, 2023-05-16:
Hi, I just assembled a new V6 board with a ESP32 WROVER-B. Tried with ED60SC4(LF) and ED060XC3(LF).  I set up in menuconfig the right dislpay type for both, and the VCOM value for de 60SC4 as the other one does not have any label.  I also enabled the SPI RAM there.

Tried powering from a USB pc port and from several 2A usb supplies, same result.

Checked VCOM at the board and it is ok as well as the -20/22/-15/15 pads.    

in the serial console, everything is looking right, no reboots.

I (3724) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
current temperature: 31
I (4617) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
EPD update 682ms. err: 0
I (4773) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4927) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5081) gpio: GPIO[15]| InputEn:

Here is a video of the 60X  running the examples/demo app, looks like there is a kind of progress bar ..

https://youtu.be/KFZbKpk3CzM  

The display arrived with a blank screen, after several runs it went gradually turning black, first runs I could barely see some letters displaying.

Does someone got into those similar results?  any clues ?

I can read  -20/22/-15/15 from the pads ONLY when the display is updating, o somehow rendering somthing.  Is this right or they should have constant values?

thank you so much

martinberlin, 2023-05-16:
Hi Roberto,

> can read -20/22/-15/15 from the pads ONLY when the display is updating

yes that’s the way it should be. Most examples do a epd_poweroff and turn the voltages down to save power after updating the display.
Voltages seem all the way correct I wonder if there is some power issue but if TPS power good pin is high should not be.
Please update the video privacy settings since only you can see the YouTube link at the moment.

grelasroberto, 2023-05-16:
Oh sorry, I did update the privacy on the video, thank you for your reply!

grelasroberto, 2023-05-18:
I solved!.. TPS power good was high so started digging each ESP32 pin.  Turns out that due to a falied soldering (residual paste)  EP_D0 and EP_CKV were shorted.

got it working now.

thank you so much!

martinberlin, 2023-05-19:
Great that it worked out!

closed 2023-05-19 by martinberlin