vroland/epdiy #252: Is there support for Cyrillic in fonts?

SerjPr, 2023-08-11:
Good afternoon. I tried to add Cyrillic through fontconvert.py script. The font is converted, but the Latin works and the Cyrillic does not. Or did I make a mistake due to little experience in the code, or is there no Cyrillic support? I tried different fonts but the result is the same.

martinberlin, 2023-08-15:
Please add the .ttf you are using or a link to download it. Add also:

- the command you are using in fontconvert.py to generate the font
- the generated font file

Please when you write an issue give full details of what you are trying to do and explain how to reproduce it adding all necessary information.

martinberlin, 2023-08-25:
Dear @SerjPr 
Cyrillic means: "an alphabet used for writing Old Church Slavic and for Russian and a number of other languages of eastern Europe and Asia."

Right? So you are asking about using specific characters that are used for example in Polish/Russian alphabet.
If you open **fontconvert.py** you will see inside they are defined intervals to render only the characters you need and avoid making a huge fonts header file:

```
# inclusive unicode code point intervals
# must not overlap and be in ascending order
intervals = [
    (32, 126),
    (160, 255),
    # punctuation
    (0x2010, 0x205F),
    # arrows
    (0x2190, 0x21FF),
    # math
    #(0x2200, 0x22FF),
    # symbols
    (0x2300, 0x23FF),
```

Try adding there the interval for Cyrillic fonts. Not all font viewers in Linux will show you that part, please get ahold a Font software like Font Forge or similar where you can check the interval range you want to add:

[image]

After that check if the font itself is in font.h file.  Then what I'm not sure if works and you have to try for yourself is to check if drawing that string, the matched character will look-up and print the character. But I think that part you where missing since I don't see any Cyrillic font in your font header file.

martinberlin, 2023-09-14:
Posts will be closed after 2 months without replies after the **waiting for feedback** label is applied. @SerjPr please give some feedback if we can help you further

closed 2023-10-20 by martinberlin