vroland/epdiy #253: V5 board cannot render ENS060XCD (-1.85v) correctly.

lanistor, 2023-08-11:
I know `ED060XCD(v)` is supported, and i had tested it.
But `ENS060XCD(-1.85v)` cannot render correctly, it only can render half screen correctly:

[image]

I'm guessing it's because the waveform doesn't support it, but I don't know how to create the correct waveform.

I got the specification and waveform file of this model screen, can someone help me to see how to solve it?

[6 inch XGA EPD Specification For Approval-ENS060XCD-20210631.pdf](https://github.com/lanistor/assets/blob/master/epaper/ENS060XCD/6%20inch%20XGA%20EPD%20Specification%20For%20Approval-ENS060XCD-20210631.pdf)

[ENS060XCD_R5H013_2022112201.wbf](https://github.com/lanistor/assets/blob/master/epaper/ENS060XCD/ENS060XCD_R5H013_2022112201.wbf)

martinberlin, 2023-08-13:
Check #226 there is an on-going conversation about waveforms.

But this:

> cannot render correctly, it only can render half screen correctly:

Does not look to me like a waveform issue. Please document correctly this issue and tell details about how to reproduce it.

lanistor, 2023-08-14:
@martinberlin Here is a video of origin v5 board run `epdiy/examples/demo` code:

[https://github.com/lanistor/assets/blob/master/epaper/ENS060XCD/20230814-ENS060XCD.mp4](https://github.com/lanistor/assets/blob/master/epaper/ENS060XCD/20230814-ENS060XCD.mp4)

<iframe 
src="https://github.com/lanistor/assets/blob/master/epaper/ENS060XCD/20230814-ENS060XCD.mp4" 
scrolling="no" border="0" frameborder="no" framespacing="0" allowfullscreen="true"> 
</iframe>

martinberlin, 2023-08-15:
Did you try with the other 6" configuration?
What is your menuconfig setting. That's when I ask: Give me full details to reproduce the issue. That should include all details, menuconfigs, etc. If this is doing the same with all different 6" settings then is for sure not a waveform issue. 
If it was some timing issue then the grays might be off or something like this but not refresh only half of the display. You have another issue there. Please try also with another size of epapers and in next batch add also at least one FPC connector more so you can try with a different display and then you also discard that is not a bad display.

lanistor, 2023-08-18:
I tried another two new screens of ENS060XCD, both have the same behavior, and now i'm sure this model is not supported.

martinberlin, 2023-08-18:
Did you check the resolution settings match your display?

lanistor, 2023-08-18:
Yeah, it matched, i use `ED060XC3`(Component config > E-Paper Driver > Display Type), and the screen resolution is `1024*758`, match the model.

martinberlin, 2023-08-25:
Try this also with the V7 board. If you have an additional spare display and you need help, I would pay for the shipping to me and in exchange I will help you to make it work.

lanistor, 2023-08-26:
> Try this also with the V7 board. If you have an additional spare display and you need help, I would pay for the shipping to me and in exchange I will help you to make it work.

V7 board can support this screen model 👍, i don't know what v7 did fixed this problem.

vroland, 2023-08-29:
Hm, the only major difference should be the LCD vs I2S peripheral... Maybe it's sending a few clocks too many or smth. Probably you'd need an oscilloscope to debug this. But you got it working now?

lanistor, 2023-08-31:
> Hm, the only major difference should be the LCD vs I2S peripheral... Maybe it's sending a few clocks too many or smth. Probably you'd need an oscilloscope to debug this. But you got it working now?

Is there a way to change `I2S peripheral clocks` in v5 board? In v7 seems like we can change it in `displays.c` by `bus_speed`.

vroland, 2023-08-31:
There is, but not an easy way: in `i2s_data_bus` at line ~210 ish there is the APLL and clock divider setup which you can change. But you'll have to lookup the valid dividers in the datasheet, there is no conversion from a frequency in the code.

lanistor, 2023-09-01:
> There is, but not an easy way: in `i2s_data_bus` at line ~210 ish there is the APLL and clock divider setup which you can change. But you'll have to lookup the valid dividers in the datasheet, there is no conversion from a frequency in the code.

Thanks, is should be this line:
```c++
  dev->sample_rate_conf.tx_bits_mod = 8;
```

But this looks like I don't know how to adjust it

vroland, 2023-09-02:
No, that's the i2s bus width. The frequency will be adjustable with some combination of `rtc_clk_apll_enable(` and `clkm_div_a / b`. But how exactly will be specified in the ESP technical reference manual in the I2S clock section.

lanistor, 2023-09-05:
> No, that's the i2s bus width. The frequency will be adjustable with some combination of `rtc_clk_apll_enable(` and `clkm_div_a / b`. But how exactly will be specified in the ESP technical reference manual in the I2S clock section.

Thanks, but it really far beyond my knowledge, i don't how to change adjust the parameters to test.

lanistor, 2023-09-07:
@martinberlin @vroland I have a very embarrassing request: can I send you a screen of this model, and may you debug this screen?

martinberlin, 2023-09-07:
Hi Lanistor,
Sure you can send it to me and I will try to help, please reach me at martin (AT) cale.es
But I have one question before: If you got this working with V7, why you are fighting now to make it work in version 5 ?
v7 will be the way to go forward in this project. The S3 LCD mode seems to me much easier to understand that the old way with I2C LCD transmission mode.

> Thanks, but it really far beyond my knowledge, i don't how to change adjust the parameters to test.

I understand what you are saying and I was exactly in the same position when I started testing this project. It was about 3 years ago and I was just doing my first steps with IDF trying to learn how to build components, my C / C++ really in newbie level at that time. But trying it and reading documentation, failing many times, and fighting back I could understand many things without being an expert (Actually not even close to it, just by trial and error method)
So my advice to you is that if someone tells you go and check this part of the documentation just try and do it. Espressif is one of the companies that really documents well their stuff and this big part of their success so reading their technical datasheets, checking their examples, modifying them and trying them yourself is a good idea.
That's the way to expand your knowledge. You have to push your boundaries, because no-one will do it for you, you have to do it yourself.

lanistor, 2023-09-10:
@martinberlin Thanks a lot, had sent you an email for contacting.

Our product is about to be mass-produced and we have been using v5, so we cannot directly use v7 recently. We will use v7 in the next PCB version.

martinberlin, 2023-09-18:
I’ve been testing on this displays that @lanistor sent me and so far I get the same effect. Tried to change the recommended:

> The frequency will be adjustable with some combination of rtc_clk_apll_enable( and clkm_div_a / b

But so far I also get the half of the display only will keep on trying stuff. Question: this could be because of the timing waveform?   Asked Ace to help me trying the original vendor framework.

martinberlin, 2023-09-18:
@vroland I played like one full hour with the timings and could not get it to work. However disabling the **tx_stop_en** in I2S seems to fix it for this display.

@lanistor please try this:

In File: /src/output_i2s/i2s_data_bus.c b/src/output_i2s/i2s_data_bus.c

@@ -243,7 +243,7 @@ void i2s_bus_init(i2s_bus_config *cfg, uint32_t epd_row_width) {
 
```C
   // Stop after transmission complete
   dev->conf1.val = 0;
-  dev->conf1.tx_stop_en = 1;  // set the tx_stop_en to 0
+  dev->conf1.tx_stop_en = 0;
```

[image]

Tried this with IDF 5 since I don't have installed the 4.4 but that line seems independent of the IDF version.
Test done with latest branch s3_lcd (Again this should not matter since this is a one line update)

Using this initialization:

```C
epd_init(epd_board_v5, &ED060XC3, EPD_LUT_64K);
```

Both the dragon and the demo examples seem to work well on this display after that small update.

lanistor, 2023-09-19:
@martinberlin This works !!! Amazing!!! you are so awesome!!!
Thank you so much!

BTW: It seems run perfect with other screen models, so if there are problems that i keep `dev->conf1.tx_stop_en = 0` for all screen models?

martinberlin, 2023-09-19:
Awesome is really Valentin that understand how all this code works. I just changed a line.
I guess this could work with other displays too. But he ia really the one that should evalueta that since I've just a few of them in my office.

vroland, 2023-09-20:
Congrats on finding the issue! According to the TRM, this bit stops the word clock after the transmission is over:
```
I2S_TX_STOP_EN Set this bit and the transmitter will stop transmitting BCK signal and WS signal
when tx FIFO is empty. (R/W)
```
So apparently this display needs a couple of clocks more to fill the source register completely. According to the datasheets that have the signal timelines, the EPDs should be fine with a continuous pixel clock though, so we could set that to 0. Let's just confirm with some other displays to make sure this doesn't break anything :)

@martinberlin Does the issue also appear with the V7 board on this display?

martinberlin, 2023-09-20:
> Does the issue also appear with the V7 board on this display?

In v7 board that uses S3 and LCD module this seems to work without any additional change. I started studying slowly the whole LCD new module so I will try to collaborate more getting some self-taught knowledge.
By the way Ace python wbf parser starts to work and I tried this display also with the original framework #226 and it’s a good result but a bit too clear (only 10 darker grays seen then all mixes with white too much)
But very interesting result maybe with some adjustments we could have a great parser that takes WBF and exports an epdiy waveform.

martinberlin, 2023-09-25:
Hey @lanistor with your permission please keep this reopen since we need to test the fix in other displays with @vroland and then we can fix it also for other users when it's merged in master

lanistor, 2023-09-25:
OK, very good, thanks for your job.

martinberlin, 2023-09-29:
@vroland this was tested in 6", 9.7" and 13.3" displays. All seem to work so far with this update. 
So I think it will be safe to leave the config for I2S as:

     dev->conf1.tx_stop_en = 0;

martinberlin, 2023-10-24:
Closing here since is going to be merged soon in master with #251

closed 2023-10-24 by martinberlin