vroland/epdiy #254: Partial screen update.

SerjPr, 2023-08-16:
Good afternoon. I couldn't find the partial screen refresh feature. For example, display the readings of the clock and temperature and humidity sensors on the general screen of the weather station implemented on LilyGo-EPD47. Option to override the epd_clear_area(area) function; causes the area being updated to flicker. I used this option.

Rect_t area = {
    .x = 0,
    .y = 60,
    .width = 240,
    .height = 170,
  };
  int cursor_x = 50;
  int cursor_y = 120;
  
  for (int i=240; i >=200; i=i-10){
     epd_push_pixels(area, 20, i);
   }
  writeln((GFXfont *)&OpenSans24B, tt, &cursor_x, &cursor_y, NULL);

But if you turn off the power to the panel, the highlighted area becomes gray. But maybe there is already a ready-made function?

SerjPr, 2023-08-16:
epd_push_pixels(area, 20, 200);
At one time does not brighten the screen. That's why I used a loop.

martinberlin, 2023-08-17:
Sorry but this is not an issue. Please try to use this library instead of Lilygo own implementation that is old and based on a V2 of epdiy.

Check the demo in current master where there are examples of how to do partial refresh using this library.
https://github.com/vroland/epdiy/tree/master/examples/demo

closed 2023-08-17 by martinberlin

martinberlin, 2023-08-17:
Proof of work https://www.youtube.com/watch?v=9W8mtOKtRi8

SerjPr, 2023-08-18:
I am sorry. I didn't give full information when I made the consultation request. I am using the new version of the LOLOGO board on the ESSP 32-s. Do you have support for this version?
I'm trying to migrate to your library, but your library's documentation doesn't have all the information.

martinberlin, 2023-08-18:
I think there is another Issue already asking for this. Check the S3 issues.
Use branch: s3_lcd 
And check to add Lilygo S3 as a new board with the correct pin-up checking their schematic.
Later will send you a link to my own fork with that GPIo config.
And yes please add all details to reproduce the issue next time including link to the Lilygo Model so there is no confusion.

martinberlin, 2023-08-18:
You can try also this branch in my own fork
https://github.com/martinberlin/epdiy-rotation/tree/lilygos3

SerjPr, 2023-08-19:
Thank you! I will try to study your library. I recently started learning programming, purely as a hobby. I'm afraid there will be many questions in the future. ))

martinberlin, 2023-08-20:
Great. For questions and things alike please join: https://epdiy.slack.com Just send me your email to martin AT cale.es and I will send you an invitation