vroland/epdiy #256: Color support and next collaboration examples

martinberlin, 2023-09-14:
One #251 is merged I'm planning a new code sprint that I'm already testing but I need to mature both in my mind but also in the code after some research and discussions on what is the best way to implement it.
This is the summary of what I have in my mind, which still needs to be matured and probably explained extensively and documented in the WiKi section. 
First I should make a clear explanation of how color works with the filters in Color Filter displays (CFA):
Basically each pixel has on top a small transparent filter with color, so if the pixel below is white (255) in actual grayscale, then the light will pass through and it will reflect on the white. Hence you will see red. The panel that I'm testing that you can see in the WiKi page has an RBG pattern. 
Let's say you want to print only Red. Then you have to only set to white those pixels below Red. And the others to gray or some value below 255. So now that is how to show color. But you can also completely ignore the filters and print as always with a grayscale epaper the Dragon example. Then you will just see the dragon in grayscale as always since it completely ignores color.
Note that the background in this CFA epapers is always a bit grayish since they have an additional layer on top. Said that now that we drawn the color basic principles when you use color, your resolution will be always a bit lower, since a color pixel is really drawn by combining 3 dots (RBG). 
But you can of course draw image color parts and still a font rendered only in black completely ignoring color. And that's the difficult part: How to make it easier for the user to do that. 

- [ ] Add a way to manage color. For more detailed info please check [this page in the WiKi](https://github.com/vroland/epdiy/wiki/C-ideas-to-add-color-filter-support)

Best idea so far after discussing this with @vroland would be to add a **epd_set_color(uint8_t r, uint8_t g, uint8_t b)**
Then drawing a pixel or calling any GFX function will move a pixel in the display if the R, G, B parameters match. Otherwise it will be left "as is".
But the question is: Is it this linear way of "asking what filter is on top" the best way to handle color?
I've no idea. It's the simpler way that for sure.

- [ ] Add a way to handle gamma curve for each pixel printed. Gamma can be described as how smoothly black transitions to white on a digital display.

- [ ] New v7 PCB has a RTC chip on board.
Let's make one full example to make a precise timer with it that wakes up every N minutes, 2 maybe, and shows the correct hour, date, month and year in the display. We can use the [PCF8563](https://github.com/martinberlin/esp-idf-pcf8563) interrupt pin, that get's low on defined alarms (It's pulled high), to wake up ESP32S3 and draw only the hour part.

martinberlin, 2023-09-28:
[image]
Further color bugs:
Sometimes, not every render, a pixel shifts from R to G (or to B don’t really know) making this effect in the display:
[image]
Once it’s done color changes in many lines altogether.

vroland, 2023-10-09:
Hm, maybe this is because on the frame restart hack on the LCD peripheral for displays higher than 1024 pixels? It seems weid, because in that case it should be shifted by a block of 4 pixels (1 cycle), but maybe this is correlated. What's the resolution on this display again? Right now it's set to do this after 1000 lines. Does the color line shift if you change the LINE_BATCH value in output_lcd/lcd_driver.c?

martinberlin, 2023-10-09:
Thanks for the hint. **Yes this looks exactly because of this.**
Resolution is: 2232x1680 (WxH)
Setting the LINE_BATCH to 840 that is exactly half of this epd_height() then this effect of color-shifting appears on the middle of the display. Trying to set it higher than 1000 it just crashes.

> It seems weird, because in that case it should be shifted by a block of 4 pixels (1 cycle), but maybe this is correlated. What's the resolution on this display again?

Sure I don't really know how many pixels are, just see that is one color shifted to the right, will do more tests and let you know.
I guess it does not have anything to do with color, right?   It should be possible to see a pixel shifting in any other display after certain amount of rows (Example: the 13.3" I will try that also)
We should try also on the 13.3 (The PCB I sent you has this connector) and draw some vertical lines, check if they look all plain vertical all over the display with some magnifier lens.

mataide, 2023-10-18:
Hello, I have here the Kaleido display https://www.eink.com/product/detail/SC1452-FOA, let me know how I can help to test this Color filter.

martinberlin, 2023-10-18:
Hello @mataide 
If you have some information / datasheet please send it to me with a private email to martin [AT] cale.es and get a working epdiy v7 board. Then we can try something together. So far I had only this Wuxi Weifeng (DES) color displays that are sold by Good-Display.

mataide, 2023-10-18:
@martinberlin Done. Let me know if you got my email. I am very interested in making it work, and I do have contact with the EINK Engineers, so hope that I can help you with this.

mickeprag, 2023-10-20:
Is the kaleido display different from the one you have @martinberlin?
@mataide Where did you get it from?

martinberlin, 2023-10-20:
I'm quite sure the colors arrangement is not the same. But we need to try it so first thing I will do is to sell one of my v7 boards to him so he can try it out. And tonight will take a deeper look to the datasheet and check if the Pinout is the same as v7 current 16 bit, 40 pin connector.

martinberlin, 2025-03-11:
Not sure if we will ever merge this so moving it to a discussion.
Let’s make an issue when there is a real plan to implement it