vroland/epdiy #258: Suggestion, split repo inte multiple repos

mickeprag, 2023-10-20:
Currently this repo contains a lot of stuff, both hardware, software, and breakout boards.

I propose we create an epdiy organisation/group on GitHub where the epdiy software and hardware is in separate repos. Also, if someone makes a support library/integration (like a LVGL driver) this can easier live it's own life in an own repo. All under the common name "epdiy". 
Releases can also be done separately. Releasing like the v7 boards and releaseing v1.0.1 of the software makes more sense in different repos.

What do you guys think?

martinberlin, 2023-10-20:
I always thought this will be a smart idea 💡
And I remember some time ago commenting about this in Slack. But the decision is really @vroland and how he decides to organize this 

+1 is my vote. Because this will also make the component lighter. Ideally Firmware should be only component plus a few useful examples (one nice is waiting to be merged also)

I would suggest also to discuss this with Valentin in next talks and maybe also organizing a bimonthly meeting between collabotors to help each other shaping what is coming and how we can help making epdiy a better tool for everyone.
First priority now should be to merge v7 and make a new release.

vroland, 2023-11-01:
Indeed, we already talked about this a while back and it makes a lot of sense. I would also factor out some larger software examples to not be maintained in-tree.

It will also make epdiy easier to package, e.g., for arduino, platformio, etc because the repository will be much smaller. 
I see this as the next step after the v7 branch is merged. 

I didn't want to do it before because I want to rewrite the history to actually reduce repository size but not loose the software history. That will break local clones / forks though, so we should first get all outstanding PRs merged.

vroland, 2023-11-01:
I created an epdiy-hardware repository here: https://github.com/vroland/epdiy-hardware

I plan to clean this repo up a bit more, separating all the epdiy revisions and adapters into folders and possibly running https://github.com/INTI-CMNB/KiBot to automate up-to-date file exports.
Once it is in good shape, I'll delete the files from this repository and eventually filter them out of history :)

martinberlin, 2023-11-03:
Thanks Valentin,
Labeling this as done since the new hardware repository is already created.
 
I guess we can close this issue after hardware files are cleaned from this repository. It should be also clearly linked from the Readme file and also instead of deleting the hardware folder initially we could also add a Readme saying that is moved to a new repository (Just an idea, should not be necessary if it's properly documented).
The best design for me to make it completely clear is to add a table header where we have a clear separation of terms:

| Hardware | Examples  | 
| ------------- |--------------- |
| [epdiy KiCad](https://github.com/vroland/epdiy-hardware/) | Link |

martinberlin, 2023-11-03:
@vroland I also wanted to mention that since latest PR was merged in master the display table is screwed up
[image]
This will be important to fix since many users refer to it.

vroland, 2023-11-11:
As of 8b960dfda74c9000471965c4292dcd1c80f0ef09 I'd consider this done. Any further proposals to improve structure?

vroland, 2023-11-11:
And I decided I'm actually not going to filter history, if you do a fresh clone now without the old hardware submodules you get ~100MB, which is not great, but with docs and some example data it won't be tiny anyway, and that way we don't break history for all the forks.

martinberlin, 2023-11-12:
Good job Valentin, I think it became much clearer and lighter now the repository.
Question: What will be the destination of the moved examples? ( terminal & mpd )

vroland, 2023-11-13:
They are moved to the repos linked in the commit message and in the README: https://github.com/vroland/epdiy#get-inspired

mickeprag, 2023-11-13:
> As of [8b960df](https://github.com/vroland/epdiy/commit/8b960dfda74c9000471965c4292dcd1c80f0ef09) I'd consider this done. Any further proposals to improve structure?

I am happy. Not a separate organization as suggested, but that's fine!

closed 2023-11-13 by mickeprag