vroland/epdiy #260: About costs

3058132083, 2023-11-05:
[image]
[image]
These two chips only have different output currents in the manual. I would like to know if cheaper TPS65185 can be used instead of TPS651851
[image]

seishinkouki, 2023-11-05:
Roughly, if your panel size lower then 9", tps65185 is acceptable
小尺寸面板可以用65185

vroland, 2023-11-05:
Indeed, they are functionally equivalent, except that the TPS651851 has better performance. Make sure to get the RSLR version though, because there is also an RGZ which has a larger package and won't fit.

martinberlin, 2023-11-06:
Interesting issue!  Since epdiy has now also a [hardware repository](https://github.com/vroland/epdiy-hardware/issues) I suggest that further issues about hardware can be discuss directly there.
In my opinion the best would be to make a page in the WiKi section on the hardware side, that can be edited without making commits, etc (Or yes it does but internally and works just like a WiKi Content manager) and there we build a nice table with possible alternatives.
Like @vroland mentioned above the most important thing is that the replacement part has the same footprint.

vroland, 2023-11-06:
@martinberlin I think putting the wiki page into a markdown file that's also built into the pages website. That makes it easier for people to make a PR for that file to add new suggestions. In my experience, the wiki is also a bit hard to find.