vroland/epdiy #261: About use IDF Project

dengfengdz, 2023-11-06:
Hi!  I want to use the ESP-IDF bulid it but have some problem, the error messages is : HINT: The component epdiy could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.

Can you help me analyze it？Thank you!

martinberlin, 2023-11-06:
Hello @dengfengdz 
Please write in detail what are the steps to reproduce what you are experiencing. As far as I know just downloading latest **master** and building one of the examples I don't get this.
It should be explained in detail so another person can reproduce it. Otherwise is very hard to help you.

dengfengdz, 2023-11-06:
Hi ! I downloaded the master and Direct use ESP-IDF 5.1 PowerShell open it.As I enter \epdiy-master\examples\dragon to bulid it ,the Warning message is ：The component epdiy could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
[image]
[image]

martinberlin, 2023-11-06:
Sorry but I cannot reproduce in Linux. Will try later using a Windows PC but at the moment I don't have access to it. My steps:

```
git clone https://github.com/vroland/epdiy.git
cd epdiy/examples/dragon
$ prep_idf 
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...

idf.py build
Successful build.
```

dengfengdz, 2023-11-06:
Thank you，I'm also trying to figure out what's wrong，I used an older version that was compilable.The old version step have  to use idf.py menuconfig

martinberlin, 2023-11-19:
Update @dengfengdz I will still have a week more to wait until I can grab a Windows PC and test it since I do not really own any laptop. So if you have any news please keep us posted. I still do not understand why you are getting this

martinberlin, 2023-12-02:
I tried reproducing this steps and it works correctly. I really do not know what you are doing wrong but please send me your email (martin@luckycloud.de) and I will add you to join SLACK.

[image]

LOG:

```
Using Python in D:\Espressif\python_env\idf5.1_py3.11_env\Scripts\
Python 3.11.2
Using Git in D:\Espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Checking Python compatibility
Setting IDF_PATH: D:\Espressif\frameworks\esp-idf-v5.1.2

Adding ESP-IDF tools to PATH...
    D:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    D:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    D:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    D:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    D:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    D:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    D:\Espressif\tools\cmake\3.24.0\bin
    D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin
    D:\Espressif\tools\ninja\1.10.2\
    D:\Espressif\tools\idf-exe\1.0.3\
    D:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    D:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    D:\Espressif\frameworks\esp-idf-v5.1.2\tools

Checking if Python packages are up to date...
Requirement files:
 - D:\Espressif\frameworks\esp-idf-v5.1.2\tools\requirements\requirements.core.txt
Python being checked: D:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of idf-driver, idf-python-wheels use command 'python.exe D:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe D:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

D:\Espressif\frameworks\esp-idf-v5.1.2>cd
D:\Espressif\frameworks\esp-idf-v5.1.2

D:\Espressif\frameworks\esp-idf-v5.1.2>cd
D:\Espressif\frameworks\esp-idf-v5.1.2

D:\Espressif\frameworks\esp-idf-v5.1.2>cd ..

D:\Espressif\frameworks>cd ..

D:\Espressif>cd ..

D:\>ls
'ls' is not recognized as an internal or external command,
operable program or batch file.

D:\>list
'list' is not recognized as an internal or external command,
operable program or batch file.

D:\>l
'l' is not recognized as an internal or external command,
operable program or batch file.

D:\>show
'show' is not recognized as an internal or external command,
[... 356 lines cut ...]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
[919/932] Building C object esp-idf/epdiy/CMakeFiles/__idf_epdiy.dir/src/board/epd_board_common.c.objIn file included from D:/Espressif/m/epdiy/src/board/epd_board_common.c:2:
D:/Espressif/frameworks/esp-idf-v5.1.2/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from D:/Espressif/m/epdiy/src/board/epd_board_common.c:3:
D:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
[926/932] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.objD:/Espressif/m/epdiy/examples/dragon/main/main.c: In function 'idf_loop':
D:/Espressif/m/epdiy/examples/dragon/main/main.c:30:23: warning: unused variable '_err' [-Wunused-variable]
   30 |     enum EpdDrawError _err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
      |                       ^~~~
[931/932] Generating binary image from built executableesptool.py v4.7.dev3
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated D:/Espressif/m/epdiy/examples/dragon/build/dragon_example.bin
[932/932] cmd.exe /C "cd /D D:\Espressif\m\epdiy\examples\...spressif/m/epdiy/examples/dragon/build/dragon_example.bin"dragon_example.bin binary size 0xb4720 bytes. Smallest app partition is 0x177000 bytes. 0xc28e0 bytes (52%) free.

Project build complete. To flash, run this command:
D:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe ..\..\..\..\frameworks\esp-idf-v5.1.2\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\dragon_example.bin
or run 'idf.py -p (PORT) flash'
```

martinberlin, 2023-12-05:
Reopening here since @dengfengdz is not the only getting this

mataide, 2023-12-06:
Hey guys, I found a few issues to solve this problem:

1. Check if you have installed:

- C++ Clang Compiler for Windows
- C++ Cmake tools for Windows

[image]

2. If you renamed the folder from epdiy to xxxx-epdiy then you will need to replace:

- REQUIRES epdiy to REQUIRES xxxx-epdiy in all CMakeLists.txt.

@martinberlin The first one we can add this info into the docs, the second one, I dont know if its possible to avoid this problem and add a tag that wont change even if the folder is renamed.

martinberlin, 2023-12-06:
Thanks Marcus,
@dengfengdz can please check if this works also for you?

martinberlin, 2023-12-22:
Hello @vroland 
do you have any idea on how to solve this issue?
I really don't since as you can see in the comments I grabbed my Partner Windows laptop and could not reproduce it. It seems, reading between lines, the issue arises when you rename the epdiy folder into something else. 
And that makes sense, since now epdiy in this version, is ready to be used "as a component". If this is correct then we can safely close here.

> "git clone ..."This will make my program work.

Yes @dengfengdz that is the correct way to do it. You just clone it and by default will come with the **epdiy** name as the component name is.

vroland, 2024-01-21:
If that's the root of the issue, I think we can indeed close it. Looks like an issue specific to the build environment.

closed 2024-01-21 by vroland