vroland/epdiy #264: I copied the pcb of epdiy-v5 version, and now I want to produce it in bulk, and I want to get your approval

ShiningMan123, 2023-11-16:
Dear Valentin Roland

I am an engineer from Good Display, and I admire this project very much, so I copied pcb, I think you know this company, now I want to mass produce the pcb I copied, and sell it together with our electronic paper. I will also hang the link of the warehouse on our website, which is a commercial behavior. I have to get your permission. I hope I can get your permission

vroland, 2023-11-16:
Hi, generally the terms of the board and software licenses allow that, so that is not a problem. I'll just mention that this project is developed in my free time, so there is no guarantee of support to you or your customers for any software or hardware issues, especially since I don't own your hardware and displays. Are you prepared to handle support for your customers?

Also I'd advise against using v5. It is slower thatn v7, uses legacy parts with limited availability and is a two-layer board with questionable grounding, which I won't vouch for in a production setting. I'd opt for V7 instead, possibly without the optional RTC.

martinberlin, 2023-11-17:
Valentin the only problem I see here is that producing in Bulk the PCB, no matter what version, and not giving proper support the Issues will inevitably come to this repository. GoodDisplay is mostly a hardware manufacturer and as far as I know is not github friendly and does not have the structure to give support to the epdiy community.
Apart of the Licenses, that allow this as you mention, I would like GoodDisplay to bring back something to the community in case this works somehow. What I oppose as a long-term collaborator of this project is that a company produces in bulk the PCB without giving something in return.
And that is a decision needs to happen at a higher level than @ShiningMan123 mentions.

GoodDisplay, as they are resellers of https://en.wf-tech.com DES color epapers, should at least help somehow to implement color or with the waveforms part. If they want to sell PCBs in bulk they have also the responsibility to give support and to get somehow involved in the project.
At least I would like to have the understanding that a company that chooses to make PCB's in bulk with somehow commit to the project and give back something when it generates profit from an open-source project. If that does not happen and it's not a compromise, I oppose as a collaborator to use my free time to give support, knowing that a company that produces hardware will be mostly interested in producing in bulk and bringing all the support back here.

ShiningMan123, 2023-11-17:
Of course, this does not require a higher level decision, and I am willing to contribute to this warehouse

vroland, 2023-11-17:
@ShiningMan123 Again, do you have a plan for providing support for your customers? Because if you don't they will inevitably ask here. And this is not what this project is for.

ShiningMan123, 2023-11-20:
Dear Valentin Roland
Of course, we will provide relevant technical support for customers. Please rest assured

vroland, 2023-11-20:
Ok, thanks, good to know. Btw, any reason why you are using v5 instead of the newer revisions?

martinberlin, 2023-12-22:
No feedback as I've seen there is no activity on this. @ShiningMan123 if you want to further comment on this please use the Slack channel just send me a Twitter message and I invite you

closed 2023-12-22 by martinberlin