vroland/epdiy #266: EC060KH1 color panel support

vanarebane, 2023-11-20:
Hi

Can we add support for this color display?

The datasheet for it is [here](https://1drv.ms/b/s!ApRLRqw9Uoyuvpo5pdrEOj-P9l8xAQ?e=9OrQ1s)

martinberlin, 2023-11-20:
Hello @vanarebane ,
And welcome to epdiy. Please notice that there was an existing issue about this display #88 (Reopened it since I think might be a duplicate of this one since is the same display)
There the author mentions:

> Do you think there is a way to use this board for a 6 inch CFA (Color Filter Array) panel?  The controller timing is the same as ED060SC4, however it seems need another lookup table to show colors.

Did you research how this displays work? Please check my color issue.

> Can anyone add support for this [color display](https://www.panelook.com/SA1452-FOA_E%20Ink_6.0_EPD_overview_49310.html)?

This is open-source Firmware to support epapers but you need to do your part, research, understand a bit how epdiy works and also help to implement it yourself. Please get some background info in the #256 color support issue and also try to provide more background information about how Kaleido works.
Then if you want to keep on forward with this research we leave this issue open and close the old one. Just opened it so it's more visible in case has something that helps you here.

martinberlin, 2023-11-20:
I suspect the pixel arrangement is very similar to the DES color from Wuxi Weifeng https://en.wf-tech.com/pro_1341.html
so if you have one epdiy board you can try my branch where I'm trying to implement color. But the only function implemented is draw_cpixel (draw color pixel) there is still no GFX functions supporting color.

vanarebane, 2023-11-26:
@martinberlin, 
No, I have not yet looked deeper into the color display datasheet. I think I need to get to know some epd basics first.
I'm working on getting myself a copy of epdiy, I was thinking to make a lighter version of the v7, with touch and pwm background also implemented, and right connectors positioned just for what I have in my project.
When I was researching the colour eink display, I believed I could get a cheap second hand Pocketbook Color, but someone was faster than me and purchased it. Then I looked only for a panel separately from [Aliexpress](https://vi.aliexpress.com/item/1005004285847831.html), but it's a bit too pricey for just testing. And I started to doubt if color displays epd-s are good at the moment, as they are too dark. I have PocketBook InkPad Color 7,8" for reading and I almost always have to have backlight on.

My personal project right now is going to be [ED060KC1 with backlight](https://vi.aliexpress.com/item/1005003368019572.html) and some touch-glass, either a too-big 8" with GT911 or see if I'll re-use Kindles 6" touch panel. I just need some touch-enabled-glass that is close to 118mm wide.

TL:DR; Perhaps I can contribute with some PWM backlight support and I2C touch chip support to this project.

martinberlin, 2023-11-26:
Hello @vanarebane 

> And I started to doubt if color displays epd-s are good at the moment, as they are too dark. I have PocketBook InkPad Color 7,8" for reading and I almost always have to have backlight on.

Yes that I think happens with all color filters, it has an additional layer on top (or two if touch), and this of course reduces the light. They look all grayish without backlight. I just tested DES from Wuxi Weifeng, sold by Good-Display. Also too expensive for my tests and I just have one.

> Perhaps I can contribute with some PWM backlight support and I2C touch chip support to this project.

About this I think it will be better to collaborate with an LVGL driver. epdiy is really only the epaper component. I don't think @vroland has plans to add touch or backlight support in this repository. 
But I know that is possible to use epdiy in LVGL. Please check this repository https://github.com/martinberlin/lv_port_esp32-epaper/ if you are interested in testing. I already added there a GT911 touch driver for LVGL (Really the driver itself is a Espressif component)

vroland, 2023-11-29:
Yes, it looks like lvgl is the way to go for this, it should theoretically already support color / grayscale drawing, touch integration and all that nice stuff...

martinberlin, 2023-12-08:
Thanks for your feedback Valentin.
Wanted to share more insight in supporting color in this display.
[image]

Started to add the display and it works good with the 34 pin connector. I guess it should work in any board no matter what version is. 
But I realized when I tried color that this has a completely different pixel arrangement than color DES. So using the same function to try color comes out all black.

More info: https://github.com/koreader/koreader/issues/6479 in that Issue this interesting guys figured out what is the pixel arrangement. Although I'm starting to realize this color thing is different for each brand and it's actually a pain to handle. 
@vroland I'm actually not 100% sure if it's a good idea to integrate it in epdiy or just make it on the outside. Because if we take epdiy as epaper controller and consider the color filter or CFA, like an external addition to the epaper, then there are like 2 different layers. And epdiy is already working great to support the Eink panel below. What I don't really know if we need to really understand how are the CFA setups above, that might change from each fabricant (For now Eink and WF-Tech, but maybe there will be more?)

martinberlin, 2023-12-08:
Advancing slowly with this one. Took me a while to get the color lookup function. Still the colors are way off, but at least they show up. This displays without front light are kind of useless must say, colors look always like a dim pastel (Watercolor painting)
[image]
[image]
(A green field...)
Many adjustments have to be made. Specially would like to see it with front-light

martinberlin, 2023-12-22:
Hello @vanarebane 
you asked for this and I accidentally had a Kaleido display coming to my studio and implemented the color. Please check [s3_color_implementation](https://github.com/vroland/epdiy/tree/s3_color_implementation) branch and try it out.
I will only make a Pull request if I see there is real interest on this. And then I will find a display to send Valentin so he can double check it and see if there is any possibility to improve it.

martinberlin, 2024-01-05:
@vanarebane can you check if this works or you?
I'm working on getting one of this Displays and send it to @vroland so we can analyze if there is a way to add color support to epdiy. If there is no feedback or interest anymore in this topic for months then I will close the issue.  I don't like to have issues open forever when there is no feedback since they do not bring anything more than being an open list

vanarebane, 2024-01-05:
@martinberlin sorry for late reply. I missed the notification that you wrote. 
As I wrote earlier, I decided to not invest in a colour display because they are too dim. 
So for my sake, you can close this topic for now.
Thanks for all the effort, I hope there is some use for colour epds in the future :)

closed 2024-01-05 by vanarebane