vroland/epdiy #270: ED133UT3 Screen Suport

dengfengdz, 2023-12-14:
Hi ，guys！
I tried to use the ED133UT2 code to light up ED133UT3, but it didn't work properly and the screen didn't respond at all. I only have the ED133UT3 specification at present, and I didn't see any difference between them。

dengfengdz, 2023-12-14:
[VB3300-NCC (ED133UT3).pdf](https://github.com/vroland/epdiy/files/13668042/VB3300-NCC.ED133UT3.pdf)

martinberlin, 2023-12-17:
[image]
Hi there @vroland if I see this correctly in the datasheet mentions Gage positive supply to be 27 V. If that is different from ED133UT2 it could be that the Gate of the pixel FETs is not receiving enough voltage to open and let pass the +15 volts right?
Can that be the issue @dengfengdz is experimenting here?

vroland, 2023-12-27:
Hm, it could be. In practice it looked like +22V worked for all displays, although some specify a higher voltage. But it could be that this batch is particularly sensitive. The voltage can be raised by writing to some registers in the TPS65185, but this is not implemented in the public interface of epdiy. 
But it could also be that something is different with the pinout from UT2, I don't have a datasheet available on this machine, but it should be available somewhere to compare.

martinberlin, 2024-03-12:
Hello @dengfengdz please provide feedback about this or close the issue. 
I'm not a big fan of keeping issues forever open since then we are lost trying to give support.

jonardan, 2024-03-18:
Hello, i am using a ED133UT3 with an epdiy v5 and actually it works, but compared to a board from gooddisplay the colors are a bit washed out and don't look too good, I just bought an epdiy v7 to see if it will improve the image quality (because right now i'm setting vcom manually and i'm not too sure the value is exactly correct) do you think that issue with image quality is related to waveform file or it could be the vcom or something else?

martinberlin, 2024-03-19:
@jonardan please send me an E-Mail to martin@cale.es and I will send you a new Waveform to try in the 13.3" display?

martinberlin, 2024-07-31:
Planning to test this one too: https://github.com/vroland/epdiy/discussions/341
@jonardan did you designed an adapter for this?