vroland/epdiy #274: v7 firmware and the switch to a ESP32-S3 module

second-string, 2024-01-04:
I made a custom PCB that uses epdiy to drive an e-ink display. The only modifications I made [in my fork](https://github.com/vroland/epdiy/compare/master...second-string:epdiy:master) was to switch a pin mapping and disable the RTC sleep for that pin. The esp module I used was an ESP32-WROVER-E with 8MB external flash and 8MB PSRAM.

I can see that the new v7 board is using one of the S3-based modules, switching over to WROOM from WROVER, and going to 16MB flash / 8MB PSRAM.

My question is: are there any underlying firmware changes in the v7 firmware that wouldn't work on a WROVER chip with the same memory specs? I know the S3 is still an Xtensa core and not their RISC-V one, but I'm not sure if there are underlying esp-idf changes or something that would cause the newer repo firmware to not work on a WROVER. I'd be fine making small tweaks to a fork like I did previously, but would like to avoid a deep dive into the firmware if avoidable.

martinberlin, 2024-01-05:
> there any underlying firmware changes in the v7 firmware that wouldn't work on a WROVER chip with the same memory specs?

The v7 Firmware uses LCD Module internally instead of I2S as it did in the ESP32 WRover version. So you definitely can’t mix one with the other.
But this is resolved and there are precompiler conditionals that will adapt to one tatted or the other. 
There are many things that change if you take a deep dive. Sending the pixels is different, the hor. sync and vsync is different, they are two different ways to send the framebuffer. Depending on the target you need to use one method or the other.
V7 using S3 is also faster than ESP32 WRover

second-string, 2024-01-05:
Makes sense. I might lay out two different versions of the board, one for each ESP32 version, as I'd like to use the newer chipset and newer firmware but I'd rather not get stuck on this right now if it doesn't transfer easily.

I can update if I find anything that I need to debug or that isn't know yet. Thanks for the answer!

closed 2024-01-05 by second-string

martinberlin, 2024-01-05:
Sure no problem. With the new boards way to do it that @mickeprag implemented is easier than before to add new custom boards. 
Just keep in mind that the easier way to have v7 with S3 working is to stick to the same schematics and signals, meaning to use same IO expander and TPS65185 to drive the voltages. If you do it that way then it should work out of the box provided that the layout is right.

second-string, 2024-01-05:
The switch to the TI TPS6518X chip is the main driver to upgrade this board, it's a cheaper BOM, a simpler layout, and available :)