vroland/epdiy #276: S3 board on ED060SC screens seems rendering not clear?

lanistor, 2024-01-12:
Using the lastest `s3_lcd` branch, rendering the screen `ED060SCP`, the rendering seems not clear, here is the result:
[image]

I'm using the the s3 board made at `2023-07-25`, i don't if it's the newest hardware design.

And here is the datasheet of `ED060SCP`: https://github.com/lanistor/assets/blob/master/epaper/ED060SCP/ED060SCP%20(A58)%20Ver%203.0_600-800.pdf

martinberlin, 2024-01-14:
I believe I have one of this displays you sent me so I will make some tests over the week and let you know. 
Please copy your epdiy initialization since I would like to know what display you are using ( **EpdDisplay_t**  declaration )

lanistor, 2024-01-14:
> I believe I have one of this displays you sent me so I will make some tests over the week and let you know. Please copy your epdiy initialization since I would like to know what display you are using ( **EpdDisplay_t** declaration )

Here is my code, changed `example/demo/main/main.c` at branch `s3_lcd` a little:
```
/* Simple firmware for a ESP32 displaying a static image on an EPaper Screen.
 *
 * Write an image into a header file using a 3...2...1...0 format per pixel,
 * for 4 bits color (16 colors - well, greys.) MSB first.  At 80 MHz, screen
 * clears execute in 1.075 seconds and images are drawn in 1.531 seconds.
 */

#include <esp_heap_caps.h>
#include <esp_log.h>
#include <esp_sleep.h>
#include <esp_timer.h>
#include <esp_types.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <inttypes.h>
#include <stdio.h>
#include <string.h>
#include <time.h>

#include <epdiy.h>

#include "sdkconfig.h"

#include "firasans_12.h"
#include "firasans_20.h"
#include "img_beach.h"
#include "img_board.h"
#include "img_zebra.h"

#define WAVEFORM EPD_BUILTIN_WAVEFORM

// choose the default demo board depending on the architecture
#ifdef CONFIG_IDF_TARGET_ESP32
#define DEMO_BOARD epd_board_v5
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
#define DEMO_BOARD epd_board_v7
#endif

EpdiyHighlevelState hl;

void idf_setup() {
    epd_init(&DEMO_BOARD, &ED060SCT, EPD_LUT_64K);
    // Set VCOM for boards that allow to set this in software (in mV).
    // This will print an error if unsupported. In this case,
    // set VCOM using the hardware potentiometer and delete this line.
    epd_set_vcom(1560);

    hl = epd_hl_init(WAVEFORM);

    // Default orientation is EPD_ROT_LANDSCAPE
    epd_set_rotation(EPD_ROT_LANDSCAPE);

    printf(
        "Dimensions after rotation, width: %d height: %d\n\n", epd_rotated_display_width(),
        epd_rotated_display_height()
    );

    heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);
    heap_caps_print_heap_info(MALLOC_CAP_SPIRAM);
}

#ifndef ARDUINO_ARCH_ESP32
void delay(uint32_t millis) {
    vTaskDelay(millis / portTICK_PERIOD_MS);
}
#endif
[... 174 lines cut ...]
        .width = img_beach_width,
        .height = img_beach_height,
    };

    epd_draw_rotated_image(img_beach_area, img_beach_data, fb);
    epd_poweron();
    checkError(epd_hl_update_screen(&hl, MODE_DU, temperature));
    epd_poweroff();
    printf("#### use time: %ldms\n", clock() - start);

    printf("going to sleep...\n");
    epd_deinit();
    esp_deep_sleep_start();
}

#ifndef ARDUINO_ARCH_ESP32
void app_main() {
    idf_setup();

    while (1) {
        idf_loop();
    };
}
#endif
```

vroland, 2024-01-21:
Hm, that looks strange. Can you test with the ED047TC2 waveform? That could give us a hint to if it's a waveform or a hardware issue.

lanistor, 2024-01-24:
> Hm, that looks strange. Can you test with the ED047TC2 waveform? That could give us a hint to if it's a waveform or a hardware issue.

OK, i will test it soon.

lanistor, 2024-01-30:
> Hm, that looks strange. Can you test with the ED047TC2 waveform? That could give us a hint to if it's a waveform or a hardware issue.

Yeah, i found `ED047TC2 waveform ` rendering much better than `ED060SCP waveform `.

vroland, 2024-02-07:
Hm, that's weird, because the vendor waveforms which the ED047TC1/2 are should work exactly the same on both boards. I guess you could try to reduce the bus speed on the S3 and see if that changes anything? Maybe it's pushing the screen too fast?

lanistor, 2024-02-08:
Got it, i will test it again.

lanistor, 2024-03-05:
Now i'm using `ED047TC2 waveform` to let `ED060SC7` have a good behavior, but `ED060SCE` still not so perfect, will these screens have a better waveform file?

lanistor, 2024-03-06:
[image]
I got the differrence between `ED060SC7` and `ED060SCE`, the `ED060SC7` has an extra IC, i heard this is an Amazon encryption chip.

lanistor, 2024-03-14:
> Hm, supposedly you can dump the waveform from these chips somehow, but the format is not documented afaik. I hope to eventually build something like a waveform generator that can generate different waveforms to try, but that requires some more investigation. Waveforms are kind of the bane of driving the screens yourself, they are hard to get and can even vary from batch to batch of the same display...

Yeah, i also found the problem using my displays: `vary from batch to batch of the same display`.

Now i will close this issue, i know it's really hard to compatible.

closed 2024-03-14 by lanistor