vroland/epdiy #278: Can I use it with rust? If so, are there any examples?

anuger, 2024-01-18:


martinberlin, 2024-01-18:
I don't think you can because this uses specific instructions in the ESP-IDF components to handle LCD module communications. Please show us another projects using parallel LCD transmission mode and we will take a look. First thoughts in my humble opinion is that this component is quite complicated to be ported to another language.

mickeprag, 2024-01-18:
I think he means using epdiy from a rust application. I think it is doable since rust is good at interfacing C-code. But we have no working examples, you are on your own.

martinberlin, 2024-01-18:
> I think he means using epdiy from a rust application 

Thanks Micke.
@anuger has please enlighten us and tell what you wanted to do since we just have a title from you.
Added the “waiting for feedback” tag to ensure you can explain us what is the plan to in the next weeks

anuger, 2024-01-19:
@martinberlin 
I currently have an ed060kc1 E-Paper and an epdiy V7.
I want to use rust to control E-Paper, but I don’t know if it is possible. At present, I only know these https://esp-rs.github.io/std-training/

martinberlin, 2024-03-12:
Since there is no more feedback since almost 2 months from @vroland I'm closing here.
@anuger please make a new one providing more details if there is a serious intention to port this component to Rust. I would be also interested in seeing that happen (And into learn Rust)

closed 2024-03-12 by martinberlin