vroland/epdiy #279: epdiy v7 - no border connection?

tomek-szczesny, 2024-01-18:
Hi,

Been checking out your electronic design and I noticed there is nothing connected to BORDER pins of e-inks... Or maybe I'm not looking hard enough?

Anyway, this information appears to be scarce on the internet, so I'll just say it: Border should be connected to VCOM.

martinberlin, 2024-01-18:
I don’t think there is a strict rule of what you need to do with border since in v7 PCB it’s just offered as an external pin.
And I really don’t think Valentin would leave it as is if should be connected to Vcom

tomek-szczesny, 2024-01-19:
Truth is none of us has NDA with e-ink and access to the proper manual of their products.
I checked epdiy schematic because I didn't know what to do with BORDER pin and thought you guys have it figured out.
Perhaps your vast experience with actual e-inks shows it doesn't matter much, which might be true.

To me it's not even clear what the purpose of this pin is. Some datasheets give as much description as "Border connection". Great..

I use schematic of PineNote as a reference, the only mass produced open source project with e-ink that I know. I assume they got in touch with e-ink at some point, as they purchased thousands of their products. 
They connected BORDER to VCOM, even though schematic shows they were not sure either what to do with it (dummy resistors to VPOS and VNEG).

What somewhat convinces me this is this Waveshare e-ink documentation, which looks like rebranded e-ink datasheet:
https://files.waveshare.com/upload/5/5d/13.3inch-e-paper-specification.pdf

[image]

But, if it's supposed to be always connected to VCOM, why making it a separate pin on each e-ink model? I can't answer that. I can only speculate this connection serves the same purpose as separate VCOM for outermost regions of the screen. Perhaps it makes sense in some applications (such as extreme temperatures), but then again, de facto standard TPS PMIC has only one DAC for VCOM.

I also have seen reports of refresh problems around the edges, but I cannot remember if it was one of the issues here or elsewhere. Perhaps driving BORDER pin with any voltage rather than leave it floating could help solving that.

martinberlin, 2024-01-19:
I really don't know but I can just say that I tried connecting the BORDER to VCOM or even to GND, and I don't notice any difference. Or maybe there is, I don't know. But I recommend taking a look at all great tech stuff written here:
http://essentialscrap.com/eink/ I'm aware that **Petteri Aimonen** was one of the first, if not the first tech developer, that managed to make a parallel epaper flash controlling it with a Microprocessor. And this repository is basically based on his work and research, of course amplificated and reworked by Valentin for epdiy.

tomek-szczesny, 2024-01-19:
That resource is the bible of hacky e-ink driving, I read it many times already. :)

That's what I found disturbing once I realized in most applications BORDER is driven together with VCOM, and some datasheet actually puts limits on that voltage. Certainly +15 or -15 is not a good idea, with all that I know so far.

Okay then, perhaps my assumption as to what exactly BORDER pin does are false. Frankly I'm only in the middle of designing my first dedicated e-ink hardware so I'm not even sure what I'm taking about.

As for future epdiy hardware, perhaps an upopulated 0R resistor to VCOM wouldn't hurt, until some new evidence sheds light on the true purpose of BORDER pin.

vroland, 2024-01-21:
Sounds like reasonable idea. I believe it even varies from display to display, some models definitely react to +-15V to adjust the border color, but as you said I'm not sure if it's a good idea to drive it constantly. The best setup would probably a transistor pair / MUX so it can be driven to GND, +15, -15 and VCOM. But in practice I'd hide the border in a case anyway.

tomek-szczesny, 2024-01-21:
If my understanding is right, driving VCOM to +15V and -15V should also set all pixels to white or black, but we can agree that is not the right thing to do.
I suspect this is what BORDER input is for - to act as a separate VCOM for border region. However as everyone have pointed out, nobody's sure what applications this feature might have.

I looked into my donor e-ink board and confirmed this one does not have BORDER and VCOM shorted, but didn't have time to investigate further at that time.

My point is, unless we are sure that BORDER actually can be safely driven by +15V or -15V, I wouldn't make it a selectable mode of operation.

martinberlin, 2024-02-23:
Hi Tomek,
Is this issue still relevant?
Version 7 of the board has Border in the headers, so it is connected and available to do whatever you desire. The discussion is if it really matters to do so and to achieve what goal.
But in my opinion that is up to everyone particular needs. For me it’s actually fine if the board just exposes Border as an external pin. Epdiy is open source then anyone can expand and drive this up to their needs just modifying the PCB to their particular project