vroland/epdiy #283: Support for ED060KG1 screen

alex180500, 2024-03-03:
Hello, I want to know if the board supports the ED060KG1 screen found in my Kobo Glo HD. This is the technical specifications I found online:
https://jpa.kapsi.fi/stuff/other/eink_information/P-511-708-V1_ED060KG1%20Ver%201.0.pdf

martinberlin, 2024-03-04:
Hi Alex ,
It should support it you should give it a try.
In the table already exist similar models with the same resolution like ED060KC1:
[image] uses same 34 pin connector as mos5 6” displays.

If you are looking for a board I will have some ready in the next weeks just send me a message.

alex180500, 2024-03-06:
Hi, sorry for the late answer but I was super busy. I tested the device (the screen comes from a kobo glo hd) and it's completely unresponsive so I think the screen is dead. Do you know if there is a way for me to test the screen only with a multimeter or other equipment?

martinberlin, 2024-03-07:
Hi Alex,

> Do you know if there is a way for me to test the screen only with a multimeter or other equipment?

No sorry that will not work. This displays need precise timing and signals to work. Please try first your board with other panels and if that works and not this display is probably dead (If you are using the right configuration of course)

martinberlin, 2024-04-07:
Closing here. Please open a new discussion topic if you need help trying the display

closed 2024-04-07 by martinberlin