vroland/epdiy #285: Prepare v7 lcd_driver.c to be ready for IDF 5.3

martinberlin, 2024-03-11:
When trying to build this component with IDF 5.3 we came upon this errors (5.1 works correct)

### 1st error: lcd_ll_set_data_width

```
src/output_lcd/lcd_driver.c:501:5: error: implicit declaration of function 'lcd_ll_set_data_width'; did you mean 'lcd_ll_set_data_wire_width'?
```

This one I don't know why it comes but in the documentation of [ESP32S3 I see only a reference to data_width](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/lcd.html?highlight=lcd%20module#_CPPv4N26esp_lcd_rgb_panel_config_t10data_widthE) but I don't find this function. Anyways it compiles correctly in IDF 5.1, and it of course exists:

_$ grep -rne lcd_ll_set_data_width ._

```
./components/esp_lcd/src/esp_lcd_panel_rgb.c:503:    lcd_ll_set_data_width(rgb_panel->hal.dev, rgb_panel->data_width);
./components/hal/esp32s3/include/hal/lcd_ll.h:277:static inline void lcd_ll_set_data_width(lcd_cam_dev_t *dev, uint32_t width)
```

### 2nd error: Missing argument to function 'lcd_hal_cal_pclk_freq'

```
src/output_lcd/lcd_driver.c:546:21: error: too few arguments to function 'lcd_hal_cal_pclk_freq'
  546 |     uint32_t freq = lcd_hal_cal_pclk_freq(&lcd.hal, 240000000, lcd.config.pixel_clock, flags);
      |                     ^~~~~~~~~
```

This happens because in idf 5.3 there is a new argument at the end called hal_utils_clk_div_t* **lcd_clk_div**

Reference in [5.3 LCD HAL](https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/components/hal/lcd_hal.c#L16).

vroland, 2024-03-14:
Good to know, unfortunately the HAL is not guaranteed to be stable, but the high-level drivers don't work for our purposes :/

martinberlin, 2024-04-25:
Interesting I didn’t know that.
Just wanted to mention that in 5.2 version of IDF there are also issues https://github.com/vroland/epdiy/issues/300 so we should prepare to handle those cases because many people will fall there @vroland 
Otherwise we need to very explicitly advice to try v7 esp32S3 Firmware only with IDF version 5.1 till is solved

vroland, 2024-04-29:
Hm, though something else also plays into it, I think I'm using 5.2.1 and that works without problems for me...

vroland, 2024-04-30:
Ah, nvm, 5.2.1 also crashes for me.

martinberlin, 2024-08-26:
Closing this one since there is another Issue about the same subject and better documented in #345

closed 2024-08-26 by martinberlin