vroland/epdiy #286: is lower psram usage possible?

jonardan, 2024-03-11:
I was working on a project and I was working with psram but then I saw that there was very low psram available, then I saw the the function epd_hl_init() allocates almost all of the psram available so I was wondering, are all those malloc necessary? couldn't they be optimized to use less psram? I was juust asking because without knowing well how the library works i didn't want to modify it too much.

martinberlin, 2024-03-11:
You can not use PSRAM at all:
https://github.com/martinberlin/platformio-epdiy-monochrome-GFX

But the idea of using the external RAM, which is far bigger in the v7 version since uses ESP32S3, is that you can use GFX functions to draw and then output all together to the display. 

If you find new creative ways to do that then please keep this Issue open and we discuss about it. 

Other option as I mentioned before is to use a monochrome smaller buffer and not use PSRAM at all (check that link I sent you)
But for that is also essential you get deep in the library and understand the Eink signals to be able to use your own way to draw / send stuff to the panel. 
On that link I've used Adafruit GFX as a library to draw over a monochrome canvas.

jonardan, 2024-03-11:
as for now i don't think I know enough about the library to optimize this aspect, but as I go thru with my project i might work on it, in the mean time i think i'll buy the v7 to use that additional ram! Thank you for your support.

martinberlin, 2024-03-11:
There is the [last one on stock in my store](https://www.tindie.com/products/fasani/epdiy-v7-816-bit-parallel-epaper-controller/) and I will make a pause in my Tindie activities till mid-April. Good luck either if you get the last or sending some to production

closed 2024-03-11 by martinberlin