vroland/epdiy #297: Areas of the screen not in the rendering area will become lighter in color.

lanistor, 2024-04-14:
Areas of the screen that are not in the rendering area will become lighter in color. For example, screen is 1024*758, rendering area `{x: 550, y: 30, width: 474, height: 60}` for several times, the area `{x: 76, y: 90, width: 474, height: 668}` black color will fade.

Like this:
[image]

No matter how I adjust the voltage, I can't fix this problem.
Is there a way to keep areas of the screen that are not in the rendering area unaffected?

Testing environment:
- board: v5
- screen: 060XD4
- waveform: tested `epdiy_ED060XC3` and `epdiy_ED060SCT`, same behavior.
- UI library: lvgl
- mode: MODE_DU, using `epd_hl_update_area`
- esp idf: release 4.4

Demo code:
```
epd_poweron();
epd_hl_update_area(&hl, updateMode, temperature, first->area);
epd_poweroff();
```

martinberlin, 2024-04-15:
My feedback w/Kindle paper white 
https://youtu.be/HrZnDrv0QJ8?si=tgrA1ztUJoJH-S-3

lanistor, 2024-04-15:
> My feedback w/Kindle paper white

Got it👍

lanistor, 2024-04-15:
> Areas of the screen that are not in the rendering area will become lighter in color.
> 
> For example, screen is `1024*758`, rendering area `{x: 550, y: 30, width: 474, height: 60}` for several times, the area `{x: 76, y: 90, width: 474, height: 668}` black color will fade. Like this: <img alt="截屏2024-04-15 01 43 46" width="889" src="https://private-user-images.githubusercontent.com/15553840/322293127-52cd406f-e07f-4b45-a61c-b2fff3cf6d19.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyMDQzODQsIm5iZiI6MTcxMzIwNDA4NCwicGF0aCI6Ii8xNTU1Mzg0MC8zMjIyOTMxMjctNTJjZDQwNmYtZTA3Zi00YjQ1LWE2MWMtYjJmZmYzY2Y2ZDE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE1VDE4MDEyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwNWU2MGMxOGMzZTk1Yjk1NzAzMWZlMzljNTc4NmE3YjBjNGUwOTUyMjc2MjY2ZDQxYjUzYzhjYjk2N2ZiNTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Z-XZNJR4lZJtfJArahWj8ZSB67x_7cg3HO4cwywf9PE">
> 
> No matter how I adjust the voltage, I can't fix this problem. Is there a way to keep areas' color of the screen that are not in the rendering area unaffected?
> 
> Testing environment:
> 
> * board: `v5`
> * screen: `060XD4`
> * waveform: tested `epdiy_ED060XC3` and `epdiy_ED060SCT`, same behavior.
> * UI library: `lvgl`
> * mode: `MODE_DU`, using `epd_hl_update_area`
> * esp idf: `release 4.4`
> 
> Demo code:
> 
> ```
> epd_poweron();
> epd_hl_update_area(&hl, updateMode, temperature, first->area);
> epd_poweroff();
> ```

@martinberlin Is there a way to avoid color changes in non-rendering areas?

I think if no voltage is applied to an area, its color should not change.

martinberlin, 2024-04-15:
Can you please post some program to replicate this?
I know this happens mostly with SPI epapers. But still never saw it in parallel. It might be that display has some ghosts after partial refreshes, that's why e-readers make a full refresh every 10th or so page turn, but other than that it works good for me. 
But it can be that with this particular eink and program does not work for you. At least you should make a repository and post test cases so we can replicate it

lanistor, 2024-04-16:
@martinberlin  I found the problem, it was a problem with the code I rewrote.

I'm using below code to replace `epd_hl_update_area()`:
```cpp
epd_clear_area_cycles(area, 1, _clear_cycle_time);
epd_hl_update_area_directly(&hl, updateMode, temperature, area);

enum EpdDrawError epd_hl_update_area_directly(EpdiyHighlevelState* state,
                                              enum EpdDrawMode     mode,
                                              int                  temperature,
                                              EpdRect              area) {
  assert(state != NULL);
  // Not right to rotate here since this copies part of buffer directly

  bool previously_white = true;
  bool previously_black = false;

  memset(state->dirty_lines + area.y, 1, area.height);

  enum EpdDrawError err;
  if (previously_white) {
    err = epd_draw_base(epd_full_screen(), state->front_fb, area,
                        MODE_PACKING_2PPB | PREVIOUSLY_WHITE | mode,
                        temperature, state->dirty_lines, state->waveform);
  }

  for (int l = area.y; l < area.y + area.height; l++) {
    if (state->dirty_lines[l] > 0) {
      uint8_t* lfb = state->front_fb + EPD_WIDTH / 2 * l;  // to
      uint8_t* lbb = state->back_fb + EPD_WIDTH / 2 * l;  // from

      for (int x = area.x; x < area.x + area.width; x++) {
        if (x % 2) {
          *(lbb + x / 2) = (*(lfb + x / 2) & 0xF0) | (*(lbb + x / 2) & 0x0F);
        } else {
          *(lbb + x / 2) = (*(lfb + x / 2) & 0x0F) | (*(lbb + x / 2) & 0xF0);
        }
      }
    }
  }
  return err;
}
```

This method will clear afterimages and ensure faster rendering speed, but the problem is: the area rendered by this function, will become lighter in color, when other area rendering.

I wan't to rewrite `epd_hl_update_area_directly`, but not works, such as set `state->difference_fb[l * EPD_WIDTH + x] =
            ((*(lfb + x / 2) & 0x0F) << 4) | (*(lbb + x / 2) & 0x0F);` will not work.

Could you tell me the wrong of this function `epd_hl_update_area_directly`, and how to repair it?

martinberlin, 2024-04-17:
> Could you tell me the wrong of this function epd_hl_update_area_directly, and how to repair it?

I don’t really know since I never try to use this myself. I either use the high level library or I push the raw framebuffer and do a hl_update

lanistor, 2024-04-18:
> > Could you tell me the wrong of this function epd_hl_update_area_directly, and how to repair it?
> 
> I don’t really know since I never try to use this myself. I either use the high level library or I push the raw framebuffer and do a hl_update

It should be because of i don't known the rendering logic of epdiy clearly, i havn't got the side effect of this way.

lanistor, 2024-04-18:
> @martinberlin I found the problem, it was a problem with the code I rewrote.
> 
> I'm using below code to replace `epd_hl_update_area()`:
> 
> ```c++
> epd_clear_area_cycles(area, 1, _clear_cycle_time);
> epd_hl_update_area_directly(&hl, updateMode, temperature, area);
> 
> enum EpdDrawError epd_hl_update_area_directly(EpdiyHighlevelState* state,
>                                               enum EpdDrawMode     mode,
>                                               int                  temperature,
>                                               EpdRect              area) {
>   assert(state != NULL);
>   // Not right to rotate here since this copies part of buffer directly
> 
>   bool previously_white = true;
>   bool previously_black = false;
> 
>   memset(state->dirty_lines + area.y, 1, area.height);
> 
>   enum EpdDrawError err;
>   if (previously_white) {
>     err = epd_draw_base(epd_full_screen(), state->front_fb, area,
>                         MODE_PACKING_2PPB | PREVIOUSLY_WHITE | mode,
>                         temperature, state->dirty_lines, state->waveform);
>   }
> 
>   for (int l = area.y; l < area.y + area.height; l++) {
>     if (state->dirty_lines[l] > 0) {
>       uint8_t* lfb = state->front_fb + EPD_WIDTH / 2 * l;  // to
>       uint8_t* lbb = state->back_fb + EPD_WIDTH / 2 * l;  // from
> 
>       for (int x = area.x; x < area.x + area.width; x++) {
>         if (x % 2) {
>           *(lbb + x / 2) = (*(lfb + x / 2) & 0xF0) | (*(lbb + x / 2) & 0x0F);
>         } else {
>           *(lbb + x / 2) = (*(lfb + x / 2) & 0x0F) | (*(lbb + x / 2) & 0xF0);
>         }
>       }
>     }
>   }
>   return err;
> }
> ```
> 
> This method will clear afterimages and ensure faster rendering speed, but the problem is: the area rendered by this function, will become lighter in color, when other area rendering.
> 
> I wan't to rewrite `epd_hl_update_area_directly`, but not works, such as set `state->difference_fb[l * EPD_WIDTH + x] = ((*(lfb + x / 2) & 0x0F) << 4) | (*(lbb + x / 2) & 0x0F);` will not work.
> 
> Could you tell me the wrong of this function `epd_hl_update_area_directly`, and how to repair it?

> @martinberlin I found the problem, it was a problem with the code I rewrote.
> 
> I'm using below code to replace `epd_hl_update_area()`:
> 
> ```c++
> epd_clear_area_cycles(area, 1, _clear_cycle_time);
> epd_hl_update_area_directly(&hl, updateMode, temperature, area);
> 
> enum EpdDrawError epd_hl_update_area_directly(EpdiyHighlevelState* state,
>                                               enum EpdDrawMode     mode,
>                                               int                  temperature,
>                                               EpdRect              area) {
>   assert(state != NULL);
>   // Not right to rotate here since this copies part of buffer directly
> 
>   bool previously_white = true;
>   bool previously_black = false;
> 
>   memset(state->dirty_lines + area.y, 1, area.height);
> 
>   enum EpdDrawError err;
>   if (previously_white) {
>     err = epd_draw_base(epd_full_screen(), state->front_fb, area,
>                         MODE_PACKING_2PPB | PREVIOUSLY_WHITE | mode,
>                         temperature, state->dirty_lines, state->waveform);
>   }
> 
>   for (int l = area.y; l < area.y + area.height; l++) {
>     if (state->dirty_lines[l] > 0) {
>       uint8_t* lfb = state->front_fb + EPD_WIDTH / 2 * l;  // to
>       uint8_t* lbb = state->back_fb + EPD_WIDTH / 2 * l;  // from
> 
>       for (int x = area.x; x < area.x + area.width; x++) {
>         if (x % 2) {
>           *(lbb + x / 2) = (*(lfb + x / 2) & 0xF0) | (*(lbb + x / 2) & 0x0F);
>         } else {
>           *(lbb + x / 2) = (*(lfb + x / 2) & 0x0F) | (*(lbb + x / 2) & 0xF0);
>         }
>       }
>     }
>   }
>   return err;
> }
> ```
> 
> This method will clear afterimages and ensure faster rendering speed, but the problem is: the area rendered by this function, will become lighter in color, when other area rendering.
> 
> I wan't to rewrite `epd_hl_update_area_directly`, but not works, such as set `state->difference_fb[l * EPD_WIDTH + x] = ((*(lfb + x / 2) & 0x0F) << 4) | (*(lbb + x / 2) & 0x0F);` will not work.
> 
> Could you tell me the wrong of this function `epd_hl_update_area_directly`, and how to repair it?

@vroland May you help me?  I want to get the side effect of this rendering logic, and how to repair it.

lanistor, 2024-04-18:
> > @martinberlin I found the problem, it was a problem with the code I rewrote.
> > I'm using below code to replace `epd_hl_update_area()`:
> > ```c++
> > epd_clear_area_cycles(area, 1, _clear_cycle_time);
> > epd_hl_update_area_directly(&hl, updateMode, temperature, area);
> > 
> > enum EpdDrawError epd_hl_update_area_directly(EpdiyHighlevelState* state,
> >                                               enum EpdDrawMode     mode,
> >                                               int                  temperature,
> >                                               EpdRect              area) {
> >   assert(state != NULL);
> >   // Not right to rotate here since this copies part of buffer directly
> > 
> >   bool previously_white = true;
> >   bool previously_black = false;
> > 
> >   memset(state->dirty_lines + area.y, 1, area.height);
> > 
> >   enum EpdDrawError err;
> >   if (previously_white) {
> >     err = epd_draw_base(epd_full_screen(), state->front_fb, area,
> >                         MODE_PACKING_2PPB | PREVIOUSLY_WHITE | mode,
> >                         temperature, state->dirty_lines, state->waveform);
> >   }
> > 
> >   for (int l = area.y; l < area.y + area.height; l++) {
> >     if (state->dirty_lines[l] > 0) {
> >       uint8_t* lfb = state->front_fb + EPD_WIDTH / 2 * l;  // to
> >       uint8_t* lbb = state->back_fb + EPD_WIDTH / 2 * l;  // from
> > 
> >       for (int x = area.x; x < area.x + area.width; x++) {
> >         if (x % 2) {
> >           *(lbb + x / 2) = (*(lfb + x / 2) & 0xF0) | (*(lbb + x / 2) & 0x0F);
> >         } else {
> >           *(lbb + x / 2) = (*(lfb + x / 2) & 0x0F) | (*(lbb + x / 2) & 0xF0);
> >         }
> >       }
> >     }
> >   }
> >   return err;
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This method will clear afterimages and ensure faster rendering speed, but the problem is: the area rendered by this function, will become lighter in color, when other area rendering.
> > I wan't to rewrite `epd_hl_update_area_directly`, but not works, such as set `state->difference_fb[l * EPD_WIDTH + x] = ((*(lfb + x / 2) & 0x0F) << 4) | (*(lbb + x / 2) & 0x0F);` will not work.
> > Could you tell me the wrong of this function `epd_hl_update_area_directly`, and how to repair it?
> 
> @vroland May you help me? I want to get the side effect of this rendering logic, and repair it.

It seems the problem of `epd_clear_area_cycles`.

vroland, 2024-04-18:
Do you have an image / video of the area that gets lighter? Because to me that sounds a lot like improperly calibrated VCOM. Additionally, you could try if using the ED047TC2 waveform changes anything.

Also, why are you using your own `epd_hl_update_area_directly` instead of `epd_hl_update_area`?

lanistor, 2024-04-19:
> Do you have an image / video of the area that gets lighter? Because to me that sounds a lot like improperly calibrated VCOM. Additionally, you could try if using the ED047TC2 waveform changes anything.
> 
> Also, why are you using your own `epd_hl_update_area_directly` instead of `epd_hl_update_area`?

1.  I tried `ED047TC2`, has same behavior.
2. I adjusted vcom to a suitable value, so it shouldn't be a problem with vcom.
3. I use `epd_clear_area_cycles` + `epd_hl_update_area_directly` to eliminate afterimages, it will rendering  `black` -> `white` -> content image, this will clear the afterimage,`epd_hl_update_area` cannot do this.

And, i tried not using `epd_clear_area_cycles` + `epd_hl_update_area_directly`, instead using rendering `black` -> `white` -> content image, with `epd_hl_update_area `, this problem disappears, so i think it may be `epd_clear_area_cycles`

martinberlin, 2024-04-19:
Please answer this one:

> Also, why are you using your own epd_hl_update_area_directly instead of epd_hl_update_area?

I also would like to understand what is the reason to use your own high level update area instead of the one that is already built in

lanistor, 2024-04-19:
> 3\. eliminate afterimages

This will eliminate afterimages (`epd_clear_area_cycles` + `epd_hl_update_area_directly`), while `epd_hl_update_area` cannot.

lanistor, 2024-04-19:
> Please answer this one:
> 
> > Also, why are you using your own epd_hl_update_area_directly instead of epd_hl_update_area?
> 
> I also would like to understand what is the reason to use your own high level update area instead of the one that is already built in

This way can eliminate afterimages (`epd_clear_area_cycles` + `epd_hl_update_area_directly`), while `epd_hl_update_area` cannot.

vroland, 2024-04-22:
Can  you show a video of both methods? I don't really see why they would behave different, so seeing the outcome would help.

lanistor, 2024-04-25:
> Can you show a video of both methods? I don't really see why they would behave different, so seeing the outcome would help.

### Before
[image]
### After
[image]
### Video
https://github.com/lanistor/assets/blob/master/epaper/060XD4_out_paint_area.mp4

lanistor, 2024-04-28:
> > Can you show a video of both methods? I don't really see why they would behave different, so seeing the outcome would help.
> 
> ### Before
> ![IMG_9951](https://private-user-images.githubusercontent.com/15553840/325732074-cefa3657-2aed-4a2d-8d0e-f0551503fa7e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzMjQwNzksIm5iZiI6MTcxNDMyMzc3OSwicGF0aCI6Ii8xNTU1Mzg0MC8zMjU3MzIwNzQtY2VmYTM2NTctMmFlZC00YTJkLThkMGUtZjA1NTE1MDNmYTdlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI4VDE3MDI1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwYTgwNjU5OTQzYmIwYjUyYWUxMDhhY2UzNWE4ODU3ZDI2NTFhMGJiZGFjYzUyOGMyNzUzNGFmMzhmMzQ2YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XDG6i6ooGKAF3IT-eVk7ZmU94eKJcSWHLLLcwqocb3Y)
> 
> ### After
> ![IMG_9953](https://private-user-images.githubusercontent.com/15553840/325732085-3ec0eeaa-1862-4de9-896f-ca4417ee249f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzMjQwNzksIm5iZiI6MTcxNDMyMzc3OSwicGF0aCI6Ii8xNTU1Mzg0MC8zMjU3MzIwODUtM2VjMGVlYWEtMTg2Mi00ZGU5LTg5NmYtY2E0NDE3ZWUyNDlmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI4VDE3MDI1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3ZDI3MzEzODgyMmY2ODVjODQ5YmI0NTA5MGUwNmFlYzc0MWMyMTFlMDVkMDQ3NGVjZmUxY2RmYmQxYzlmODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8fhhB4hT31AgSc6b-aQhGS0OQSfkNEJFqR9qsbCCtgw)
> 
> ### Video
> https://github.com/lanistor/assets/blob/master/epaper/060XD4_out_paint_area.mp4

@vroland Is this helpful for resolving the problem?

vroland, 2024-04-29:
That makes it more clear, thanks. Curious, I don't really know why that is. I'll mark it as a bug for now and try to reproduce it some time. Do you have a minimal example for reproducing the problem handy?

lanistor, 2024-04-30:
> That makes it more clear, thanks. Curious, I don't really know why that is. I'll mark it as a bug for now and try to reproduce it some time. Do you have a minimal example for reproducing the problem handy?

I made a minimal example : https://github.com/lanistor/epaper-tester

I can reproduct problem with `ED060XD4`, but cannot reproduct with `XC3`, `XCH`, do you have a `ED060XD4`?

vroland, 2024-04-30:
Argh, I think I only have an XC3... But I can try my luck.

martinberlin, 2024-04-30:
I will check because Lanistor sent me 2 XD4 so if I find will send it to you in next package 📦

lanistor, 2024-05-01:
> I will check because Lanistor sent me 2 XD4 and still have one around so if I find it I can send it to you in next package 📦 Just checked so far have a 060XCD

As i remembered, i sent you 2 XCD, not XD4.

lanistor, 2024-05-01:
> Argh, I think I only have an XC3... But I can try my luck.

Could you give me your address? I could send you 2 XD4. Here is my email: `lanistor@163.com`.

lanistor, 2024-05-09:
Hi, @vroland, @martinberlin,

I found another very interesting performance that will help solve this problem: when the circuit board is powered on again, it will slowly display the last rendered content.

Here is a video of my experiment:
https://github.com/lanistor/assets/blob/master/epaper/060XCD_2024-05-10.mp4

And here is my experimental using case:
1. Turn on the power and a black background will appear on the entire screen.
2. Render the block a few times at the top, then a few times at the bottom
3. Disconnect the power and then turn it on again. When rendering the top block, the shape of the block will appear at the bottom.
4. Repeat the above steps, but turn the power off after 3 minutes and then turn it on again. At this time, the shape of the bottom block will fade; if you turn the power on again after 5 minutes, the shape of the bottom block will be almost invisible.
5. When rendering without calling `epd_hl_update_area_directly` but only calling `epd_clear_area_cycles`, the problem disappears.
6. I tested the brand new XCD and XCH and both had this problem; on the second-hand XD4, this problem will be more serious.

This is really a very strange behavior. The only explanation I can think of is: the electrode on the back of the screen has a weak capacitance, which stores a small amount of electricity. When the screen is powered on, these weak currents will cause the displacement of the particles. Because the second-hand screen (XD4) has been used for a long time, the capacitance will increase, causing the particle displacement to increase.

I found the problem, but I don't know how to solve it. This problem is really interesting.

May you find out the way to resolve this problem? I think the problem still lies in `previously_white=true` of `epd_hl_update_area_directly`.

The test code for video recording is here https://github.com/lanistor/epaper-tester/tree/test/case-2, the branch is `test/case-2`.

vroland, 2024-05-10:
Interesting, looks like this particular display model is very sensitive to residual charge / charge buildup. Maybe if we could dump the on-chip waveform we can fix this? Or we have to experiment with the waveforms we have. But this would be my guess that with the waveforms we have there is some imbalanced charge buildup.

lanistor, 2024-05-10:
> Interesting, looks like this particular display model is very sensitive to residual charge / charge buildup. Maybe if we could dump the on-chip waveform we can fix this? Or we have to experiment [...]

1. In the previous logic, how does epdiy eliminate residual charges?
2. In my test, it seems that the more times `epd_push_pixels(area, white_time, 1)` is called, the more residual charge there is. and using same times`epd_push_pixels(area, black_time, 0)` can neutralize the charge. Why is this? Isn't the voltage of the screen fixed?

lanistor, 2024-05-11:
> Interesting, looks like this particular display model is very sensitive to residual charge / charge buildup. Maybe if we could dump the on-chip waveform we can fix this? Or we have to experiment [...]

I got the knowledge here: https://github.com/vroland/epdiy/issues/226#issuecomment-1895893829 .

vroland, 2024-05-16:
@lanistor I got your displays today, now I'm going to try replicating it when I have the time :) I already tried the display with a V7 board, there it didn't behave so different, but I'll try to reproduce it with a V5 board when I have the opportunity.

vroland, 2024-05-16:
I tried to compile your example repo and I get an error with idf 4.4.5:
```
/home/vroland/dev/misc/epaper-tester/components/lvgl_epaper_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
/home/vroland/dev/misc/epaper-tester/components/lvgl_epaper_drivers/lvgl_helpers.h:48:25: error: 'LV_HOR_RES_MAX' undeclared (first use in this function); did you mean 'LV_HOR_RES'?
```
Any particular version of the dependencies I need?

martinberlin, 2024-05-17:
Probably a different version of LVGL but here I packed a way to get all in a single git clone call:

https://github.com/martinberlin/lv_port_esp32-epaper/wiki/LVGL-9.0

LV_HOR_RES_MAX & LV_HOR_VER_MAX  correspond if I remember well to display Width and Height.

lanistor, 2024-05-19:
> I tried to compile your example repo and I get an error with idf 4.4.5:
>
> ```
> /home/vroland/dev/misc/epaper-tester/components/lvgl_epaper_drivers/lvgl_helpers.c: In function 'lvgl_driver_init': [...]

Have you changed the version of lvgl? The default version is `v8.3`, and it doesn't need the `LV_HOR_RES_MAX `.

If using the following code to start project, it won't need `LV_HOR_RES_MAX`:
```cpp
git clone git@github.com:lanistor/epaper-tester.git
cd epaper-tester
git submodule update --init --recursive
```

vroland, 2024-05-25:
Ok, now it compiles but the screen doesn't update. Maybe I'll have to try with a different board when I have the chance.

lanistor, 2025-12-01:
> [@lanistor](https://github.com/lanistor) if there is any update with this we can keep it open otherwise we can close it since it has no feedback since May 2024

I'll close this issue; it seems to be caused by improper VCOM voltage adjustment.

closed 2025-12-01 by lanistor