vroland/epdiy #300: epd_renderer_init failing on v7 board

bgrassy, 2024-04-24:
Hello! I recently got a v7 board from JLCPCB. I've been able to get things running on the v5 board before, but I'm running into issues with the dragon example project with this new board. I installed IDF version 5.2.1 through the VSCode extension, and was able to successfully build and flash the project to my board. The only change I made in the code was to adjusts the VCOM in `epd_set_vcom`. When monitoring after flashing, I see the following message repeatedly in the logs:

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403754a9
0x403754a9: esp_restart_noos at /home/brock/esp/v5.2.1/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1568
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe3c
load:0x403cc700,len:0x2f88
entry 0x403c992c
I (31) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (31) boot: compile time Apr 24 2024 00:34:02
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : QIO
I (53) boot.esp32s3: SPI Flash Size : 2MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00177000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=85b28h (547624) map
I (192) esp_image: segment 1: paddr=00095b50 vaddr=3fc93300 size=0240ch (  9228) load
I (194) esp_image: segment 2: paddr=00097f64 vaddr=40374000 size=080b4h ( 32948) load
I (205) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=1c15ch (115036) map
I (224) esp_image: segment 4: paddr=000bc184 vaddr=4037c0b4 size=07238h ( 29240) load
I (236) boot: Loaded app from partition at offset 0x10000
I (236) boot: Disabling RNG early entropy source...
I (248) cpu_start: Multicore app
I (248) octal_psram: vendor id    : 0x0d (AP)
I (248) octal_psram: dev id       : 0x02 (generation 3)
I (251) octal_psram: density      : 0x03 (64 Mbit)
I (257) octal_psram: good-die     : 0x01 (Pass)
I (262) octal_psram: Latency      : 0x01 (Fixed)
I (267) octal_psram: VCC          : 0x01 (3V)
I (272) octal_psram: SRF          : 0x01 (Fast Refresh)
I (278) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (284) octal_psram: BurstLen     : 0x01 (32 Byte)
I (290) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (296) octal_psram: DriveStrength: 0x00 (1/1)
I (301) MSPI Timing: PSRAM timing tuning index: 5
I (306) esp_psram: Found 8MB PSRAM device
I (311) esp_psram: Speed: 80MHz
I (605) esp_psram: SPI SRAM memory test OK
I (614) cpu_start: Pro cpu start user code
I (614) cpu_start: cpu freq: 240000000 Hz
I (614) cpu_start: Application information:
I (617) cpu_start: Project name:     dragon_example
I (622) cpu_start: App version:      2.0.0-12-g29b2b19
I (628) cpu_start: Compile time:     Apr 24 2024 00:33:58
I (634) cpu_start: ELF file SHA256:  ec3e906bfdd45e35...
I (640) cpu_start: ESP-IDF:          v5.2.1-dirty
I (646) cpu_start: Min chip rev:     v0.0
I (650) cpu_start: Max chip rev:     v0.99 
I (655) cpu_start: Chip rev:         v0.2
I (660) heap_init: Initializing. RAM available for dynamic allocation:
I (667) heap_init: At 3FC96218 len 000534F8 (333 KiB): RAM
I (673) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (679) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (685) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (692) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (700) spi_flash: detected chip: gd
I (703) spi_flash: flash io: qio
W (707) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (721) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (731) sleep: Configure to isolate all GPIO pins in sleep state
I (738) sleep: Enable automatic switching of GPIO sleep configuration
I (745) main_task: Started on CPU0
I (755) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (755) main_task: Calling app_main()
I (765) epdiy: using resolution 300x832
I (765) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (775) epdiy: pclk freq: 22000000 Hz
I (785) epdiy: line width: 14us, 308 cylces
I (785) epdiy: LCD init done.
I (785) epd: Space used for waveform LUT: 64K

assert failed: 0x4037d39f
0x4037d39f: prvInitialiseNewTask at /home/brock/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:1061 (discriminator 1)

Backtrace: 0x40375546:0x3fc99d20 0x4037be79:0x3fc99d40 0x40381669:0x3fc99d60 0x4037d39f:0x3fc99da0 0x4037e7a2:0x3fc99dd0 0x420074fe:0x3fc99e10 0x420071dd:0x3fc99e60 0x420070ac:0x3fc99e80 0x420070cf:0x3fc99ec0 0x4201b55f:0x3fc99ee0
0x40375546: panic_abort at /home/brock/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
0x4037be79: esp_system_abort at /home/brock/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40381669: __assert_func at /home/brock/esp/v5.2.1/esp-idf/components/newlib/assert.c:40
0x4037d39f: prvInitialiseNewTask at /home/brock/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:1061 (discriminator 1)
0x4037e7a2: xTaskCreatePinnedToCore at /home/brock/esp/v5.2.1/esp-idf/components/freertos/esp_additions/freertos_tasks_c_additions.h:284
0x420074fe: epd_renderer_init at /home/brock/Projects/epaper/epdiy/src/render.c:326
0x420071dd: epd_init at /home/brock/Projects/epaper/epdiy/src/epdiy.c:491
0x420070ac: idf_setup at /home/brock/Projects/epaper/epdiy/examples/dragon/main/main.c:37
0x420070cf: app_main at /home/brock/Projects/epaper/epdiy/examples/dragon/main/main.c:44
0x4201b55f: main_task at /home/brock/esp/v5.2.1/esp-idf/components/freertos/app_startup.c:208
```

The assertion here is failing when calling `epd_renderer_init`: https://github.com/vroland/epdiy/blob/main/src/render.c#L326. As a result, there's nothing displaying on the screen. Has this issue been seen before, and if so are there any recommended mitigations? Thank you!

martinberlin, 2024-04-24:
> I installed IDF version 5.2.1

Never seen this error before but I had an [issue trying with 5.3](https://github.com/vroland/epdiy/issues/285), although very different since it does not build. I strongly recommend you to install 5.1 and maybe on first try just flash the dragon example with IDF.
It seems is dying when trying to initialize the render_thread task didn't see this before.

mickeprag, 2024-04-24:
The crash is due to an assert. Maybe you can see what the assert is checking? That can be a clue to what is going on. 

Also check your sdkconfig that both cpu cores is enabled.

bgrassy, 2024-04-25:
It looks like downgrading to 5.1 works- I was able to build and flash the demo on v6 and v7 boards without any issues after doing so. Thanks for the guidance!

As another note- is there a recommended place to ask general questions about this project beyond Github issues? It looks like the Slack link is broken.

martinberlin, 2024-04-25:
Sure please try https://github.com/vroland/epdiy/discussions
Closing this but I will link it from my 5.3 Issue

closed 2024-04-25 by martinberlin

vroland, 2024-04-30:
@bgrassy There was a bad priority value used in task creation, #307 should fix the issue.