vroland/epdiy #313: please change license to MIT

mars3142, 2024-05-24:
As the title said, I would like to see a MIT license for this package. Because with the current license it can't be used in any commercial products, without open source the complete firmware - I can only speak for ESP32 (don't know other systems).

Are there any points against it?

schuhumi, 2024-05-25:
So, why not build a product with open firmware and hardware? Are there any points against it?

Developing open e-paper drivers is hard enough, because the whole ecosystem is riddled with proprietary mysteries. So why should the people pouring their heart into reverse engineering all these things let you sell their work for profit, and instead of seeing any contributions to this effort all that remains is another locked down disposable blackbox?

If you want to stick to the proprietary side of things, buy/license from the many proprietary solutions out there, also E-Ink themselves have those. If you want to feast on the open ecosystem, become part of the open ecosystem. [There](https://www.tindie.com/products/fasani/epdiy-v7-816-bit-parallel-epaper-controller/) [are](https://www.modos.tech/blog/modos-paper-monitor-pre-launch-on-crowd-supply) [many](https://mntre.com/reform.html) [many](https://www.opulo.io/products/lumenpnp) [many](https://sienci.com/) [others](https://pine64.org/devices/).

vroland, 2024-05-25:
> Are there any points against it?

Aside from keeping the ecosystem open, we already have had the experience of derived boards by LilyGo, who also forked the driver. With a copyleft license, it is way easier to add support for these boards in epdiy than if we need to reverse engineer things.
I also don't see open firmware to be a dealbreaker for most products: Their value proposition is in continued support, backend services, convenience, availability, etc. You could even separate open / proprietary components on separate microprocessors. However, if that is a problem for you specific application, epdiy may not be for you.

mars3142, 2024-05-25:
Okay, you raised valid points. It was just a question and now I know, that epdiy is maybe not for me.

closed 2024-05-25 by mars3142