vroland/epdiy #317: Errors on esp32s3 compile

malachib, 2024-06-02:
Getting a linker errors:

```
/Users/malachi/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/epdiy/libepdiy.a(epd_board_lilygo_t5_47.c.obj):(.literal.epd_board_poweron+0x4): undefined reference to `i2s_gpio_attach'
/Users/malachi/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/epdiy/libepdiy.a(epd_board_lilygo_t5_47.c.obj): in function `epd_board_poweron':
/Users/malachi/Projects/ext/esp/epdiy/src/board/epd_board_lilygo_t5_47.c:113: undefined reference to `i2s_gpio_attach'
```

Seems like a component referencing issue.  Using ESP-IDF 5.1.4

martinberlin, 2024-06-02:
Sorry but we decided not to support https://www.lilygo.cc/products/t5-4-7-inch-e-paper-v2-3 Lilygo ESP32S3 board since it's made based on epdiy version 2 (While epdiy is already in version 7 of the Hardware / Firmware)
It's possible to do it, please check my branch here:
https://github.com/martinberlin/epdiy-rotation/tree/lilygos3 

But we are not going to implement it since it requires adding a lot of pre-compiler IF's and it's really not our fault that Lilygo choose this awkward way to build their PCB. 
For more information about why we took this decision please watch [this Video](https://www.youtube.com/watch?v=r4-dz_x7K4k&t=7s).

closed 2024-06-02 by martinberlin

malachib, 2024-06-03:
I can understand that, given the general support I've received from them.  $40 not well spent, that's for sure.  Thanks for your efforts.  Time to leave them a lousy review

martinberlin, 2024-06-03:
No problem. Still keep in mind like I mentioned you can just use my branch in the epdiy fork mentioned and build stuff with IDF.
But yes I agree it’s not a very useful board in its current state and the display failed for me having that hardware dead lines…