vroland/epdiy #318: The `ED060XC3` waveform on s3 cann't render 060XCD, 060XCH clearly with MODE_DU mode.

lanistor, 2024-06-05:
Always using the `ED060XC3` waveform with v5 board, to paint `060XCD`, `060XCH` and `060XD4` with  `MODE_DU` mode (faster), It can paint clearly.
But when changing to s3 board, It's color become light, seems half dark than v5 board. When changing the waveform to `ED047TC2`, it become clearly, but `ED047TC2` is really slow.

Is this can be fixed with some simple adjust? Or must create another waveform?

Here is the paint result:
**ED060XC3 waveform**  
[image]

**ED047TC2 waveform** or **v5 board**  
[image]

vroland, 2024-06-21:
I guess it can be improved by hacking the builtin epdiy waveform to have more cycles pushing the pixels in the respsective direction. Or using epd_push_pixels() a number of times.

lanistor, 2024-06-22:
> I guess it can be improved by hacking the builtin epdiy waveform to have more cycles pushing the pixels in the respsective direction. Or using epd_push_pixels() a number of times.

Yeah, i guess so, and i don't have the ability to create waveforms yet.

martinberlin, 2024-11-08:
@lanistor  please give a good read to https://github.com/vroland/epdiy/issues/226 because this is open since June and I don't see any advancement on it. So either we take action or this issue will be close end of December.
I would like to keep open issues "waiting for feedback" in a maximum wait of 2 months if there is no follow-ups then it will be closed. Because it does not make any contribution to a project just to open issues and leave it there to get stale. There is either **proactive action** or it get's closed

lanistor, 2024-11-11:
I will close it, s3 board works fine with `ED047TC2` waveform