vroland/epdiy #319: The s3 board cannot adjust vcom automaticlly with some XD4 screen.

lanistor, 2024-06-05:
The s3 board can adjust vcom automaticlly depend on screen model, so you don't need to adjust the variable resistance by hand, it's really cool, and i tested it can adjust vcom of `060XCD`, `060XCH` and give them a perfect painting result (in v5 board you must adjust vcom).

But on `060XD4`, 40% of the screen can be rendered perfectly, the other `60%`, the screen will be lighter the expected, and it seems that there is no way to adjust it?

It this can be resolved by other way?

martinberlin, 2024-06-05:
You can adjust it using what you can see in the demos:
https://github.com/vroland/epdiy/blob/main/examples/dragon/main/main.c#L38
epd_set_vcom(1560); // That will be -1.5V

> But on 060XD4, 40% of the screen can be rendered perfectly, the other 60%, the screen will be lighter the expected.

I really do not think that this a VCOM related issue.

lanistor, 2024-06-06:
> You can adjust it using what you can see in the demos: https://github.com/vroland/epdiy/blob/main/examples/dragon/main/main.c#L38 epd_set_vcom(1560); // That will be -1.5V
>
>
> I really do not think [...]

Got it, thanks.

closed 2024-06-06 by lanistor