vroland/epdiy #320: Will epd_push_pixels_lcd in s3 support partial area?

lanistor, 2024-06-08:
`epd_push_pixels` will always paint full screen on s3 board, for the macro `RENDER_METHOD_LCD`.
What should i to do, for partial area clear?

vroland, 2024-06-21:
Ah indeed. Probably it would need pre-population of a masked line buffer for the affected lines, and then orchestrate the callbacks appropriately.

lanistor, 2024-06-22:
> Ah indeed. Probably it would need pre-population of a masked line buffer for the affected lines, and then orchestrate the callbacks appropriately.

It this need to modify the hardware circuit board? or only modify the software code?

vroland, 2024-08-05:
Only software modifications. Shouldn't be too crazy.

lanistor, 2024-08-06:
Waiting for your update👍.

> Only software modifications. Shouldn't be too crazy.

lanistor, 2024-09-26:
@vroland Is this problem in plan?

lanistor, 2024-10-14:
@vroland Is there any i can do to support it? May you guide me to do it?

vroland, 2024-10-15:
Hey, can you test if #364 fixes it for you? I finally found some time to just do it, it's not a lot of work at the end of the day.

lanistor, 2024-10-16:
> @lanistor hey, can you test if #364 fixes it for you? I finally found some time to just do it, it's not a lot of work at the end of the day.

OK, i will test it soon, thanks a lot👍👍👍

lanistor, 2024-10-17:
@vroland I has tested it (https://github.com/vroland/epdiy/pull/364), it can works perfect! Thanks a lot. 👍👍👍

PR #364 (2024-11-03): Implement area clearing on the ESP32-S3
Makes `epd_push_pixels` respect the given area on ESP32-S3 chips. This should fix #320 .
commit d12021ef (2024-11-03): Implement area clearing on the ESP32-S3

Makes `epd_push_pixels` respect the given area on ESP32-S3 chips. This should fix #320 .
closed 2024-11-03 by vroland