vroland/epdiy #321: kicad schematics

SpandexWizard, 2024-06-09:
is it possible to get ahold of a schematic file so i can make large edits to the board? my usecase involves a 50pin connector, and possibly io expanders, and i'd like to change the formfactor entirely. i can rebuild it from the pdf probably but it'd be nice if i didnt have to.

as an aside how hard would it be to get the code to run on an orange pi zero 2w instead of an esp32? i'd rather have as few devices draining power as possible, but i understand it's going to be something of an undertaking. i'm just not sure how big.

martinberlin, 2024-06-09:
Sure please pull this repository 
https://github.com/vroland/epdiy-hardware Moving your Issues there later today

> orange pi zero 2w instead of an esp32? i'd rather have as few devices draining power

1st that would be a monumental work
2nd the Pi won’t have enough speed to run this. But if you manage to do it I would love to see the results!

SpandexWizard, 2024-06-09:
ooh, awesome!

why wouldnt the pi have enough speed? an orange pi zero 2w operates at 1.5ghz. an esp32 operates at, if i'm reading this right, 240mhz. the orange pi should blow it out of the water.

mickeprag, 2024-06-09:
If you run inside the kernel you may be able to use all speed. In userspace the scheduler would make the code both slower and more unpredictable.

mickeprag, 2024-06-09:
Also, in esp32 the actual bitbanging is done using hardware blocks (LCD and i2s) and the code only forwards the data. This way esp32 may be able to push the speed even more than normally.

martinberlin, 2024-06-09:
I don’t think that there is currently any PI solution to control parallel panels.
I also agree with Michael that having a PC that goes at that speed does not mean you can do parallel at that fast rate.
You should explore Rockchip since they do have a bird that is design to control parallel with their own dedicated chip. Or also Waveshare has IT8951*** board so you could talk SPI with them albeit is closed source the format inside the chip and you cannot switch Eink displays at least with the ones I’ve tried.
Passing this to Discussions since it’s not really a software Issue but maybe helps someone.