vroland/epdiy #325: Implement Wokwi testing

martinberlin, 2024-06-18:
Wokwi is a simulator. The idea is that we could be able to simulate a v7 board, including it's main two chips:

- PCA9535 , I2C 16 bit IO expander, manages eink slow signals
- TPS65185, also talks I2C, generates the needed voltages to drive the pixels (+15 and -15V) plus to open the FET gates that let pass whose (higher ones + VCOM)

First step is to simulate this two as "[Custom chips](https://blog.wokwi.com/custom-chips-cd4051b-analog-multiplexer/)" and attach them to an ESP32S3-DEV board. Then we will simulate the existing v7 board attaching this 2 chips in the I2C line and some LEDs as data output. 

If everything turns out as expected, this will simulate that the voltages are on (wokwi does not know about voltages), and we should see how the framebuffer signals are outputted on the data LEDs.
I propose to use different colors:
- Red: Fast signals
- Orange: Slow signals
- Blue: Data in pins

@mickeprag if you are interested on this also you can help out in any way. 
This will take many iterations. The end result would be to truly simulate a v7 board, adding an SVG drawing, and the custom chips inside. At that point it will look and behave as a real board and the idea is that we could try to put this also in the Github CI, so apart of running the tests, it will also try to run the generated Firmware in our "virtual board"

mickeprag, 2024-06-18:
This sounds cool. I am on. Although I have almost zero experience from wokwi...

martinberlin, 2024-06-19:
> This sounds cool. I am on. Although I have almost zero experience from wokwi...

Thanks for your interest. Same here. I'm still figuring out the very basics. For example [here](https://wokwi.com/projects/401153784260252673) I don't have any idea why IO8 does not turn HIGH. Note I'm lazy to put a resistor to each LED but is not that for sure!

mickeprag, 2024-06-19:
Maybe because IO8 is internally connected to the flash?

martinberlin, 2024-06-20:
Ok it was me that I was connecting the wrong pins. The problem with most computer related thing is ourselves except on some rare moments ;)
https://wokwi.com/projects/401153784260252673 -> And next blocker is... that I can't seem to tell the Simulator that the MCU has external RAM. Why I don't know, I'm adding the attribute:

```json
{ "type": "board-esp32-s3-devkitc-1", "id": "esp", "top": -0.18, "left": -5.03, 
    "attrs": { "psramSize": "8", "flashSize": "4" }
```

martinberlin, 2024-06-24:
Blocker solved thanks to a hint of Wokwi team: 

```json
{
    "type": "board-esp32-s3-devkitc-1",
    "id": "esp",
    "top": -0.18,
    "left": -5.03,
    "attrs": {
        "psramSize": "16",
        "flashSize": "4"
    }
}
```

@vroland and @mickeprag PSRAM is detected now. What remains is to make the 2 mentioned custom chips otherwise the firmware will hang without being able to communicate with the IO expander or the TI Power mic.

martinberlin, 2024-06-25:
Implemented first time custom chip and it seems the CI action works however I still fail to link it on VSCode diagram.json
https://github.com/martinberlin/wokwi-pca9535

martinberlin, 2024-06-26:
@vroland I will need the behaviour of "simulated chip" PCA9535 to be reviewed. What I'm getting in the VSCode console only placing this chip with the pins connected as in v7 Schematic is this:

```
[chip-pca9535] Input mask is now 0xc007
[chip-pca9535] Input mask is now 0x3
[chip-pca9535] Input mask is now 0x2103
[chip-pca9535] Input mask is now 0x2903
[chip-pca9535] Input mask is now 0x3903
[chip-pca9535] Read: reset INT flag
```

Custom chips are not really easy to use. The fastest way that I've found is to make a repository, add some CI, so it get's compiled (output is Web assembly or WASM), and that's how it's loaded into Wokwi. It's a bit of black magic for me, a bit hard to understand, the internals of how all this works together. Hence I need a bit of a review and some help making custom chip TPS65185. On that topic if I understand that right:

To simulate the power Mic I just need to hear on 0x68 address. And when it get's the TPS_REG_ENABLE I2C signal activates the voltages, pulling TPS_PWR_GOOD low. Is that correct?

martinberlin, 2024-07-22:
Advancing very slowly here. In the next days I'm adding the wokwi-tps65185 custom chip.
What it will do really is just to receive the signal to power up the grid and just turn hi a GPIO. It won't really generate any voltages since wokwi does not know anything about volts ;)

vroland, 2024-08-05:
No problem, things are slow during the summer ;) Curious where this goes.

martinberlin, 2024-08-17:
Finished to add new chip:
https://wokwi.com/projects/401153784260252673

Chip itself is in this repository: https://github.com/martinberlin/wokwi-tps65185/
This is just a small wrapper to "emulate" a reply with PWR_GOOD signal since Wokwi does not know about voltages for Eink :electric_plug: 
Let's see if I can test it in the next days. Would be cool if someone helps to verify if all connections / signals match the schematic.

vroland, 2024-08-26:
Sounds promising! Did it compile in the end?

martinberlin, 2024-08-26:
No. There must be something in my tps65185 chip that does not like. But I must say that at this point is a bit weird because it can take forever, instead of prompting some kind of useful error message.
My first thoughts where "Ok this is great to make simple simulations" but when going deeper then unexpected things happen, without proper guidance to the developer trying to implement it. Hopefully I'm wrong and it just deserves some more rounds of fight. But I guess you understand what I'm trying to say.

martinberlin, 2025-01-06:
Closing here I tried my best but could not make the TPS65185 simulation work. Might take the idea again when I recover forces from last crash

closed 2025-01-06 by martinberlin