vroland/epdiy #346: The V5 and V7 versions drive different colors for the same display

zengcym, 2024-08-22:
I have a problem, I use v5 and v7 versions to drive ED060KD1 display, waveform uses epdiy_ED060XC3, v5 is normal, v7 is obviously weak, I have checked VCOM and other voltages, it is consistent. What causes this?Can you give me some help, thank you。
[image]
[image]

zengcym, 2024-08-26:
Hello Martin, I will read this carefully, thank you for your great work, and look forward to fixing this bug.

martinberlin, 2024-10-07:
Hello @zengcym 
Additionally in case it's of your interest I've started writing an article in the epdiy WiKi that covers with more detail how the eink panels are driven with V7. As soon as I research more, will add code snippets and examples there:

https://github.com/vroland/epdiy/wiki/How-pixels-are-driven-in-a-parallel-epaper-with-epdiy

zengcym, 2024-10-12:
Hello Martin, I'll be following the content of the page, thank you for your work!

vroland, 2024-10-15:
For a workaround, you can try some vendor waveforms like the ED047TC2 waveform and see how it does on your display.

martinberlin, 2024-11-08:
@zengcym please give feedback about this issue or close it if you are not more interested in finding a solution for it

zengcym, 2024-11-12:
I'm sorry, but I came late, and the actual display refresh will be slow when I test it with a ED047TC2 waveform file, and it will gradually return to normal from dark color in the process。

martinberlin, 2025-04-15:
@vroland my wish in this matters is that one day we can enable a way to edit matrix transitions in a human way, just similar to what is being done in FastEPD.
Then you can edit this by hand or even make a Serial Console “talking” client to test your GRAY1 to GRAY2 transitions.
Probably only 2% of the people will use, but still there is a chance to modify a custom waveform, while at the moment is really hard to do so.

closed 2025-12-25 by martinberlin