vroland/epdiy #355: The Readme states ED060SC4 is compatible but it isn't

floers, 2024-09-03:
The [Displays](https://github.com/vroland/epdiy?tab=readme-ov-file#displays) section states that ED060SC4 is compatible since version 2. My understanding was that this would mean it is working with the latest version of epdiy but v7 has no 39 pin port so it can't be compatible and the only adapter documented is specifcally not compatible for ED060SC4.

martinberlin, 2024-09-03:
Hello Florian the table does not really specify if it has the connector that is the issue.
It’s impossible to put all connectors in the same PCB. Before sending it to fabrication or buying a unit you have to always check what connector it has or design an adapter (there are some in that hardware repository)

mickeprag, 2024-09-03:
Maybe we can make a separate table in the readme that lists the different connector on the boards?

martinberlin, 2024-09-04:
Yes that’s a good idea. Because is confusing at least with this 39 pin connector since there are two (39 for 6”, 39 for 13.3”, different layout and pins)

martinberlin, 2024-09-10:
@vroland maybe we could come up with some better idea to hold the "supported displays" table.
Or even to not show board version, just add a note in top: 

**Check the hardware repository before ordering display to make sure the PCB has the right connector.**

martinberlin, 2024-09-16:
Also @floers we should state what compatible is:
**For this repository is that it can drive the display.**
A very different thema is if it has the connector. Because if it does not you can also check if there is an adapter for it. And if there is not, one can be designed, I did this myself for example to get a 6" working from the 40 pin FPC.

vroland, 2024-09-24:
Indeed, we should make this more explicit in the readme :+1:

martinberlin, 2024-11-08:
Closing here I will make a PR this weekend making more "explicit" what the readme states
Supported in vX means that the PCB supports the display but you need to make sure that has the right connector to handle it. Is impossible to cover in a single PCB all possible connectors for dozens of different displays

closed 2024-11-08 by martinberlin