vroland/epdiy #356: Crashed when used in conjunction of NimBLE-Arduino + Wi-Fi

rascove, 2024-09-08:
Hi, I'm working on a PlatformIO project that requires the ESP32 (LilyGo T5 4.7) to subscribe and publish to an MQTT broker using Wi-Fi and scan nearby BLE devices, while displaying some data on the e-Paper. I'm using [NimBLE-Arduino](https://github.com/h2zero/nimble-arduino) and built-in Wi-Fi libraries in conjunction with the epdiy.

If I use only two of these libraries (epdiy + Wi-Fi, epdiy + NimBLE-Arduino, or NimBLE-Arduino + Wi-Fi), the project can be successfully executed on the device, however as soon as I added all three libraries at the same time, the device keeps on rebooting. These are the source code and the configuration file for your reference:

`#include <Arduino.h>
#include <Epd_Highlevel.h>
#include <Epdiy.h>
#include <FiraSans.h>
#include <NimBLEDevice.h>
#include <NimBLEAdvertisedDevice.h>
#include <WiFi.h>

#define EPD_WIDTH 960
#define EPD_HEIGHT 540
#define WIFI_SSID "SSID"
#define WIFI_PASSWORD "PWD"

uint8_t* fb;
EpdiyHighlevelState hl;
WiFiClient wifi;
NimBLEScan* scanner;
enum EpdDrawError err;

void setup()
{
	Serial.begin(115200);
	NimBLEDevice::init("");

	scanner = NimBLEDevice::getScan();

	scanner->setActiveScan(true);
	Serial.print("Connecting to ");
	Serial.println(WIFI_SSID);
	WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

	while (WiFi.status() != WL_CONNECTED)
		delay(500);

	Serial.print("WiFi connected, IP address: ");
	Serial.println(WiFi.localIP());
	epd_init(&epd_board_lilygo_t5_47, &ED047TC1, EPD_LUT_64K);

	hl = epd_hl_init(EPD_BUILTIN_WAVEFORM);

	epd_set_rotation(EPD_ROT_INVERTED_PORTRAIT);

	fb = epd_hl_get_framebuffer(&hl);
	int cursor = 0;

	epd_clear();
	epd_write_default(&FiraSans, "Initialising...", &cursor, &cursor, fb);
	epd_poweron();
	epd_hl_update_screen(&hl, MODE_GC16, 27);
	delay(500);
	epd_poweroff();

	NimBLEScanResults foundDevices = scanner->start(10);

	Serial.print("Devices found: ");
	Serial.println(foundDevices.getCount());
	Serial.println("Scan done!");
}

void loop()
{
}`

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600
monitor_speed = 115200
lib_deps =
    https://github.com/vroland/epdiy.git
    h2zero/NimBLE-Arduino
build_flags =
    -DBOARD_HAS_PSRAM
    -DCONFIG_EPD_DISPLAY_TYPE_ED047TC1
    -DCONFIG_EPD_BOARD_REVISION_LILYGO_T5_47
    -DCONFIG_ESP32_WIFI_SW_COEXIST_ENABLE`

Thank you very much in advance.

martinberlin, 2024-09-10:
Hi @rascove 

> I'm working on a PlatformIO project that requires the ESP32 (LilyGo T5 4.7)

You will be much better off writing an issue in [their Repository](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/). Because we do not produce this Hardware and not even their code which is based on epdiy v2 ( And as you can see here the PCB is already version 7 )

You could try the same code in a different hardware maybe. And also make some tests with the latest version of the Firmware of this repository.

closed 2024-09-10 by martinberlin

rascove, 2024-09-10:
Thanks for the feedback @martinberlin , I'll do that as well.

I'm posting this here because the error is caused when the `EpdiyHighlevelState` is being initialised, as shown in the serial log: `assert failed: epd_hl_init highlevel.c:50 (state.dirty_columns != NULL)`.

Anyway, thank you very much and have a great day ahead.

martinberlin, 2024-09-10:
Sure, but you are using a library that is from Lilygo and since long time updated here.
If you switch to this repository with the latest library then esp32 version of Lilygo is still supported. (Not S3)
Anyways buying a product from Lilygo is the hardware provider who should give support.

rascove, 2024-09-11:
Hi @martinberlin , thanks for the response but please refer to my code and PlatformIO configuration. I'm not using LilyGo T5 library at all, in fact, I'm using [this repository's own example](https://github.com/vroland/epdiy/tree/main/examples/lilygo-t5-47-epd-platformio) to build the project.

martinberlin, 2024-09-11:
Don't really understand why then: 

#include <Epd_Highlevel.h>
#include <Epdiy.h>

Use first letter uppercase. The example code in this repositories are not like this and it won't find the include if you uppercase first letter.
I will try when I find sometime your code, but I need to find my Lilygo first. I highly recommend you to switching to a v7 board though, to use the latest code using LCD Module. I've still one v7 unit available in tindie.com if you search for "epdiy"

rascove, 2024-09-11:
Ah, I'm so sorry about that. I'm using Windows so the filename is not case-sensitive, and coming from Java background, it's my habit to capitalise the first letter of the library.

But thank you very much for your kind assistance, @martinberlin , I really appreciate that. Unfortunately I don't have a control for the type of e-Paper used as it was purchased by the client.

martinberlin, 2024-09-11:
Also this line of the assert where it stops:

    state.dirty_columns
        = heap_caps_aligned_alloc(16, epd_width() / 2, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);

Creates the RAM space. If it's not allocated then the assert will fail. Try to locate it in PSRAM, but that will be slower, and not sure if it will work. It might be that using this combination of libraries won't work with the hardware.
That's why I was suggesting to update to v7. You will need to deep debug this and check how much Heap RAM you have available, I guess this could be the most probable cause.

rascove, 2024-09-14:
Hi @martinberlin ,

Sorry for the late response and thank you very much for your advice about the insufficient RAM. It gave me the idea to initialise the BLE stack in the PSRAM instead, which is already provided by the NimBLE-Arduino library. And now, all three libraries can be used altogether.

Once again, thank you very much and have a great day ahead.