vroland/epdiy #366: Will 7 colors epaper be supported in S3 board? Such as  EC060KH3, EC078KH5?

lanistor, 2024-10-21:
It would be really cool if such a color screen could be driven.

[image]

This is a color implementation of the filter. The driving logic should be the same as the 16-grayscale black and white screen, but it requires a color algorithm converter. 
The interface is also the same as the `060XCH`, and as far as I know, the rendering speed is the same as the black and white screen.

martinberlin, 2024-10-21:
I made a humble approach for the Kaleido 6” display here
https://github.com/vroland/epdiy/tree/s3_color_implementation
you can read more details in the wiki
[image]
[image]

lanistor, 2024-10-22:
@martinberlin This is really cool, I'll try it right away.

martinberlin, 2024-11-08:
Great @lanistor  please give some feedback in the next weeks of your experience trying this.
I've made this branch mostly as a personal experimentation. Probably there are wiser ways to deal with color, just write about your feedback please.

lanistor, 2024-11-11:
> Great @lanistor please give some feedback in the next weeks of your experience trying this. I've made this branch mostly as a personal experimentation. Probably there are wiser ways to deal with color, interested to know how it was the experience for you.

Thanks, i tested it for once, but didn't found a demo to rendering with 7 colors directly, i will test it few days later, for too much development work recently.

martinberlin, 2024-11-11:
You can just try the www-image example.
This should download a JPG and call my clone of the función epd_draw_pixel then for every pixel will be aware of what filter is on top

lanistor, 2024-11-11:
> You can just try the www-image example. This should download a JPG and call my clone of the función epd_draw_pixel then for every pixel will be aware of what filter is on top

Got it, thanks 👍

martinberlin, 2025-02-20:
This has been way to long without any activity. You can find it converted to a discussion into General section