vroland/epdiy #367: Boot / reset buttons in v7 board?

BackSlasher, 2024-10-27:
I'm aware it's probably a stupid question, but I'm afraid of burning something
[image]
I got the board I ordered, and trying to follow the [getting started](https://epdiy.readthedocs.io/en/latest/getting_started.html) guide. 
I got as far as:
> Hold down the `BOOT` button on your board, while quickly pressing the `RESET` button. 
I'm unsure how to do that on tis board, as it doesn't seem to have buttons.
Is shorting the `B+/-` holes (lower left) an equivalent for pressing the boot button?

Thank you for your patience

martinberlin, 2024-10-27:
There are no Boot or Reset buttons in v7.  You actually do not need them since it has an UART chip that should handle it for you.   If it does not flash or does not get in download more there is another thing, but is not because of that.
The getting started guide seems not updated to current version since this v7 uses ESP32S3 and there it mentions the WROVER module.

BackSlasher, 2024-10-27:
Thanks for the info. Could you give me a hint on how I go about flushing code into the v7?

martinberlin, 2024-10-27:
Sorry but not with this information at least make a detailed description of what’s going on and why you can’t flash your board

BackSlasher, 2024-10-28:
Understood. I'm at the very beginning - I connected the device I got from the factory to my laptop over USB-C, and I see `/dev/ttyUSB0`.
My result as I'm trying to flush:
```
/tmp/epdiy/examples/demo$ idf.py flash -b 921600
Executing action: flash
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory /tmp/epdiy/examples/demo/build
Executing "ninja flash"...
[1/5] cd /tmp/epdiy/examples/demo/build/esp-idf/esptool_py && /home/nitz/.espressi...ld/partition_table/partition-table.bin /tmp/epdiy/examples/demo/build/firmware.bin
firmware.bin binary size 0x10abe0 bytes. Smallest app partition is 0x177000 bytes. 0x6c420 bytes (29%) free.
[1/1] cd /tmp/epdiy/examples/demo/build/bootloader/esp-idf/esptool_py && /home/nit... 0x8000 bootloader 0x1000 /tmp/epdiy/examples/demo/build/bootloader/bootloader.bin
Bootloader binary size 0x6f50 bytes. 0xb0 bytes (1%) free.
[4/5] cd /opt/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/esp...iy/examples/demo/build -P /opt/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32 -p /dev/ttyUSB0 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 firmware.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/nitz/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/opt/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed.

FAILED: CMakeFiles/flash /tmp/epdiy/examples/demo/build/CMakeFiles/flash 
cd /opt/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/esp-idf -D "SERIAL_TOOL=/home/nitz/.espressif/python_env/idf5.3_py3.12_env/bin/python;;/opt/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/tmp/epdiy/examples/demo/build -P /opt/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /tmp/epdiy/examples/demo/build/log/idf_py_stderr_output_3001064 and /tmp/epdiy/examples/demo/build/log/idf_py_stdout_output_3001064
```
Am I running the wrong command? Missing some way to put the device in flush mode?

mickeprag, 2024-10-28:
It seem you are able to connect to is and responds as it should.
`This chip is ESP32-S3 not ESP32. Wrong --chip argument?`

Basically you are building for the wrong ESP32 chip. Check your build settings in your project.

BackSlasher, 2024-10-28:
Is that in `idf.py menuconfig`? If so, which setting should I change there?
The project I'm trying to flush is `examples/demo` in this repo

BackSlasher, 2024-10-28:
I was missing `idf.py set-target ESP32-S3`. I still need to figure out how to set the display type, but I'll try it by myself.
Will reopen if I get stuck.
Thank you for your patience.

closed 2024-10-28 by BackSlasher

BackSlasher, 2024-10-28:
Things work perfectly.