vroland/epdiy #370: Add support for ED052TC4 (5.2" 720×1280)

jonasschnelli, 2024-11-05:
`ED052TC4` is a neat and affordable (10-20$) 5.2" high resolution screen with 720×1280 (WXGA) with 283 PPI similar to the one used in the LilyGo T5 but with hight resolution.

It uses a 50pin `WP27D-P050VA3` connector (see images).
Not sure if an adaptor board for V7 would be possible.

[image]
[image]

https://www.panelook.com/ED052TC4_E_Ink_5.2_EPD_parameter_48664.html

martinberlin, 2024-11-05:
My 0.2 cents about this: If Lilygo has in fact the same connector then before planning any hardware adaptation it should be tested with that device.
@martinberlin has really not much interest to add this AXE connector to a epdiy v7 clone so someone else should give it a try.

martinberlin, 2024-11-08:
@jonasschnelli please give some feedback in the next weeks about what is the plan with this display.
Requests of displays that are not supported need some effort of the person wanting to make this work.

jonasschnelli, 2024-11-08:
@martinberlin My skills in PCB/electronics are limited but I'm happy to contribute (provide screen or work on the firmware part)

martinberlin, 2024-11-08:
I ordered one of this displays already. But like I mentioned there is existing hardware to try this (That is not super expensive) so you might also get it and try it out first. Just to check if this behaves the way you want. 
I've already some 40 pin to 50 AXE connector adapters that I've designed some time ago and will check this display when it arrives and write some feedback about it.

My skills on electronics are also limited. But I just want to give the message that everyone wanting to drive a new display needs to put his effort and time on it to make it happen. As you know there are hundreds of different panels out there and it's impossible to cover all of them even having the advantage that most parallel have the same driving principle (But as you know, different connectors, timings and so on)

martinberlin, 2024-11-09:
Hello @jonasschnelli  if you get it earlier than me can you please post a Zoomed photo of the connector?
The datasheet is here https://www.data-modul.com/sites/default/files/products/VB3300-EBA-specification-12053269.pdf
And all pictures are quite blurry so it's hard to see that detail. I'm not sure if [C3647224](https://jlcpcb.com/partdetail/JaeElectronics-WP27D_P050VA3R15000/C3647224) is the right female connector where this can be plugged.

martinberlin, 2024-11-09:
I've designed an adapter for this, still needs time, since I've to wait till display comes and test it. If it works I will share it in the hardware repository
[image]

jonasschnelli, 2024-11-20:
[image]
[image]

jonasschnelli, 2024-11-20:
I reviewed the new adapter schema and have some observations (with my limited understanding):

### Unconnected Pins
* **SPI Pins (1.8V PIN5)**: All SPI pins are unconnected. Purpose of flash memory unclear.
* **TP_1V8 (PIN29)**: Not connected - likely negligible since there's no touch panel
* **Thermistor Connections**:
  * TS (PIN37) - Thermistor sensor not connected
  * TS_GND - Thermistor analog ground not connected
  * Question: What's the purpose of these thermistor pins?

### Connector Verification
* Panel uses: JAE WP27D-P050 VA3
* New adapter: WP27D-S050 VA3
* Note: R15000 designation's significance unknown (likely not relevant)

All other connections appear to match specifications.

martinberlin, 2024-11-21:
P is Plug and S correspond to Receptacle (in PCB)
Waiting for the part to arrive to JLCPCB

martinberlin, 2025-02-11:
V7 raw had a few errors and I still don’t know why PWRGOOD pin does not get high. But it works. @vroland left a branch: board/v7-raw for you to test it.
If you like the idea I send you a PCB

https://youtu.be/quSVbgvOYXM?si=UyFBScEK1o5a5Do5

martinberlin, 2025-02-20:
This is already working converting it to a discussion