vroland/epdiy #371: The timing diagram for driving the 9-inch A090XE01 V6 ink screen is as follows

jysrfeng, 2024-11-07:
[image]
[image]
Where do I need to modify? Thank you.

martinberlin, 2024-11-07:
Is possible that you paste the full datasheet?
Is very hard to help you just looking at those 3 pages. Also what version of epdiy are you using?
Please give all possible details to reproduce what you are experimenting and what is the problem to drive this display. You are also driving a display that is not supported so it's actually your part of the job to research how epdiy works.

jysrfeng, 2024-11-07:
[A090XE01 V6 Spec.pdf](https://github.com/user-attachments/files/17661853/A090XE01.V6.Spec.pdf)
Yes, this is a very difficult thing. Take it slow, I hope for a good outcome.

martinberlin, 2024-11-07:
@jysrfeng do you know what is the part MFR number of that 50 pin connector?

jysrfeng, 2024-11-07:
Any FPC socket with a spacing of 0.5mm can be connected to it

martinberlin, 2024-11-08:
That really doesn’t reply to my question. Please research what is the specific part that gets connected with this display and report the part number

jysrfeng, 2024-11-09:
The hardware cannot be determined yet. Now I want to adjust the timing first, and then adjust the hardware. After all, the specifications are available, but the schematic diagram is not. We have to take it slow. thank you
[image]

jysrfeng, 2024-11-09:
This is the result of randomly selecting a modification to 1024x768 and adding plus or minus ± 20 plus or minus ± 15. I think we should adjust the timing first. But I won't do it.

martinberlin, 2024-11-09:
I was thinking that if you can find me the exact part number Incan help you designing an adapter:
V7 output -> 40 pins to the 50 pins connector 

It’s very probably that it works just out of the box with a standard waveform as ED097TC2

jysrfeng, 2024-11-09:
I only have one manual, please pay attention to the circuit diagram inside. The key is that the timing diagram needs to be modified in order to be suitable for AUO's EPD. How do you do it, how do I wire it. We are currently using FPC-05FB-50PH20 for reference only, it is not important.

martinberlin, 2024-11-09:
Thanks @jysrfeng 
[FPC-05FB-50PH20](https://jlcpcb.com/partdetail/Xunpu-FPC_05FB50PH20/C2856838) seems to be available in JLC parts. I was checking **1. Panel Pin Assignment** and I must admit I'm quite lost.
3 VCOM_BOT  seems to be VCOM Input setting voltage
(Does this display set his own VCOM? )
7 RST_N is unknown to me, we don't use this pin for the other supported displays (AFAIK)
8 SHD_N same with this one
9 PWR_RDY same, this is an output signal but for the voltages PMIC TPS65185, I'm not aware that any other display has this
[image]

I'm must say that I'm completely lost with this display it seems to be completely different than the ones that epdiy supports (At least with the signals I see in the datasheet)
Is parallel for sure, but it seems to have a completely different setup to handle voltages, and I'm unsure even where to start with it.

jysrfeng, 2024-11-09:
3 VCOM_BOT (Does this display set its own VCOM?) I'm not sure either.
7 RST_N is the Global reset pin. Low to reset. I directly connected it to 3.3v.
8 SHD_N is the DC-DC converter shut down pin. I directly connected it to 3.3v.
9 PWR_RDY is the Power ready output. I left it floating.

VCOM only affects the display effect. For now, the priority is to get it to display something. Therefore, I need to rewrite the program according to AUO's timing diagram.

This is what I hope you can help me with. Don't expect it to be compatible with Eink. Sorry for any inconvenience. If you're too busy, it's fine. I'll learn and do it step by step. After all, this is just a small hobby for me.

I've tested with four voltages (positive and negative), and there is display but it's messy.

Thank you very much.

Thanks!!!

martinberlin, 2024-11-09:
Awesome. I will try to check this of the timings and maybe also @vroland can help to point out where it is.
I like that people try to drive new panels is always an interesting but time taking experience. Best of the luck!!!

jysrfeng, 2024-11-09:
Come on, everyone! Let's work hard together!
I'm thinking of buying an AUO EPD reader to try out the circuit.

jysrfeng, 2024-11-15:
[image]

martinberlin, 2025-03-11:
Any luck @jysrfeng did you get it working?

jysrfeng, 2025-03-11:
Thank you for remembering this. I haven't made any progress yet, but I'll do it again when I get the chance. If there are any signs, they will be published here. thank you

谢谢你还记得这件事。还没有进展，不过有机会了还会折腾的。如果有迹象了，一定会在这里发布的。谢谢！

martinberlin, 2026-10-16:
Dear @jysrfeng is there any progress on this one?