vroland/epdiy #381: Adding Lilygo new S3 board

martinberlin, 2024-12-06:
This one finally is very similar to epdiy v7 so support should work without any problem.
Please refer to this issue: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/93
[Schematics](https://github.com/user-attachments/files/18019780/T-EPD47-PRO.V1.0.pdf)

bmorcelli, 2025-01-16:
I have this board (Lilygo T5 E-Paper S3 Pro), and their ["official" library](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) isn't updated yet.. 

the version that is working is from the [device's repo](https://github.com/Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO/tree/master/lib/LilyGo-EPD47-esp32s3)

I copied it into my fork to mess with it and develop my firmware over it [here](https://github.com/bmorcelli/LilyGo-EPD47)

The main difference is in the /src/utilities.h and src/ed047tc1.h

I'm looking forward to make it work with lvgl 9.2 (got my first "Hello World" for this display today.. tomorrow I might have it running better), but I want to move to this Library, so I can use it on more devices

jslawek, 2025-02-01:
I think you should clearly outline the various versions of Lilygo T5 E-Paper S3, and the fact that EPDIY will only support the upcoming hardware that is not released yet as of today - as this might be really confusing. So there is the [T5 e-Paper S3](https://lilygo.cc/products/t5-4-7-inch-e-paper-v2-3) version 2.3 or 2.4 (depending on options, model number H716, H578 or H580), and also there is the new [T5 e-Paper S3 Pro](https://lilygo.cc/products/t5-e-paper-s3-pro) (model number H752). As far as I understand, in terms of EPD driving the "Pro" and "non-pro" (v2.3, 2.4) versions differ just in the pinout, but share the same issue: shift register on the way that prevents using the latest S3 LCD driver directly. So neither the v2.3/2.4 nor current "Pro" will be supported in EPDIY(?), we will have to wait for Lilygo to release the new hardware. Please correct me if I'm wrong.

martinberlin, 2025-02-01:
Thanks for the request. I clearly stated in this PR

> Please Note: This board is still under development. I leave this here so the tech Team of Lilygo tests their new PCB.

It’s not my fault that there are several different versions of the Lilygo S3. But I really tried to state that this new version of Lilygo is still under development and I left this branch there ready for them to test.

closed 2025-02-03 by martinberlin

redstoneleo, 2026-08-19:
I have successfully made an EPDiy patch for this board
https://github.com/redstoneleo/EPDiy-T5S3