vroland/epdiy #391: Row  update is limited to 2000 pixels

martinberlin, 2025-01-13:
It seems so. 
Doing it larger it will crash.
Recently got this one in Aliexpress:
ED115OC1 Eink Screen 2760x2070 EPD Display quite cheap and big actually.
up to 2000x2070 works but adding any row size larger than 2000 it crashes:

```
assert failed: 0x42009593
0x42009593: epd_hl_init at /home/martin/esp/projects/epdiy/src/highlevel.c:27 (discriminator 1)
```

Digging a bit deeper:

```c
    state.back_fb = heap_caps_aligned_alloc(16, fb_size, MALLOC_CAP_SPIRAM);
    assert(state.back_fb != NULL);
```

It seems that it fails allocating that kind of huge buffer: 
2760/2*2070 = 2856600

@vroland any idea if it could be possible to drive this big thing with the S3?

vroland, 2025-01-26:
Hm, how much RAM does the esp module you have have in total? Because the highlevel API is allocating a front buffer, a back buffer, and a difference buffer. In total, this would be 4x this calculated size: https://github.com/vroland/epdiy/blob/1bee9c3fd990b48f8916de13bc7d80c8384299e6/src/highlevel.c#L39
So 4x ~2.7MB is almost 11 MB of space needed. This shouldn't be a problem with a 16MB PSRAM chip, but it doesn't fit into 8, which many modules have. Maybe that is the issue?

martinberlin, 2025-01-26:
Thanks Valentin,
that seems to be the issue. But it should be possible also to use one framebuffer instead of front and back, right?
I mean if you just want to display a single image over white, that could be an option, to avoid using so much external RAM.

martinberlin, 2025-01-27:
Alright it seems that I won’t have an easy way to deal with big framebuffers unless in the future there is an option to avoid 2 frame buffers and diff.

closed 2025-01-27 by martinberlin